bigtreetech/BIGTREETECH-TouchScreenFirmware #1182: [FR] CHAMBER temoerature

DrumClock, 2020-10-18:
Please be able to display BED and CHAMBER temperature alternately.

[image]

oldman4U, 2020-10-18:
hi.

This is a duplicate of #707 where the user was not able to give the ticket a useful title and now he is not able to close the ticket or change the title.

Please close the ticket. This feature request is on the FR list which I will post very soon.

Thank you

bigtreetech, 2020-10-19:
Hello, In fact, it has been supported since https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/875
If CHAMBER is enabled in Marlin, a switching state of CHAMBER will be automatically added in the heater menu of touch screen, and the actual temperature and target temperature of CHAMBER will also be displayed on the top title bar.

DrumClock, 2020-10-19:
Yes, this is the case, but is it possible to display BAD and CHAMBER temperatures 
alternately in the same way as for NOZZLE T0 and T1?

oldman4U, 2020-10-19:
DrumClock.

@bigtreetech - welcome back, we missed you!;-)

Personally I do not use a heated Chamber, but from BTT's answer I would expect that this already works as expected in case everything is setup right in Marlin.

DrumClock, 2020-10-19:
Yes, in the "Heat" menu, the Chamber is fine.
I wanted the temperature to be shown on the home screen and on the screen when printing alternately with the temperature for
"BED" as with the  `nozzle T0 / T1`, `Fan F0 / F1` and `Speed / Flow `icons

**Is it already understandable?**

bigtreetech, 2020-10-19:
Yes, I understand. You want to alternate bed and chamber on the status screen

oldman4U, 2020-10-19:
No idea if this is already possible or not, but could you please activate this in config.ini and update your TFT or activate this directly in the feature list of the TFT to see if you can see the Chamber temp there.

#### Persistent temperature info
# show persistent temperature info on all menu screens
# Options: [enable: 1, disable: 0]
persistent_info:**1**

Please let me know

DrumClock, 2020-10-19:
Hi @oldman4U 
No change, the temperature CHAMBER z is displayed only in the corner of the screen but it works even without setting.

oldman4U, 2020-10-19:
Could you pls add a picture which shows this.

Thank you

DrumClock, 2020-10-19:
Hi @oldman4U there are ..
[image]
[image]
[image]
[image]
[image]
[image]

oldman4U, 2020-10-19:
OK. Picture 3 and 4 shows clearly that the Chamber is recognised correctly. So you should see the Bed and Chamber Temp. alternating - which is not the case. You can leave it a feature request and I will add it to the FR list instead of 707, or you change it to a Bug. Whatever fits better for you.

DrumClock, 2020-10-20:
OK, add it to the FR list instead of 707
Thanks

oldman4U, 2020-10-20:
Done.

Please check #1170 and close this ticket afterwards.

Thank you

oldman4U, 2020-10-21:
Hi.

I verified this also on my printer, even I have no chamber. Turning it on there is a chamber in the upper right corner, but the Bed does not changed to show the chamber. According to BTT this should work, so maybe it would even be best to change this from FR to Bug!?

Again, as you like;-)

DrumClock, 2020-10-21:
Hi @oldman4U 
So can you please change it to BUG?

oldman4U, 2020-10-21:
You are the owner, so only you can change it.

I will keep it also in the feature requests because it is unclear if it is A or B.

DrumClock, 2020-10-21:
how do I change `[FR] CHAMBER temperature # 1182 `to **BUG**?

oldman4U, 2020-10-22:
@radek8

Can you please help with this issue. At least to find out if this is a bug or not implemented. 

Thank you

radek8, 2020-10-22:
Hello
it is not implemented to be displayed on both status screens. only in the corner of the display.
This is what he would like to program.

oldman4U, 2020-10-22:
Cool. So it is a feature request.

Thank you Radek

oldman4U, 2020-10-23:
I checked the bmp's and there is nothing for Chamber, so it has never been implemented.

@DrumClock 

It is listed in the FR list. Could you please close this ticket and I will try to find an engineer to look at it.

Thank you

PR #2084 (2021-08-06): Clean up and optimize printing screen
- Clean up and reduce RAM usage in the Printing screen.
- Optimize the Printing screen to load faster.
- Prevent unnecessary status icon redraw in the Printing screen.
- Status icons on the Printing screen are now clickable after the print is finished.
- Clicking on the Layer/Z-layer status icon now opens the Babystep menu.
-  Misc. clean-up and reduce global variables.
-  Add chamber temperature status in status screen and printing screen (#1182). The bed heating icon will toggle between bed and chamber status If chamber heating is available.

Resolves #1 [... cut]
closed 2020-10-24 by DrumClock