bigtreetech/BIGTREETECH-TouchScreenFirmware #1183: [BUG] (short description)

chill-uk, 2020-10-18:
### Description

Everytime I try and enable z-offset to configure ABL, my Ender 3 tries to move past the end-stops, crashing the gantry.

### Steps to reproduce

1. Go to Movement
2. ABL
3. z-offset -> on

**Expected behavior**
3D Printer will auto-home, then probe the bed.

**Actual behavior**
3D Printer tries to move past the end-stop, causing the gantry to crash.

### Hardware Variant
BIGTREE_TFT35_V3.0_E3

### TFT Firmware Version & Main Board Firmware details
BIGTREE_TFT35_V3.0_E3.26.x.bin

### Setup
Ender3 + SKR Mini E3 v2.0

System: Marlin 2.0.7.2 (Oct 16 2020) [https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V2.0/firmware-bltouch.bin]

### Additional Information
* [Reddit Post](https://www.reddit.com/r/ender3/comments/jdfy00/btt_zoffset_abl_crashes_gantry/)
* [Youtube video](https://www.youtube.com/watch?v=nVrJM1YBoGo)

radek8, 2020-10-18:
Our problem is not caused by the touch screen but by the wrong setting of the limit switches in Marlin.
Check the Marlin configuration and perform a new compilation of Marlin

digant73, 2020-10-19:
also remember that PR #1154 will restore the original "software endstops" state exiting from Z offset menu.
So if in your Marlin FW you disabled "software endstops", that state will be restored when you exit from Z Offset menu.
Currently the software endstops are always enabled exiting from Z offset menu

chill-uk, 2020-10-19:
Hi @radek8 @digant73 , I have only used the official pre-compiled firmwares for both SKR and TFT.

If there is a Marlin config issue, it would suggest that they are present in those builds:
[SKR Mini E3 v2.0](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/firmware/V2.0/firmware-bltouch.bin)
[TFT35](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update%20-%20Unified%20Menu%20Material%20theme/BIGTREE_TFT35_V3.0_E3.26.x.bin)

I'm not sure if that helps ?

radek8, 2020-10-19:
Compile your own Marlin.
It is difficult to determine what parameters are stored in the BIN file.

oldman4U, 2020-10-23:
Please check if #1188 solves the problem. 

Do you know how to compile the firmware or do you need help?

digant73, 2020-10-23:
@chill-uk it's better that you compile the firmware. The problem should not be present also in the current main branch. Z offset menu simply performs homing and then moves nozzle to Z 0. The endstops are disabled just to allow to move nozzle to negative Z values (so to decrease Z offset value).
PR #1188 fixed the bug related to original endstops state not restored exiting from Z offset menu. But that should not be your problem

digant73, 2020-10-23:
From the video I see a z offset set to -1.85. Is it the value you set?

digant73, 2020-10-23:
You are also using an old version of the FW. Try the latest one

oldman4U, 2020-10-26:
@chill-uk 

Is your problem solved?

Please let us know 

Thank you

chill-uk, 2020-10-26:
Hi All,
I really appreciate all of your input and help. 
Unfortunatelly, I haven't had a chance to get around to compile my own firmware yet (life commitments).
Feel free to close the bug report if needed.

digant73, 2020-10-26:
I will compile it for you my current PR

digant73, 2020-10-27:
Attached the PR #1188 compiled for your TFT35. unzip it in the root directory of the SD card.
Change first the settings in config.ini according to your printer and then flash the FW.
Let us know the result

[TFT35.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5443931/TFT35.zip)

chill-uk, 2020-10-28:
Hi @digant73, I've tried your compiled version (and modified the config.ini), but the problem persists.

Unfortunately, I won't have much time to test/diagnose the issue (new baby due very soon), so I'll close the ticket.

Thanks everyone for your time and effort with this bug.
I hope to be able to contribute back somehow.

Stay safe 👍

closed 2020-10-28 by chill-uk

oldman4U, 2020-10-28:
Good luck!