bigtreetech/BIGTREETECH-TouchScreenFirmware #1184: [Q] TFT35 Printing Status not showing properly

zeruspocus, 2020-10-19:
Hello,

I've Just installed the SKR E3 mini V2 with the Tft35 screen and compiled Marlin 2.0.7 and the latest firmware available for the screen.

Everything works well, I can control everything on the printer; motors, autohome, Bltouch, etc, except that if I print within the touch screen mode it won't show the printing process properly at the beginning, nor temperature, fan or layer (when it's actually working).

 It will start the printing however, and only after the actual print starts it will show the temperature after a while, but not when it is heating up and if I cancel the print it goes into that loading msg with black screen and I have to restart the printer.
In Marlin mode everything works perfectly, so not sure if I'm missing something here. Any ideas?

Please see attached pictures.

[image]
[image]

Thanks in advance.

oldman4U, 2020-10-19:
Please update to the latest Master branch.

zeruspocus, 2020-10-19:
> Please update to the latest Master branch.

I'm not sure if I follow, what do you mean by latest Master branch? If you are talking about firmware I have the latest Marlin with the latest firmware available for the screen.

oldman4U, 2020-10-19:
Send a picture from Info screen. Which device do you print from? Do you have Cancel and End gcode activated?

zeruspocus, 2020-10-19:
> Send a picture from Info screen. Which device do you print from? Do you have Cancel and End gcode activated?

Thanks for your help, I just figured out what you were saying and yes, I've installed the latest Master branch from here: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/tree/master
I was just trying to compile a new firmware with a few different options on the config.ini file but I don't think is going to change much. 

I'm printing from the SD card on an Ender 3 Pro. I'll check now the info screen and the options of cancel and end gcode that I never checked before.

oldman4U, 2020-10-19:
Ok. 

And for the data not updating. 

Activate start gcode and add "M155 S1" to it.

zeruspocus, 2020-10-19:
[image]

oldman4U, 2020-10-19:
Or print from the TFT to get rid of the problems

zeruspocus, 2020-10-19:
> Ok.
> 
> And for the data not updating.
> 
> Activate start gcode and add "M155 S1" to it.

That is something that I should set on the printer? or on the starting gcode on the slicer?

oldman4U, 2020-10-19:
Both should work. 

Your system is empty. Here you should see Marlin version. [image]

zeruspocus, 2020-10-19:
I've just updated the firmware with those gcode commands and the cancel and end gcode enable and still the same, also the long names not showing.... 
I see what you mean about not showing the Marlin version.... I have the 2.0.7 with all the functions I need and it's been working great.
Is there any option to have in mind when compiling Marlin so it shows all this info to the screen? What am I missing here?
Again, thanks for taking your time with me!! I appreciate it!!

zeruspocus, 2020-10-19:
[image]

> Both should work.
> 
> Your system is empty. Here you should see Marlin version. [image]

Not sure how, but now it's there
Now I'm facing new issues, ABL failed to start and I can't get into Marlin mode.... if I can sort those I will give up with the touch screen tbh, it's nice but a lot of hassle... any idea?

oldman4U, 2020-10-19:
Whats the problem with Marlin mode?

zeruspocus, 2020-10-19:
[image]
It just gets stuck there now....

oldman4U, 2020-10-19:
When you rotate the knob?

Has this been working before?

zeruspocus, 2020-10-19:
> […](#)

Yes, it was working before. I just press and hold the knob until I get the option to switch to marling mode, once I select that it stays on that black screen.... and I've tried a lot things now, I went back to Marlin 2.0.bugfix and now back to 2.0.7... it's getting quite complicated :(

zeruspocus, 2020-10-19:
Regarding the temperature not showing I just tried to change the config.ini on the screen firmware with this option set to 1 (was 0 before) and it's the same:
#### Persistent temperature info
# show persistent temperature info on all menu screens
# Options: [enable: 1, disable: 0]
persistent_info:1

oldman4U, 2020-10-19:
ONLY CR10 Stockdisplay has to be activated in Marlin and the 10 pin cable has to connect to EXP3. Thats it

oldman4U, 2020-10-19:
This setting defines if the temperature is shown in the upper bar or not.

zeruspocus, 2020-10-19:
> ONLY CR10 Stockdisplay has to be activated in Marlin and the 10 pin cable has to connect to EXP3. Thats it

Yep, it is activated on the last Marlin I compiled..... are you using a creality as well with the same board and don't have this issues?

#define CR10_STOCKDISPLAY

oldman4U, 2020-10-19:
I use 2 E3 DIP without problems.

No Marlin I get when I disconnect the cable or select wrong or two types of
Displays in Marlin.

zeruspocus, 2020-10-19:
OMG, I'm so stupid, it wasn't working cause the ribbon cable was loose, sorry for wasting your time with that and thanks for your patience. 

Anyway, all the other issues are sorted except the fact that is not updating the temperature and the layer height properly, I can live without that but I'm taking it personal now, I can't loose against the machine hahaha, I just started in the 3d printing world a bit more than a month ago and I knew 0 about everything, didn't even know how to use the arduino or the vscode, so I made quite a lot of progress and managed to get everything I wanted working always, it's hard to accept that I can't do this simple thing!! :D 

I saw on another post someone mentioning something about this on the marlin which I have commented on mine:

The only display uncommitted should be

define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

oldman4U, 2020-10-19:
Layer hight is not shown when printing from mainboard sd card, but everything else is working. This movie has been made with Marlin 2.0.7 and last Master TFT firmware:

https://www.dropbox.com/s/hfuxoeiqku00knv/IMG_7361.mov?dl=0

oldman4U, 2020-10-19:
We all make mistakes every day;-)

In any case ONLY one display can be defined in Marlin. Depending on the mainboard and therefore how the cables are connected, CR10 or RepRap has to be defines as display.

Happy printing and PLEASE close all tickets you do not need anymore.

Thank you

zeruspocus, 2020-10-19:
> Layer hight is not shown when printing from mainboard sd card, but everything else is working. This movie has been made with Marlin 2.0.7 and last Master TFT firmware:
> 
> https://www.dropbox.com/s/hfuxoeiqku00knv/IMG_7361.mov?dl=0

I would love to see that on my screen, but it is not happening like that at all.... would you mind if I leave the case opened until I find a solution? Cause that was the main reason of this post.

oldman4U, 2020-10-19:
You can keep the ticket opened as long as you want and it makes sense.

I can try to help you further and if the initial reason of the ticket still exists we can use this ticket.

So Marlin works, but there is no update of heating and fan before the print starts, right? Which slicer are you using?

zeruspocus, 2020-10-19:
> You can keep the ticket opened as long as you want and it makes sense.
> 
> I can try to help you further and if the initial reason of the ticket still exists we can use this ticket.
> 
> So Marlin works, but there is no update of heating and fan before the print starts, right? Which slicer are you using?
> 
> And does it update when you preheat before you start the print?

I've found the problem and it was me, I just realized that it will show that info only if you are printing from the USB, so far I was still using the microSD card directly into the board, which make sense to me but it apparently doesn't send the info back straight away. 
So thanks a lot for all your help and for sure I will come up with something else and I'm always happy to help anyone else with anything that I might already been through!

Cheers and now yes, happy printing! 👍

closed 2020-10-19 by zeruspocus

oldman4U, 2020-10-19:
The movie shows printing from mainboard sd, so it is possible. But I agree, it is buggy and personally, I do not like the micro SD cards and prefer the USB sticks. I only activated it because I wanted try help another user, which allows me to learn a lot about this stuff;-)

Wish you the best