bigtreetech/BIGTREETECH-TouchScreenFirmware #1186: [Q] TFT 35 V3 - Touch Screen Not Working

NevilBiscuits, 2020-10-19:
I am having an issue with the touch screen not working. The LCD works fine but not the touch screen mode.

Has anyone had this issue and also is there a way to re initiate the touch screen calibration in case this is the issue? (note the calibration did not detect touch the first time it was run previously).

oldman4U, 2020-10-19:
Maybe you reading the ReadMe and there especially the Troubleshooting area could answer your question.

oldman4U, 2020-10-19:
Did this help?

zeruspocus, 2020-10-19:
> I am having an issue with the touch screen not working. The LCD works fine but not the touch screen mode.
>
> Has anyone had this issue and also is there a way to re initiate the touch screen [...]

It happened to me when I first installed and I had to put some electrical tape around the edges of the Ender 3 display cover cause it was touching there and the screen was becoming unresponsive.... not sure if you have the same issue there but take a look, maybe trying to see if it works without being on the cabinet.

oldman4U, 2020-10-20:
Any news?  Please let us know

NevilBiscuits, 2020-10-20:
> It happened to me when I first installed and I had to put some electrical tape around the edges of the Ender 3 display cover cause it was touching there and the screen was becoming unresponsive.... [...]

No it's not in a case, it's currently hanging freely, hasn't worked since I first hooked it up.

oldman4U, 2020-10-20:
Have you connected and reset like described?

NevilBiscuits, 2020-10-20:
Where is it described sorry?

zeruspocus, 2020-10-20:
> No it's not in a case, it's currently hanging freely, hasn't worked since I first hooked it up.

Which board are you using it with? 
Maybe a silly question, but did you try to reverse the serial cable? If you look at the bottom of the mainboard it says what are the pins, the one that come loose is the reset just for reference.

NevilBiscuits, 2020-10-20:
It's connected to an SKR 1.4 Turbo, bought together and installed together.

oldman4U, 2020-10-20:
Pictures are good. 

ReadMe on the mainscreen of btt tft firmware.

Does not work is still undefined. Does it say no printer attached or is the screen dark or no reaction when touching it. Has it been working before?

radek8, 2020-10-20:
Hard work to get the necessary information so you can help someone :-)

oldman4U, 2020-10-20:
More works than reading the very basic manual😁

NevilBiscuits, 2020-10-20:
> Hard work to get the necessary information so you can help someone :-)

Hard work reading pointless comments. I have a job working 12.5hrr shifts, so sorry if I can't answer a question that exact minute.

NevilBiscuits, 2020-10-20:
oldman4U,

The printer works fine, it boots up and after around 12 seconds the printer connection is established.

The TFT serial port cable matches the pin outs like for like on the board and screen.

I'm 99.9% certain it's a fault with the actual touch surface, either a bad connection or something else, there appears to be an air gap between the screen and the touch overlay on top of the screen, not sure if it may have been incorrectly installed or damaged in transit.

I'm not a newb to 3D printers, I've read all the documentation for installation, I'm just curious if someone may have experienced this very problem and found a solution. I appreciate your assistance with trying to solve this, but I am thinking the unit is bricked and needs replacing. Just thought that maybe there was a slim chance someone had the same issue and had a solution before I fork out for a new TFT (unless BIQU come through with a solution that is, I have a support request active at the moment).

![Uploading 96622644...]

![Uploading 96622662...]

oldman4U, 2020-10-20:
We all have jobs.

Whats about the answers?

NevilBiscuits, 2020-10-20:
I just replied to your question.

radek8, 2020-10-20:
It is clear to me that you are in a different time zone and go to work and sometimes sleep :-D
I will be silent   :shushing_face:

NevilBiscuits, 2020-10-20:
lol yes, and after 6hrs sleep, I'm now back off to work :(

radek8, 2020-10-20:
Did you perform the calibration via the reset.txt file?

oldman4U, 2020-10-20:
Todd, you sleep too much.

Sorry, joke. You have to know that we are not 100% serious, more 50%.

I am 100 and Radek is 0...;-)))))

radek8, 2020-10-20:
:astonished:

oldman4U, 2020-10-20:
Oh Radek my friend. Don't be sad. You are a hero!!!

Using the reset.txt probably solves a TFT not reacting in touch mode.

I would ONLY connect the serial cable and disconnect the two other cables to see if the cable delivers power. Cable problems we have seen  a lot. Right radek8!!??

🤣

radek8, 2020-10-20:
It's the way you write.
If the calibration of the touch layer does not help, then only the damaged HW

NevilBiscuits, 2020-10-20:
Unfortunately calibration has never worked since new, tried using a stylus also to touch the calibration points, no response.

radek8, 2020-10-20:
can you take a picture of the problem with the touch layer that you described?

Portzal, 2020-10-21:
I see a few clowns, but is there anyone actually in charge of this circus?

oldman4U, 2020-10-21:
@Portzal 

And I thought I am the only one seeing them... especially this old clown, right!?;-)

Not a single person or company is in charge, but every single user. At least this is what I believe that Open Source means. BTT should monitor this, but their impact is low.

majo91, 2020-10-21:
I had the exact same issue. My display worked very good and i made an update to the current version and the touchscreen didn't work. I created an empty file named reset.txt on the SD card, plugged it in and made a reset - then i calibrated the display and now it works.

oldman4U, 2020-10-21:
FYI.

You should be able to get response from the Touchscreen as soon as it gets power from the EXP and/or RS232 cable. In case the reset.txt file on the SD Card does not help, I would recommend to contact the dealer you bought the TFT from, because it seems to be broken.

So I would test it with a single cable and the reset.txt and in case this does not help repeat the rest.txt with the other cable.

Let us know

Thank you

anilcancakir, 2020-10-21:
I had this issue after firmware updating.

Follow these steps:
1. I created an empty file that name is `reset.txt` on SD Card.
2. Plug SD Card to TFT Screen
3. Reset or Power On the device and you will see `Adjust Touch` screen.
4. Adjust touch and done.

NevilBiscuits, 2020-10-22:
So I don’t know what happened, but for some reason (as it all seemed exactly the same as previously) the screen calibration decided to work and I now have a touch screen. Thank you for your assistance 🙏 much appreciated 👍
[image]

closed 2020-10-22 by NevilBiscuits

oldman4U, 2020-10-22:
👍🏻😁

Happy printing