bigtreetech/BIGTREETECH-TouchScreenFirmware #1190: crash when entering tune menu

Thijsvanzwam, 2020-10-20:
### Description

When entering the tune menu, complete TFT freezes

### Steps to reproduce

go to setting and then to tune.

**Expected behavior**
be able to tune...
wanted to tune the PID

**Actual behavior**
freeze

### Hardware Variant
TFT3,5 v3.0 and Marlin 2.0.7.2 on BTT SKR 1.4
[TFT config files.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5409155/TFT.config.files.zip)
[Marlin.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5409159/Marlin.zip)

### TFT Firmware Version & Main Board Firmware details
latest but happend on previous version as well

### Additional Information
Because I don't know if the marlin firmware is the cause or the tft firmware, I included both.
the same happened with the previous versions or Marlin and TFT firmware, as well as with the bugfix

radek8, 2020-10-20:
While updating the FW display, you received an error message about a missing icon.
So why set up an error ticket without resolving the issue with the missing icon?

radek8, 2020-10-20:
look #1133

CBDesignS, 2020-10-20:
support from the mfr is zero and people do not know how to search on github. Maybe a common fault finding page should be added to save all these waste of time tickets. It must be easier to open a ticket than it is to search open and closed bug reports.

radek8, 2020-10-20:
People don't read manuals. it is better to ask 3x than to read the manual once :-)
But when I ask, I should provide all the information, such as the fact that an error message appeared during the installation, and write its wording.

oldman4U, 2020-10-20:
This is from the pinned README BEFORE POSTING from traffic-light:

**Q: Unexpected freezes when entering a menu after newly compiled firmware
Answer: Did you also update the config.ini on you're screen after the update? For example the freezing in preheat menu is caused by using a old config.ini.
Please try if this work. When it didn't work place an issue and tell that you did update config.ini ;-)**

and this is from the user manual:
**Errors during the update can not be ignored and must be solved before using the TFT**

Not sure you have seen that there is a new pinned Feature Request ticket, which holds (almost) all features which have been requested in the past. I am thinking about doing the same with known issues, so that all verified bugs are also in a list.

Not sure someone will read them, but it would give more structure and the search function in this ticket system is a nightmare.

Finally I believe it would be best to cover only issues and feature requests here, excluding hardware related issues or user errors. Like this is done with Marlin.

Thijsvanzwam, 2020-10-21:
> While updating the FW display, you received an error message about a missing icon.
> So why set up an error ticket without resolving the issue with the missing icon?

I already did, the missing icon I added so the error was gone. As you can read in my post, the freeze also happened when using the previous firmware version. There was no error regarding missing icons.

> look #1133

indeed, this I why I checked it first by adding the missing icon.

> support from the mfr is zero and people do not know how to search on github. Maybe a common fault finding page should be added to save all these waste of time tickets. It must be easier to open a ticket than it is to search open and closed bug reports.

true, would be nice to have a common fault finding page. 
Sorry my post was a waste of your time. Perhaps you shouldn't read it?

> People don't read manuals. it is better to ask 3x than to read the manual once :-)
> But when I ask, I should provide all the information, such as the fact that an error message appeared during the installation, and write its wording.

not true, I did read the manual, I posted because I thought this is a way of helping people who have issues after trying, just in case I missed something or there might even be a real problem. Perhaps it is better to not use this platform then as people find it annoying to react in a normal way and start pointing fingers. There was no reason to mention the error message as I fixed that earlier and way before posting this.

> Any news?

No, i do have something like a job so I am not testing the firmware all day long.

oldman4U, 2020-10-21:
Could you please be so kind and update to Marlin v2.0.7.2 and the TFT version from Oct. 20th and to attach your configs again. I would then upload this to my SKR v1.4 and TFT35 E3 v3 to see if this is reproducible somehow. Maybe also a short movie showing the crash could help.

Thank you

oldman4U, 2020-10-22:
And what about your old open ticket. Is this solved in the meantime?

oldman4U, 2020-10-24:
Hi.

Any news?

Please let us know 

Thank you

oldman4U, 2020-10-26:
Please help the community and close the ticket once you do not need it anymore. Thank you

Thijsvanzwam, 2020-10-26:
> Could you please be so kind and update to Marlin v2.0.7.2 and the TFT version from Oct. 20th and to attach your configs again. I would then upload this to my SKR v1.4 and TFT35 E3 v3 to see if this is reproducible somehow. Maybe also a short movie showing the crash could help.
> 
> Thank you

This is the version installed, tried marlin bugfix to but that made no difference. 
The problem first occurred using the TFT firmware about 4 months ago. I tuned my PID using Marlin mode so I did not need it anymore. The problem occurred in all versions of the firmware after that. Tried different icons as well but no luck.
I gonna try to redraw some icons (open them in photoshop, edit something small and then save as the same filename.) That worked on my GRBL Laser CNC for some reason.

Thijsvanzwam, 2020-10-26:
> And what about your old open ticket. Is this solved in the meantime?

closed it. solved since marlin 2.0.7.2 but forgot to close the ticket.

oldman4U, 2020-10-26:
So I can use the attached TFT firmware with the latest release?

Could you upload a short movie which shows the crash and also a picture of the Info screen?

Thank you

Thijsvanzwam, 2020-10-26:
well actually, fixed it.
replaced the name of tune_extruder.bmp to Tune_extruder.bmp (with a capital) and it does not crash anymore.
the error during upgrading the tft was gone already with the tune_extruder.bmp without a capital and replacing it with the capital one did not give any errors during upgrade.
As it works now, no short movie of crashing to attach...sorry

I attached the following files:
TFT files.zip contains both config files compiled with the latest release of the BTT tft firmware
Marlin files.zip contains the config files of Marlin 2.0.7.2

it works for me now!

[TFT files.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5441758/TFT.files.zip)
[Marlin files.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5441759/Marlin.files.zip)

oldman4U, 2020-10-26:
Even better, very good!;-)

I will check this on my machines also if this makes a difference.

Thank you

digant73, 2020-10-27:
tune_extruder.bmp is in lowercase on all the TFT variants and no problem for other users. Also many other icons have names with different naming conventions. It seems to me the problem is more related to a bug elsewhere (e.g. in the file itself) more than related to the use of a capital letter in a file name.

closed 2020-10-29 by Thijsvanzwam

oldman4U, 2020-11-01:
So I wanted to know if there is any sensitivity when it comes to the names of bmp's. So I renamed a few of them by changing the capital and small letters and I can confirm that for the name of the bmp's the format does not matter.

Even dISCONNECT was accepted. So it must be something else which caused the issue.