bigtreetech/BIGTREETECH-TouchScreenFirmware #1193: [BUG] (short description)

misiek271, 2020-10-21:
### Description
hello i have tft 35 e3 v3.0 newest firmware updated yesterday. and skr 1.4 conrctrd via rs232 cable. 
TFT 35 has got in config ini : ups enable or disable but in comment is that ups should be connected to tft. I cant find any info iabout what pin i should connact it to tft.?? can someone tell me. 
So i connected it to skr 1.4 p1.26 And turn on in marlin power loss etc. 

in tft config i add m75 at start and m77 at cancel and end gcode.
my nozzle dont rise when i unplug power. and when i connect it back and take reasum print all print start from ten moves back. 

### Steps to reproduce

1. starting print
2. wait for print first layer almost all first  layer
3. cut off power
4. connected it again
5.reasume print in tft

**Expected behavior**
after cutoff: nozzle should rise up
after resume printer starts print from the same moments

**Actual behavior**
when i cut off power nozzle dont move. just stay in place
when i reasume print it should print from the same moment that stop but it prints about 10 moves elier. 
>start printing first layer and wait almost to end of first layer >cut off power> resume> print starts form first move of first layer

### Hardware Variant
TFT 35 E3 V.30
skr 1.4 marlin 2.0.7.2

radek8, 2020-10-21:
[image]

misiek271, 2020-10-21:
Nope. Radek its for psu unit. Not for ups. Thats 2 different module.

radek8, 2020-10-22:
Sorry, you're right.
Please add a description of your problem.

radek8, 2020-10-22:
Looks like the pin isn't defined.
Try to redefine a pin that you do not use

oldman4U, 2020-10-22:
Please update the title of the ticket. 

Thank you

misiek271, 2020-10-22:
to define pins i need some file that show what pin is what function . like in marlin. But in tft i cant find something similar. can You tell me where is it?

oldman4U, 2020-10-22:
Thank you for updating the title. Hope this will help to find the right person knowing the solution.

radek8, 2020-10-22:
BIGTREETECH-TouchScreenFirmware\TFT\src\User\Variants\pin_TFT35_V3_0.h

misiek271, 2020-10-22:
i checked it but there is no info about ups or miniups pin. only ps on and filament runout.

radek8, 2020-10-22:
that is the problem

radek8, 2020-10-22:
@bigtreetech , Is UPS module support currently working?
No PIN is currently mapped.

misiek271, 2020-10-26:
i have got info from BTT and MINI UPS DON'T WORK with tft touch mode. SO its useless if you want use only touch mode. SHAME on YOu @bigtreetech. There is no official info about that. no word in user manual/video/alliexpres/your WWw site. and i checked skr 1.4 /tft 35 / miniups manuals. Thats very bad coz you selling only half working tft and You know it and don't tells customer...... Its a good subject for Youtube video tutorial if you want use tft mode with mini ups dont buy from BTT.and i dont mension that if someone want to use both mode must replug fil sensor, auto off module, from MB/TFT depend what mode you want to use i didnt see any info about that too in manuals that i read.

oldman4U, 2020-10-28:
radek8. Does it make sense to add this as a FR?

radek8, 2020-10-28:
as yes

oldman4U, 2020-10-28:
misiek271

Would you like to change the Bug label to FR?

berrenger, 2020-11-04:
Hi all. My hardware:
Skr 1.4 turbo, tft35 v3.0, btt relay v1.2, btt ups 24v, btt smart filament sensor, bltouch 3.1.

Sfs connected to tft35, relay and ups connected to skr board. On tft 35 graphic mode all functions are working but only marlin 2.0.5 version. Not work above marlin 2.0.5. why could this be?

https://youtu.be/pIcadt8rKAo

misiek271, 2020-11-04:
Man filament sensor and auto off mst be connected to tft when using tft mode. If using markin mode must be connected to skr. But mini ups dont work in tft mode only in marlin. Sell this staff and buy mks nano v2 tjey work tft on spi so all functiin works and you can connect all to MB. And mks has got eeprom and all features of btt and better tft coz its on spi connection. Btt is really bad if you want use touch mode.

berrenger, 2020-11-05:
> Man filament sensor and auto off mst be connected to tft when using tft mode. If using markin mode must be connected to skr. But mini ups dont work in tft mode only in marlin. Sell this staff and [...]

It works the way I connected it man. Watch the video

eragon001, 2020-11-18:
> It works the way I connected it man. Watch the video

which version of the 2.0.5 your using?

berrenger, 2020-11-18:
> > It works the way I connected it man. Watch the video
> 
> which version of the 2.0.5 your using?

2.0.5.4

BisSpaeterDannnoch, 2020-12-20:
> i have got info from BTT and MINI UPS DON'T WORK with tft touch mode. SO its useless if you want use only touch mode. SHAME on YOu @bigtreetech. There is no official info about that. no word in user [...]

i feel like you.

closed 2021-02-26 by stale[bot]