bigtreetech/BIGTREETECH-TouchScreenFirmware #1196: [BUG] (If select print from SD cart on board (no TFT) then halt)

dzario, 2020-10-23:
### Description
If select print from on board SD card reader, and is not card inserted then halt LCD. If insert any SD card with any gcode file 
then show error : Unknown command "M33 benchy.GCO"

### Steps to reproduce

1. TFT Touch mode
2. Touch to icon Print
3. SD card onboard

**Expected behavior**
List file on SD card or show no file

**Actual behavior**
Halt TFT.
If not inserted SD card or inserted with file.

### Hardware Variant
SKR Mini E3 V1.2
### TFT Firmware Version & Main Board Firmware details
TFT 2.4 V1.1 FW: 26 11 April 2020
Marlin 2.0.6 SD card support
### Additional Information
I don't need visible this icon, is possible disable this icon?

oldman4U, 2020-10-23:
Please update to latest TFT firmware and Marlin 2.0.7.2.

Which Icon?

dzario, 2020-10-23:
Main screen , right down -> Print . Then Show on Screen 3 Icon, TFT SD , On board SD , USB .
If touch to On board SD .. then halt.

oldman4U, 2020-10-23:
Once you updated the TFT firmware according to the description, you can edit the config.ini in a text editor and there you can change this:

**#### Onboard / Printer SD Card Support
# On Marlin Firmware, The TFT will auto detect Onboard SD Card
# Auto-detect is not available for other firmwares like Smoothieware
# Options: [enable: 1, disable: 0, auto-detect: 2]
onboard_sd_support:2**

to this:
**#### Onboard / Printer SD Card Support
# On Marlin Firmware, The TFT will auto detect Onboard SD Card
# Auto-detect is not available for other firmwares like Smoothieware
# Options: [enable: 1, disable: 0, auto-detect: 2]
onboard_sd_support:0**

and upload the new config.ini.

This will remove the button from the TFT.

Let me know

oldman4U, 2020-10-23:
Sorry for the big font, comes from copy and paste....

;-(

dzario, 2020-10-23:
..after upload https://github.com/oldman4U/BIGTREETECH-TouchScreenFirmware/tree/Oldman4U .. hide this icon .. now not possible select it .. good work..

closed 2020-10-23 by dzario

oldman4U, 2020-10-23:
Great. 

Check all the other options in config.ini and change according to your needs. 

Happy printing