bigtreetech/BIGTREETECH-TouchScreenFirmware #1197: [Q] (Disable switch TFT simulation 12864)

dzario, 2020-10-23:
Is possible disable function to switch UI between Touch Mode <-> Marlin Mode ???
I don't need this function .. also encoder ..

oldman4U, 2020-10-23:
Yes, you can. But not in software.

[image]

dzario, 2020-10-23:
:) .. no this is not way .. if you touch finger to screen and hold ... then show you menu select Marlin Mode or Touch mode .. no encoder know need hold .. software solution is possible but must change source code..

oldman4U, 2020-10-23:
The TFT can not make a difference between a short and a long press on the screen. So you have to use the encoder to initiate the switch mode, which then allows you to select the mode you want to use by rotating the encoder or by pressing the touch screen.

But you asked if it is possible to DISable the switch......!?

dzario, 2020-10-23:
...sorry... switch between .. Marlin and touch ..  
I change source code .. to disable it .. 
just must find it on c or h file 
regards.

oldman4U, 2020-10-23:
OK. Not sure what you want to achieve, but it is great if you can change the code so it fits your needs.

Please close this ticket once you do not need it anymore.

Thank you

dzario, 2020-10-23:
.. I do it ..
In file Selecmode.c in function void loopCheckMode(void) I add command return; 
After line 80 , make new line nad write `return;` 
Now do not check if encoder or touch screen pressed more then predefined time ..

closed 2020-10-23 by dzario

oldman4U, 2020-10-23:
Cool!

Sounds you can develop stuff. Want to contribute something else??

dzario, 2020-10-23:
Thanks. 
I design very special 3D printer. 
I'm totally new for STM , Platformio ...etc.. then it's like snail for me .. but still  find solution how solve my request.
For this time I use standard 12864 LCD with SKR Mini E3 V1.2 . Problem is so LCD is too big for my project. 
First prototype used ATMega2560 and small Oled LCD , but SKR Mini E3 not supported I2C LCD .. .
TFT2.4V1.1 is super , USB support ,SD cart, touch screen and all is just using gcode.. 

I need just small help to understand how firmware  work.. and then I can finish it self .. 
Best regards.

oldman4U, 2020-10-23:
Sounds like a good plan.

👍🏻

dzario, 2020-10-23:
..ehm... compile .. flash .. but only english language .. I don't know where I make mistake .. but still only one ..
Tomorrow .. is also day ..