bigtreetech/BIGTREETECH-TouchScreenFirmware #1199: [BUG] Terminal outputs "Busy paused for user" after every print

blueeagle69, 2020-10-24:
Hi.

After every print the terminal outputs a constant "busy: paused for user" and will not respond to any further commands.
However if you press the encoder wheel  once the busy command goes away and the printer will then respond.
Also if you click the encoder once after a print the screen does not respond to the first click, but it does on subsequent clicks.
It's like as if the screen is asleep and the click of the encoder wakes it up, because once I click the encoder the LCD goes to full brightness and the busy output to the terminal goes away.
The issue occurs in TFT and Marlin mode,
I can't tell if this is a Marlin firmware or TFT issue, or a "feature".  But I do now I never got it on my other printer that has a standard full graphic LCD.

Thanks.

oldman4U, 2020-10-25:
Do you have end gcode activated?

blueeagle69, 2020-10-25:
Hi.

No, End Gcode is not activated.
I could test with it activated.
But saying that, does the start/end Gcode get used in Marlin mode? I don't think it does.

Thanks!

oldman4U, 2020-10-25:
Try it and we all will know.

blueeagle69, 2020-10-26:
Hi.
I enabled End Gcode.
Printed from Marlin mode.
Same thing.
Printer constantly outputs busy message at the end of a print until the encoder is clicked.

oldman4U, 2020-10-26:
And what does the TFT. How does it look like after a print?

blueeagle69, 2020-10-26:
Hi.

I will  do a test later and respond.

Thanks.

blueeagle69, 2020-10-26:
Hi.
I'm pretty-sure when printing from TFT it brings up a "Print Complete" dialogue box with an "OK" button.
Could that be why in Marlin mode it outputs a busy message until the encoder is clicked?

blueeagle69, 2020-10-27:
Okay, when printing in TFT mode it brings up the "Print Complete" box.
While this is on the screen the printer outputs the busy message.
When I click Okay, I am left with this screen.
[image]
Requiring me to hit Stop.

oldman4U, 2020-10-27:
Well, this looks exactly the same on my screens.

I know only one who can help us, @radek8.

Do you know if this is the expected behaviour or not?

Please help

Thank you

radek8, 2020-10-27:
Check the end code in your slicing software.
A pause starts before the end of printing.
According to the icon in the photo, it looks like the print is not finished but only paused.
Please attach a file to print that you have a problem with. We'll see if my reasoning is correct

[image]

blueeagle69, 2020-10-28:
Hi.
My end Gcode consists of M300 tone commands which of course only work in Marlin mode, and an
M524 which I was testing to see if it would end the print and stop the TFT bringing up the pause at the end of the print, which not did not.

CBDesignS, 2020-10-28:
I had a problem simmilar when I tweaked the end gcode in cura a while ago. I just copied most of the code from one of my previous repg generated files thinking that it should work . it did but it caused a bit of pausing at the end of the print. the only problem is I can not remember what gcode command it was that done it. Maybe post your end code from your slicer then more knolegable users maybe able to help you ??

blueeagle69, 2020-10-28:
G91 ; Absolute positioning
G1 E-5 F300
G27
M104 S0 ; Set Hotend Temperature to 0
M140 S0 ; Set Bed Temperature to 0
M18  
M300 S860 P150 
G4 P100 ; Wait 1 second
M300 S960 P100
G4 P100 
M300 S1160 P150
M150 R0 G0 B0 W0

CBDesignS, 2020-10-28:
I am no gcode wizard but I though a G27 command needed passing parameters. 
I have just looked at the Marlin Gcode bible and it says :- 
G27 [P<0|1|2>]

[P<0\|1\|2>] | Z axis action   P0: If current Z-pos is lower than Z-park then the nozzle will be raised to reach Z-park height   P1: No matter the current Z-pos, the nozzle will be raised/lowered to reach Z-park height   P2: The nozzle height will be raised by Z-park amount but never going over the machine’s limit of Z_MAX_POS
-- | --

to me that might as well be written in greek .

blueeagle69, 2020-10-28:
No, G27 will run without parameters. You only need the P Switch if you want the Z axis to rise.

CBDesignS, 2020-10-28:
ok. just a thought. I just send the extruder to a position with a G0 command then I know it will always lower my bed (upside down printing) and be well away from the final print. G0X255Y150Z140

blueeagle69, 2020-10-28:
I'm not really bothered with the fact the TFT doesn't end correctly as I mostly print in Marlin mode.
What I was on about was the fact that when I'm printing in Marlin mode, the printer sends out a constant Busy message until I click the encoder (Marlin) or in TFT mode, close the Print Complete box.
Actually if the TFT brings up a Print Complete box, then surely it isn't pausing anywhere?

CBDesignS, 2020-10-28:
the only other thing is you use M150 R0 G0 B0 W0 and I think the G0 shouldd be U0 as G0 is a gcode command. maybe try changing M150 R0 G0 B0 W0 to M150 R0 U0 B0 W0 ???

blueeagle69, 2020-10-28:
Thanks, I will do that, although I don't think it will help the overriding issue with the constant Busy output which requires me to click the encoder.

blueeagle69, 2020-10-29:
I just decided to try a test print with no end Gcode defined in my slicer, sure enough the print ended, and in my terminal I had the normal "echo:busy: paused for user". Once I clicked the encoder the pause message went away.

radek8, 2020-10-29:
Do you have end code enabled in marlin?
The Configuration.h file
// End G-code - run this G-code after finishing print
#define PRINT_END_GCODE "G90\nG1 E-4\nG92 E0\nM18\n" // Switch to absolute positioning, reduce filament pressure by performing small retract, reset extruder position, disable steppers

blueeagle69, 2020-10-29:
Unless I'm mistaken, no I don't.
In fact I cannot find that setting in the latest Marlin.
DO you mean the TFT firmware?
Because I can find that line in the TFT firmware but not Marlin.
In the TFT I have tried with it enabled or disabled, same result, printer outputs busy at the end of a print until I click the encoder.

blueeagle69, 2020-10-29:
Just printed again (In TFT mode) with no end gcode defined in my gcode and End Gcode disabled in LCD.
Same issue.
The printer outputs this;
echo:busy: paused for user
echo:busy: paused for user
echo:busy: paused for user
I click the OKAY button on the Print Complete box and I get this; (In Marlin Mode I have to press the encoder to get rid of the paused for user.)
**M106 P0 S0**

radek8, 2020-10-30:
send the file to print you have a problem with!

blueeagle69, 2020-10-30:
Take your pick it happens with every sliced gcode with or without any end gcode in the slicer or the tft. But I will upload a gcode for you.

oldman4U, 2020-10-30:
radek8. Could you make a short movie showing what happens on your TFT at the end of a print. What blueeagle69 describes is exactly what I can see also, even I thought this would be a strange behaviour and not really a bug...

blueeagle69, 2020-10-30:
Cheers @oldman4U

CBDesignS, 2020-10-30:
This is past the limits of my help so I will have to bow out and leave this to others. but before I pop off are you using tft35 E3 v3 ? I only have the tft35 v3 and have not seen this happen since I removed a rogue unsupported gcode from my  Replicator G end code days. Maybe this is specific to E3 V3 screens

blueeagle69, 2020-10-30:
My screen is apparently a variant of the TFT35 E3 according to Biqu/Bigtreetech which came installed on my Biqu B1. It has the colour changing encoder but MicroSD instead.

blueeagle69, 2020-10-30:
> This is past the limits of my help so I will have to bow out and leave this to others. but before I pop off are you using tft35 E3 v3 ? I only have the tft35 v3 and have not seen this happen since I [...]

I did try it with or without End Gcode in my slicer (Prusaslicer) and the TFT. Same issue regardless.
Thanks.

radek8, 2020-10-30:
send the file to print

blueeagle69, 2020-10-30:
Here you go.

[xyzCalibration_cube_0.2mm_PLA_36m.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5467966/xyzCalibration_cube_0.2mm_PLA_36m.zip)

radek8, 2020-10-30:
it looks fine.
So it will be a setting in Marlin ...

blueeagle69, 2020-10-30:
It may be a setting causing the TFT screen not to end correctly.
But the printer outputting the Busy paused for user, which goes sway when I press the encoder or close the Print Complete box when in touch mode is down to the TFT.

blueeagle69, 2020-11-05:
@oldman4U Closed as it's kindly been added to the list.

closed 2020-11-05 by blueeagle69

oldman4U, 2020-11-05:
I will take care of your bugs and fr. For this particular ticket I will upload a Movie which shows the end of a print on my printer asap. So we can see if radek8 gets something different.

oldman4U, 2020-11-08:
Hi.

I wanted to know from you radek8 and blueeagle69  if your screen behaves the same after a print, so I made a short movie showing my TFT35.

Please let me know

Thank you

[End of print.mov.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5506816/End.of.print.mov.zip)

oldman4U, 2020-11-08:
So there is NO Print complete message on my screen after a print

radek8, 2020-11-08:
Oldman4U on your video has the print done correctly.
The STOP icon changes to BACK when done

[image]

[image]

oldman4U, 2020-11-08:
But I have no confirm print and using Esp WIFI connection I also see no
strange communication. So where does this confirm comes from?

radek8, 2020-11-08:
What confirmation do you mean?

oldman4U, 2020-11-08:
What blueeagle wrote a few days ago.

_Then again. In TFT mode it correctly brings up the "Print Complete" so I can't see it being a "Marlin Setting"._

radek8, 2020-11-08:
That he would use old FW?

blueeagle69, 2020-11-08:
Hi.
With my Marlin and TFT is up to date. 
I get the Print Complete in TFT but then I get the Stop button showing after clearing the Print Complete popup. 
I also noticed yesterday that if I print in Marlin mode and switch back to TFT afterwards,  I'm still greeted with the Stop button, but this time it does nothing.

radek8, 2020-11-08:
> Hi.
> My Marlin and TFT is up to date.
> I get the Print Complete in TFT but then I get the Stop button showing after clearing the Print Complete popup.
> I also noticed yesterday that if I print in Marlin mode and switch back to TFT afterwards, I'm still greeted with the Stop button, but this time it does nothing.

So did you compile your own FW into TFT?

blueeagle69, 2020-11-09:
I compile my own.
The only thing I do change is in Selectmode.

Changing "12864 Mode" to "Marlin Mode" It triggers me!

radek8, 2020-11-09:
> I compile my own.
> The only thing I do change is in Selectmode.
> 
> Changing "12864 Mode" to "Marlin Mode" It triggers me!

to nastavíš v config.INI
#### Text displayed at the top of the TFT in Marlin Mode.
marlin_title:LCD12864 Simulator

radek8, 2020-11-09:
> Co napsal blueeagle před několika dny.
> 
> _Pak znovu. V režimu TFT správně vyvolá „Tisk dokončen“, takže nevidím, že se jedná o „Nastavení Marlin“._

haven't you changed this setting?

 * Options: [OFF: 0, POPUP: 1, TOAST: 2]
 *  OFF:   No notification. The message is ignored.
 *  POPUP: Display a popup window for user confirmation.
 *  TOAST: A Toast notification is displayed for few seconds. No user interaction is needed
 *
 */
#define ACK_NOTIFICATION_STYLE 1

blueeagle69, 2020-11-09:
> What confirmation do you mean?

If you replying to me. I mean the Print Complete popup box.

blueeagle69, 2020-11-09:
> haven't you changed this setting?
>
>  * Options: [OFF: 0, POPUP: 1, TOAST: 2]
>  *  OFF:   No notification. The message is ignored.
>  *  POPUP: Display a popup window for user confirmation.
>  *  TOAST: A [...]

I believe so but in config.ini.
Do you need my config? I can upload later

radek8, 2020-11-09:
> > What confirmation do you mean?
> 
> If you replying to me. I mean the Print Complete popup box.

I don't have a pop-up window completed. Printing is complete when the STOP icon changes to BACK.
As seen in the oldman4U video.
Your print job will end with a pause, me for an unknown reason

radek8, 2020-11-09:
> I believe so but in config.ini.
> Do you need my config? I can upload later

If you changed this to Config.ini, try to return the default value
in config.ini
ack_notification: 1

blueeagle69, 2020-11-09:
Will do.
At the moment, when in tft mode I get a Print complete popup with an okay button. 
I close that and get the Stop button in the bottom right.

oldman4U, 2020-11-09:
Eagle. Could you send a picture with the info screen please.

blueeagle69, 2020-11-09:
@radek8 
Just an update.
In Config.ini I have ack_notification: 1

blueeagle69, 2020-11-09:
@oldman4U 
[image]

radek8, 2020-11-09:
FW TFT it can't be I have the same and it behaves normally. Couldn't it be in a Marlin configuration?
Can you attach configuration files from Marlin to compare what's the difference?

blueeagle69, 2020-11-09:
Certainly 😀
[Marlin Configs.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5513486/Marlin.Configs.zip)

blueeagle69, 2020-11-09:
Just noticed another weird thing to.
On the Info screen if I press the encoder button instead of touching the screen the printer homes to x/y 0 and the bed heats to 60 degrees and the nozzle to 200. (My PLA temps). I'll take that number for the exorcist if you have it 🤣

blueeagle69, 2020-11-09:
This is what I get after a print in TFT mode.
[image]
[image]

[19 more replies]