bigtreetech/BIGTREETECH-TouchScreenFirmware #1200: [Q] How do I install the TFT firmware on the Sidewinder X1?

Stellarspace, 2020-10-25:
**Your question or details you need. Please describe.**
How do I install the firmware on the Sidewinder X1? I compile, then I put the required files/folders on a microSD card, then turn the 3D Printer on, but all it does is say the following: Updating... 100%

**Specify the device variant related to your question.**
Sidewinder X1

oldman4U, 2020-10-25:
Which TFT and which firmware version (date)?

Stellarspace, 2020-10-25:
> Which TFT and which firmware version (date)?

TFT28, I’m trying to build from master.

oldman4U, 2020-10-25:
And you followed the instructions?

https://github.com/oldman4U/BIGTREETECH-TouchScreenFirmware/tree/Oldman4U

Stellarspace, 2020-10-25:
> And you followed the instructions?
> 
> https://github.com/oldman4U/BIGTREETECH-TouchScreenFirmware/tree/Oldman4U

Yes, could it be that the TFT28 is a clone?

oldman4U, 2020-10-25:
I believe you should ask the manufacturer. No idea.

radek8, 2020-10-25:
Attention TFT28 is in two versions. Did you use the correct version?

Stellarspace, 2020-10-25:
> Attention TFT28 is in two versions. Did you use the correct version?

It says TFT28 v1.0 in the Info section on the 3D Printer so I thought it would be v1.0 still.

digant73, 2020-10-25:
You have to use mks32 v1.4 for your tft

Stellarspace, 2020-10-25:
> You have to use mks32 v1.4 for your tft

Okay, thanks. I'll try it.

closed 2020-10-26 by Stellarspace

oldman4U, 2020-10-28:
@traffic-light 

Maybe something for your ReadMe?