bigtreetech/BIGTREETECH-TouchScreenFirmware #1201: [BUG] Using Encoder Wheel on any screen while printing changes print speed

taragor, 2020-10-25:
### Description

I just noticed that when using the encoder wheel (i.e. to change the temperature) while printing it changes the printspeed setting too. I'm not sure if it actually changes the Speed but the displayed percentage changes for sure.

### Steps to reproduce

1. Start an print from onboard USB
2. go to temperature menue
3. Change temp setting by using the encoder knob

**Expected behavior**
Temperature changes; Speed stays the same

**Actual behavior**
Temperature changes; shown speed percentage changes

### Hardware Variant
TFT35 3.0 + TFT35 E3 3.0

### TFT Firmware Version & Main Board Firmware details
Marlin: Current Bugfix 
TFT-FW: Current Master

### Additional Information
[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5434949/Configuration.zip)

traffic-light, 2020-10-25:
Take a look at the pinned issue that's called : read me before posting

taragor, 2020-10-25:
Sorry the pinned comment didn't show up in the android app. Solved

closed 2020-10-25 by taragor

traffic-light, 2020-10-25:
Ah okay. What a shame!

Thank you for closing the issue!

Have a nice day/evening/morning/night 😜

radek8, 2020-11-11:
guruathwal is working on a repair
Unfortunately, it will only work when connected via EXP1 and EXP2.
The fix will not work for connections via EXP3.
There is already a fix for SKR1.4 in the world.
With the latest FW tested and works :-)
guruathwal is king.

taragor, 2020-11-13:
Thanks for the headsup. I've got the exact same setup, will compile and test later.