bigtreetech/BIGTREETECH-TouchScreenFirmware #1206: PID TUNING INCREASE OR LOW PRINT SPEED / FAN SPEED BTT TFT35 E3 V3.0

mpcolman, 2020-10-25:
When printing, and adjust live PID tunning also affect at the print speed too, like the fan speed  make the same issue......
 BTT TFT35 E3 V3.0
UPDATED WITH THE LAS FIRMWARE

traffic-light, 2020-10-25:
Why would you print and do a PID tuning?
I don't get it. What do you PID tune?

Do you turn the knob? Then the cause of this is that it's hardwired to marlin.
At this moment there is PR that would solve this.

For more info about rotating encoder changes the print speed: **See the Readme before posting!** at the top of the page in the pinned section.

oldman4U, 2020-10-26:
@mpcolman 

Please read the little bit of existing documentation before posting an issue which is known since month. 

Please help the community and close this ticket. 

Thank you

oldman4U, 2020-10-26:
#680 was the first ticket reporting it and it is 6 months old

mpcolman, 2020-10-26:
@oldman4U I read the posts, but I understand that the problem was solved with the last firmware update.........

mpcolman, 2020-10-26:
@oldman4U Sorry but I don't find ENC_ACTIVE_SIGNAL  or CHECK_ENCODER_ACTIVE_SIGNAL in Configuration_adv.h. could be possible??? thanks!

oldman4U, 2020-10-26:
It is a PR but not in master branch. 

You are a lucky person that it was fixed the same time you found the problem.

😁

mpcolman, 2020-10-26:
Sorry I'm a little lost ...what is a PR?? , do I wait for a fix?? thanks for your help...

mpcolman, 2020-10-26:
Maybe I need to copy this into the adv.h ???

https://github.com/MarlinFirmware/Marlin/pull/19796/commits/9e812e79c2a426fc283907a4599171897fbeece4

oldman4U, 2020-10-26:
Pull Request. This means a developer made something and this will be added to master once approved. So nothing you can do at the moment except you know how to compile firmware using VSC and PIO.

mpcolman, 2020-10-26:
I'm compiling right noe added the missed parts thats describe in the link   MarlinFirmware/Marlin@9e812e7 but have only one an error in compilation ........

_In file included from Marlin\src\lcd\marlinui.cpp:34:
Marlin\src\lcd\marlinui.h: In static member function 'static bool MarlinUI::on_status_screen()':
Marlin\src\lcd\marlinui.h:553:52: error: 'currentScreen' was not declared in this scope; did you mean 'run_current_screen'?_
  553 |     static inline bool on_status_screen() { return currentScreen == status_screen; }
      |                                                    ^~~~~~~~~~~~~
      |                                                    run_current_screen
compilation terminated due to -fmax-errors=5.
*** [.pio\build\BIGTREE_SKR_PRO\src\src\lcd\marlinui.cpp.o] Error 1
In file included from Marlin\src\lcd\../inc/MarlinConfigPre.h:51,
                 from Marlin\src\lcd\../inc/MarlinConfig.h:28,
                 from Marlin\src\lcd\ultralcd.cpp:23:
Marlin\src\lcd\../inc/../../Configuration_adv.h:1567:1: warning: "/*" within comment [-Wcomment]
 1567 | /**
      |  
In file included from Marlin\src\libs\../inc/MarlinConfigPre.h:51,
                 from Marlin\src\libs\../inc/MarlinConfig.h:28,
                 from Marlin\src\libs\W25Qxx.cpp:23:
Marlin\src\libs\../inc/../../Configuration_adv.h:1567:1: warning: "/*" within comment [-Wcomment]
 1567 | /**
      |
In file included from Marlin\src\libs\../inc/MarlinConfigPre.h:51,
                 from Marlin\src\libs\../inc/MarlinConfig.h:28,
                 from Marlin\src\libs\BL24CXX.cpp:21:
Marlin\src\libs\../inc/../../Configuration_adv.h:1567:1: warning: "/*" within comment [-Wcomment]
 1567 | /**

oldman4U, 2020-10-26:
There is nothing you have to add manually, but to use this repository: https://github.com/guruathwal/BIGTREETECH-TouchScreenFirmware

Compile it and try it

radek8, 2020-10-27:
Oldman4U do not use user storage, it is better to use "GitHub Desktop"
It's convenient, fast and you can combine more PR into one compilation :-)
https://desktop.github.com/
Is there a link for eaten apple :-)

radek8, 2020-10-27:
Attention. Mpcolman does not write what kind of control board Marlin has.
  If the board is connected via cable EXP3 (not EXP1 and EXP2) fix the encoder behavior in Marlin suggested by guruathwal
it will not work.
The repair is only for connection via connector EXP1 and EXP2!

oldman4U, 2020-10-27:
If I understand this right the fix is only needed for EXP1 and EXP2 setups and not for EXP3 setups.

oldman4U, 2020-10-27:
> Oldman4U do not use user storage, it is better to use "GitHub Desktop"
> 
> It's convenient, fast and you can combine more PR into one compilation :-)
> 
> https://desktop.github.com/
> 
> Is there a link for eaten apple :-)
> 
> 

Not sure what you mean. I use GitHub Desktop.

radek8, 2020-10-27:
> If I understand this right the fix is only needed for EXP1 and EXP2 setups and not for EXP3 setups.

Yes, the fix is intended for connection via EXP1 and 2,
It will not work on EXP3 yet
In Marlin, PR is combined into SKR1,4 boards.
PR is created for other records but has not been merged yet.

radek8, 2020-10-27:
> > Oldman4U do not use user storage, it is better to use "GitHub Desktop"
> > It's convenient, fast and you can combine more PR into one compilation :-)
> > https://desktop.github.com/
> > Is there a link for eaten apple :-)
> 
> Not sure what you mean. I use GitHub Desktop.

pak OK

PR #1203 (2020-10-27): Change ENC_ACTIVE_SIGNAL logic to disable encoder in Marlin firmware when TFT is not in Marlin mode.
- Change `ENC_ACTIVE_SIGNAL` logic to disable encoder in Marlin firmware when TFT is not in Marlin mode. Details in https://github.com/MarlinFirmware/Marlin/pull/19796
- clean-up
- Add stale bot config to manage old abandoned issues. @bigtreetech Needs installing stale app for the repository from here: https://github.com/apps/stale

This will fix #680 after https://github.com/MarlinFirmware/Marlin/pull/19890 is merged in marlin firmware.
Resolves #1206 , Resolves #1045
closed 2020-10-27 by bigtreetech