bigtreetech/BIGTREETECH-TouchScreenFirmware #1208: [BUG] (short description)

Stellarspace, 2020-10-26:
### Description

When selecting the TFT microSD card to print models, the TFT goes back to the previous page. I've compiled from the master branch, from #1188 , and the latest from the releases page, then updated the TFT. 

**Expected behavior**
Files are supposed to appear and stay on the screen.

**Actual behavior**
Goes back to previous screen - the main menu.

### Hardware Variant
MKS 3.2 V1.4
### TFT Firmware Version & Main Board Firmware details
N/A; Marlin 2.0.6.1

### Additional Information
I'm using artillery_swx1_marlin_fw_2.0.6.1_mbl from https://www.thingiverse.com/thing:4294049/

traffic-light, 2020-10-26:
That's strange. I don't have this problem.
Maybe try another sd-card?

And I believe that the sd-card shouldn't be more then 8GB. Otherwise the screen can't handle it.

For usb memory this is, for so far I know, unlimited

oldman4U, 2020-10-26:
Can not see this problem either. Would also check another sd card.

Stellarspace, 2020-10-27:
> That's strange. I don't have this problem.
> Maybe try another sd-card?
> 
> And I believe that the sd-card shouldn't be more then 8GB. Otherwise the screen can't handle it.
> 
> For usb memory this is, for so far I know, unlimited

It's a 32 GB microSD. The previous firmware allows me to use it, so I don't see why a newer (and better) firmware wouldn't.
[Screen Files.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5442622/Screen.Files.zip)

oldman4U, 2020-10-27:
For a test it would be good to see if it makes a difference. 

A short movie showing the issue and a picture of the info window could eventually help.

Stellarspace, 2020-10-28:
> For a test it would be good to see if it makes a difference.
> 
> A short movie showing the issue and a picture of the info window could eventually help.

I can’t find a microSD that is 8 GB or 4 GB in stores and Amazon wants the same price for the larger size. Also, when I insert a USB Drive, the TFT says “U Disk inserted!”, but it doesn’t say anything when I insert the microSD Card. There should be no reason that it doesn’t accept a 32 GB microSD when https://github.com/makerbase-mks/MKS-TFT35-Firmware can.

Stellarspace, 2020-10-28:
[image]
> For a test it would be good to see if it makes a difference.
> 
> A short movie showing the issue and a picture of the info window could eventually help.

I can’t find a microSD that is 8 GB or 4 GB in stores and Amazon wants the same price for the larger size. Also, when I insert a USB Drive, the TFT says “U Disk inserted!”, but it doesn’t say anything when I insert the microSD Card. There should be no reason that it doesn’t accept a 32 GB microSD when

oldman4U, 2020-10-28:
Update to latest Marlin. There was a fix related to the serial number of those cards. It is obvious that the card can not be read. You have no other sd card to try?

Stellarspace, 2020-10-28:
> Update to latest Marlin. There was a fix related to the serial number of those cards. It is obvious that the card can not be read. You have no other sd card to try?

No, I don’t.

Stellarspace, 2020-10-28:
> Update to latest Marlin. There was a fix related to the serial number of those cards. It is obvious that the card can not be read. You have no other sd card to try?

I updated to the latest Marlin and it didn't fix the problem. It's the TFT firmware - not Marlin. I compiled from master branch and updated the TFT - same result.

oldman4U, 2020-10-28:
OK.

Then maybe the TFT Firmware has the same issue with the serial number of those SD Cards. Sounds like a bug to me.

radek8, 2020-10-28:
FW can be compiled for TFT display MKS 32_V1.4
According to the photo, you have MKS 32_V4.0
No pins are defined for MKS 32_V4.0 versions.
Can't this be the problem?

[image]

[image]

radek8, 2020-10-28:
Can you check that the card is in FAT32 format?

oldman4U, 2020-10-28:
@radek8 Do you know if the 32 is a special 35 and if the firmware is the same. Shouldn't Artillery taking over support for their hardware?

radek8, 2020-10-28:
In the attached video you have another MKS TFT 28V1.0 display
In the description of the problem you state you mention MKS 32 V1.4
The photo shows MKS 32 V4.0

Do you have 3 displays with the same problem or am I crazy?

[image]
[image]

radek8, 2020-10-28:
> 
> It's a 32 GB microSD. The previous firmware that I was using allows me to use it (attached below), so I don't see why a newer (and supposedly better) firmware wouldn't.
> [Screen Files.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5442622/Screen.Files.zip)
> [Sidewinder_manual_2.0.hex.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5442629/Sidewinder_manual_2.0.hex.zip)

You write that in the previous FW SD card worked. But you didn't write that the previous FW was from a display manufacturer who isn't a BTT and probably not an MKS.
Given your post # 1200, this is probably some clone that will otherwise have pins connected to the SD card.
Obtain the wiring diagram and check the connection of the pins to the SD card.

Stellarspace, 2020-10-28:
> FW can be compiled for TFT display MKS 32_V1.4
> According to the photo, you have MKS 32_V4.0
> No pins are defined for MKS 32_V4.0 versions.
> Can't this be the problem?
> 
> [image]
> 
> [image]

I didn't change anything in the source code after downloading it from the master branch except "default_envs = MKS_32_V1_4". I was told by @digant73 that I had to use MKS_32_V1_4 for the TFT inside the Sidewinder X1. As for the version mismatch, that's not important as it can be altered to read anything.

> Can you check that the card is in FAT32 format?

I can confirm that It's in the FAT32 format.

> @radek8 Shouldn't Artillery taking over support for their hardware?

Artillery doesn't provide firmware with Linear Advance and all the additional features that BIGTREETECH's firmware provides. Source Code wasn't always available for the MKS TFT firmware so a developer ported BIGTREETECH's TFT firmware to work with MKS TFT. Artillery doesn't even use an MKS TFT - it's a clone of the MKS TFT. I don't know which size or version to be exact.

radek8, 2020-10-28:
The problem will be that it is a clone of MKS that did not follow the pin numbering of the SD card.
If you have the source code for the original version of the FW in which the SD card works, it would be possible to find the pin numbers to which the SD card is connected.

radek8, 2020-10-28:
will you try this?

[MKS_32_V1_4.26.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5454606/MKS_32_V1_4.26.x.zip)

Stellarspace, 2020-10-28:
> The problem will be that it is a clone of MKS that did not follow the pin numbering of the SD card.
> If you have the source code for the original version of the FW in which the SD card works, it would be possible to find the pin numbers to which the SD card is connected.

If the pinout was wrong, how am I able to update the TFT firmware and how am I able to see the GCODE file with it's name and an image of the object (using the Cura plugin) momentarily before the TFT goes back to the Main Menu. It has to be reading the microSD to know this information.

> will you try this?
> 
> [MKS_32_V1_4.26.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5454606/MKS_32_V1_4.26.x.zip)

I tried that and I get a black screen. Unified Menu TFT28 is applicable to this TFT.

radek8, 2020-10-28:
Does TFT28 V1.0 work for you?

Stellarspace, 2020-10-28:
> Does FW for TFT28 V1.0 work for you?

No, only MKS_TFT32 works with this TFT, but you use Unified Menu TFT28 files with it.

radek8, 2020-10-28:
The FW I sent didn't work at all

Stellarspace, 2020-10-28:
> The FW I sent didn't work at all

Black screen, but I heard beeps and what not.

radek8, 2020-10-28:
But it was a version MKS_32

Stellarspace, 2020-10-28:
> But it was a version MKS_32

What branch did you build it from?

radek8, 2020-10-29:
the latest that was available

digant73, 2020-10-29:
@Stellarspace create the partition as FAT32 primary (not logical) and it will work.
Use a tool such as EasyUS Partition Master or similar to make that kind of partition. Don't use the Format tool from Windows OS

Stellarspace, 2020-10-29:
> the latest that was available

Maybe I did something wrong that time, but I already built from master and updated and it still didn’t work.

> @Stellarspace create the partition as FAT32 primary (not logical) and it will work.
> Use a tool such as EasyUS Partition Master or similar to make that kind of partition. Don't use the Format tool from Windows OS

I’m on a Mac, but I’ll do some testing to see if Disk Utility creates a primary partition or logical partition.

digant73, 2020-10-30:
@try to delete the partition and recreate it as FAT32 an primary partition

oldman4U, 2020-10-30:
I use Disk Utility and macOS 10.14.x for all my SD Cards.

oldman4U, 2020-10-30:
> What TFT are you using?

TFT35 E3 v3

Stellarspace, 2020-10-30:
> TFT35 E3 v3 Stellarspace <notifications@github.com> schrieb am Fr. 30. Okt. 2020 um 18:56:
> […](#)

I’ve done more investigating and the TFT in the Sidewinder X1 is a cloned MKS TFT28 with no internal connector for a WiFi module, no USB port and no microSD slot. A ribbon connector is soldered to the TFT board to connect a ribbon cable from the board to the USB port and microSD slot on the exterior casing of the Sidewinder X1. I build using BTT’s TFT firmware using MKS TFT32 parameters. Don’t different MKS TFT models need different parameters?

radek8, 2020-10-30:
https://github.com/makerbase-mks/MKS-TFT/issues/395
https://github.com/Blueforcer/MKSTFT_Marlin_Touch

digant73, 2020-10-30:
@Stellarspace as I said try to format the sd with another tool or recreate the partition. My 32gb sd card perfectly works even on the stock sidewinder tft with btt fw

oldman4U, 2020-10-30:
Maybe this particular card has a problem and another one works?

Stellarspace, 2020-10-31:
> [makerbase-mks/MKS-TFT#395](https://github.com/makerbase-mks/MKS-TFT/issues/395)
> https://github.com/Blueforcer/MKSTFT_Marlin_Touch

The issue remains the same with that firmware as well.

> @Stellarspace as I said try to format the sd with another tool or recreate the partition. My 32gb sd card perfectly works even on the stock sidewinder tft with btt fw

Disk Utility on macOS = Didn't Work
Quick Formatter on Windows = Didn't Work
Disk Management on Windows = Didn't Work
EaseUS on Windows =  Didn't Work

> […](#)

I agree, but it's pretty ridiculous. I just bought the microSD Card.

digant73, 2020-10-31:
@Stellarspace you could also try the sd in the stock fw just to verify if it's still working on that

Stellarspace, 2020-10-31:
> @Stellarspace you could also try the sd in the stock fw just to verify if it's still working on that

I already updated the TFT to stock firmware and the TFT doesn’t go back to the Main Menu like the BTT firmware. I don’t know if there have been any hardware changes, but I’m using a V3 Sidewinder X1 and not a V4.

oldman4U, 2020-11-02:
Maybe you can ask Artillery which firmware is compatible with your printer. We do not own this hardware so we can not test it.

Stellarspace, 2020-11-03:
> Maybe you can ask Artillery which firmware is compatible with your printer. We do not own this hardware so we can not test it.

They'll just direct me to MKS firmware, which doesn't have the Unified UI or any of the extra features.

oldman4U, 2020-11-05:
What do you mean with debug without updating?

All of us are not working for BTT, but we are users like you trying to help. And even we all have BTT hardware, we likely have completely different use cases.

Stellarspace, 2020-11-06:
> What do you mean with debug without updating?
> 
> All of us are not working for BTT, but we are users like you trying to help. And even we all have BTT hardware, we likely have completely different use cases.

There's no way to debug on the computer is what I'm saying. Is that a correct assumption?

oldman4U, 2020-11-07:
I am still not sure what you mean. Sorry.

oldman4U, 2020-11-27:
Have you been able to fix it. Another user reported this:

I needed to format the SD Card (not fast formating fat32 and standard-blocksize and set to 250000 Baud, than i coulded Connect to the sd card onboard from tft.

closed 2021-02-02 by stale[bot]