bigtreetech/BIGTREETECH-TouchScreenFirmware #1210: M600 code is not read by marlin mode [BUG] (short description)

John-Smith980, 2020-10-26:
While printing in Marlin mode, the m600 code is completely unread on the TFT 35 e3 V3 LCD.

blueeagle69, 2020-10-29:
Hi.
M600 works perfectly on my setup in Marlin Mode and TFT.
Print from the printer's ONBOARD SD see if it works.

oldman4U, 2020-11-05:
Hello John-Smith980.

blueeagle69 tries to help you, so it would be great if you could send him a shot answer.

Thank you

Please close the ticket once you do not need it anymore.

oldman4U, 2020-11-09:
980 seems to be a short version of nine day wonder....

@blueeagle69 Let the Bot do all the work....

blueeagle69, 2020-11-09:
> 980 seems to be a short version of nine day wonder....
> 
> @blueeagle69 Let the Bot do all the work....

Yep 😏

wildjokerss, 2020-12-01:
Hi, was just wandering around and came across this.
I'm not sure about M600 with onboard SD, but while printing via octoprint, the M600 command does not work within Touch mode, it needs to be in Marlin mode.
(and neither does print progress nor anything else but that's another story). 
Basically if the outsprint sends the M600 commands there are no "click to stop purge" "click to load new filament" nor so messages in touch mode like we have in Marlin mode. Am I missing anything?

sarvenn, 2021-01-12:
> Hi, was just wandering around and came across this.
> I'm not sure about M600 with onboard SD, but while printing via octoprint, the M600 command does not work within Touch mode, it needs to be in Marlin mode.
> (and neither does print progress nor anything else but that's another story).
> Basically if the octoprint sends the M600 commands there are no "click to stop purge" "click to load new filament" nor so messages in touch mode like we have in Marlin mode. Am I missing anything?
> 
> edit: typo

Today I encountered the same issue. I change my config.ini file with a custom g-code which is basically M600.
I started that custom gcode in TFT mode. Actually everything started fine. I parked to nozzle upper left corner, heated the nozzle to 170C. Unloaded the filament, then a pop up occurred and only choice was OK, I clicked on OK and loaded filament. It purged filament. Everything was fine up to that time. After purging stopped, a pop up appeared and only choice was OK. When I click on OK, pop up appears again with an endless loop. Only way to get rid of it is to shut down the printer.
That pop up needs to ask for purge more and finish options. Could it be done @oldman4U ? I can take a photo of it if needed.

oldman4U, 2021-01-12:
Serial always on

Is active in config.ini?

wildjokerss, 2021-01-12:
yes, it is

sarvenn, 2021-01-12:
Mine was inactive I suppose. If default config file setup is inactive then mine is inactive as well. I didn't change it. What does it for? Should I change it to enabled and re-try?

oldman4U, 2021-01-12:
I am sorry, seems I mixed up a few things reading your comments on my iPhone.

The serial always on is only relevant when switching to Marlin mode on dual mode TFT's. Using the touchscreen, serial is on by default. 

Because I have no Octoprint, I can not help with this, sorry.

@blueeagle69 Are you still available to help? Seems @sarvenn uses a custom gcode. Are you using the same or are you using M600 as part of the print file gcode?

blueeagle69, 2021-01-12:
Hi.

When I tested it, I was using the m600 that was in the gcode generated in Prusaslicer. Barring things stopping me I will test again tomorrow from Marlin mode and TFT.

oldman4U, 2021-01-12:
Hi, good to see you.

It would be great if you could help. I have no such device, so it is really hard to help;-(

THANK YOU

blueeagle69, 2021-01-13:
@oldman4u.

Sorry I'm a little late reporting back.
I sliced a test file which contained the M600 Filament change.
Printing from my customary onboard SD went as follows.

In Marlin mode the printer successfully read the M600, parked the nozzle and removed the filament waiting for new filament to be inserted. After this the printer resumed as normal.
I then repeated the same print from within TFT mode.
When the printer read the M600 the printer parked the nozzle, The  Resume button appeared on screen.
After hitting Resume the printer unloaded the filament and showed "Nozzle Parked" on screen with the button below it. Upon clicking that the printer fed in new filament and displayed the "Purge More" or "Continue".
I hit continue and the printer resumed as normal.
So I couldn't see any issues with M600 in either screen mode.
The M600 was contained in Gcode generated by Prusaslicer.

Hope this helps someone.

sarvenn, 2021-01-14:
Yesterday I updated my TFT35 E3's firmware and tried M600 from the custom command in my custom list.
My gcode is just simply : M600\n
Same problem happened. I send M600 from TFT terminal. Nozzle heated up to 170C. Nozzle park to upper left corner after homing as expected. Filament is unloaded. A pop up messege appeared just as info. After unloaded finished I loaded filament. It purged and another pop up appeared. only choice was OK again. if I clicked to OK, same pop up appear endlessly. At that time I switched to marlin mode simply just long touching to a point only screen (just like press and wait the encoder rotary knob). I switched to marlin and expected continue and purge more screen appeared. So in marlin mode everything is OK. Not in TFT mode. In my config ini file serial always on option is at off state as it is off in default config file in repository.

Here are my screenshot. File names are in chronological order.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

blueeagle69, 2021-01-15:
Weird.

When the M600 is generated in the slicer (Prusaslicer) it worked perfectly in either mode.

sarvenn, 2021-01-15:
@blueeagle69 I tried a color change layer with M600 command with PrusaSlicer and it behaved even worse.
It messed everyting. No pop up appeared. (touchscreen SD card)
My serial always on is disabled as default in my config file as well. I can not understand where this difference comes from between us. Could you try to send a M600 command via onboard terminal and take some pictures for me?

[image]

blueeagle69, 2021-01-15:
It looks like the issue is where the file is printed from.

I only use Onboard SD(SKR1.4) and it appears to work fine from there.

oldman4U, 2021-01-15:
@blueeagle69 could you please share the sliced gcode so we can reduce parameters.

blueeagle69, 2021-01-15:
Hi.

All it was, was an added primitive shape in Prusaslicer with a filament change added.

But if you need it I can send it later.

I'm sure once again the problem is using the tft sdcard. 
I will confirm later.

blueeagle69, 2021-01-15:
Hi.
It's as I expected.
Printing from the onboard SD, m600 works as expected.
When printing from the TFT the printer just stops printing. Hitting the Resume button resumes the print but no filament change takes place.

oldman4U, 2021-01-15:
So it works in Marlin and Touchscreen mode, as long as you print from on board sd, but does not work when printing from TFT related storage, right?

wildjokerss, 2021-01-15:
unsure if I ever tried from the TFT SD. I know that octoprint + TFT mode it breaks things. octoprint + marlin mode works flawlessly.

blueeagle69, 2021-01-15:
> So it works in Marlin and Touchscreen mode, as long as you print from on board sd, but does not work when printing from TFT related storage, right?

From my four or five tests, that is what it is looking like.
I just repeated the test from the TFT SD and sure enough when the screen gets to the M600 command it just stops in place. Hitting resume continues it.

oldman4U, 2021-01-15:
But eagle, your M600 comes from the gcode file and at least the joker talks about external commands, right?

blueeagle69, 2021-01-15:
> But eagle, your M600 comes from the gcode file and at least the joker talks about external commands, right?

I think he was using Octoprint and possibly an m600 invoked from the terminal.
As you said my M600 was gcode generated.
I will try that now printing from onboard SD and report back.

wildjokerss, 2021-01-15:
Sorry if i wasn't clear.
I use octoprint to send stuff to the printer. What I'm saying is that at least the M600 sent through octoprint does not work properly IF the screen is in TFT mode, if the screen is in Marlin mode it works just fine, from octoprint.
I am unsure about SD Card either on the TFT or SKR itself since they are on a "not so easy" to reach place and i just prefer the simplicity of octoprint.

blueeagle69, 2021-01-15:
I can't comment on Octoprint unfortunately, but I just did another test print that contained _no_ M600 in the gcode.
I printed from the onboard SD and sent an M600 command via Pronterface terminal.
This also worked perfectly,
It looks like that unless the gcode is on the onboard SD M600 is working correctly.
Also can someone check their TFT config.
There use to be a setting pertaining to M600/M601 but in mine I can't find it.

oldman4U, 2021-01-15:
This feature is enabled by default in Configuration.h line 300.

wildjokerss, 2021-01-15:
i might take another shot one one day. see if it runs as it should. also have to see if any update came to the TFT in the meantime

blueeagle69, 2021-01-15:
Yah.
I don't think it matters where the m600 is invoked from as long as the gcode is on the onboard SD.
Great for me. Not so good for others.

sarvenn, 2021-01-15:
Weird things had just happened.

First of all, I printed the same file (prusaslicer color change at layer with M600) from on board sd card.
This time instead of purging on the part, it parked the nozzle at it's proper place as expected and unloaded the filament. Some beep sounds heard and that's all. No response, no pop up page to continue with loading sequence. Print time counter was still counting and pause button stayed as it is. So I stopped the print.

Afterwards I tried to do a unload/load from the relevant menu which was improved recently. I clicked on the load/unload menu, nozzle is heated up to 170 automatically, unloaded the filament, a pop up appeared (the one I shared above pics-paused for user  continue) with only OK option. I clicked on OK. Loaded the filament. Again same pop up appeared, clicked ok, same pop up appeared...Endless loop again. I had to shutdown the printer.
As I noticed prebuild firmware has updated 6 hours ago. I will install that firmware today and try this again along with some other bugs I found and let you know

These results had confused me furher more :) It is obvious that there is a significant difference from printing from TFT SD and onboard SD. As I told before I also tried to send a M600 command from TFT terminal. Results were the same with printing from onboard SD.

blueeagle69, 2021-01-15:
Agreed.
It's a strange one. 
For me at least when printing from the onboard SD, the filament change hasn't failed once  on my Biqu B1.

closed 2021-03-23 by stale[bot]