bigtreetech/BIGTREETECH-TouchScreenFirmware #1211: [FR] Move X, Y, Z Axes During Pause Printing

Stellarspace, 2020-10-28:
**Is your feature request related to a problem? Please describe.**
If the end-user would like to switch to a filament with a different color and/or material while printing is paused, the filament could end up extruding on the incomplete printed object. This also applies to when the end-user runs out of filament.

**Describe the solution you'd like**
The end-user should have access to the Move menu to move the Extruder and/or Bed, but the Home button should be unavailable as that would be destructive. Confirmation as POPUP might be required as moving the Extruder and/or Bed while the print is on pause could be destructive in certain situations.

**Describe alternatives you've considered**
Maybe when the end-user wants to change filament (while on Pause) via the Extrude menu, the Extruder and/or Bed would automatically move to a position on one of the sides (after pressing the Unload button). The end-user could extrude as much filament as they wanted in that position with buttons from the Extrude menu, then when Resume is pressed, the Extruder and/or Bed would return to the original position. We'd have to test when to show confirmations and when not to show confirmations.

**Additional context**
A user by the name of [caroloking](https://www.thingiverse.com/caroloking/designs) on Thingiverse suggested this feature.

Stellarspace, 2020-10-28:
I didn't realize this is M600.

bigtreetech, 2020-11-02:
Hello, It sounds like adding babystep to the XY axis?

radek8, 2020-11-02:
I think that in Marlin it can be set, when printing is paused (for example to change the thread), the head is parked in the specified place. When the pause is canceled, the print returns to its original position.

Stellarspace, 2020-11-03:
> Hello, It sounds like adding babystep to the XY axis?

You could say that.

> I think that in Marlin it can be set, when printing is paused (for example to change the thread), the head is parked in the specified place. When the pause is canceled, the print returns to its original position.

When printing is paused, it already parks in a specific place, but for some reason, caroloking wanted access to the Move menu while paused as well. There is also M600, which has to be manually typed into the Terminal menu.

oldman4U, 2020-11-04:
Personally there are FR's where I can not see why I would need it and this is one of theses requests. Anyway, it is for sure more important then the ability to execute PID and Steps tuning WHILE printing, so I added this FR to the FR list.

If it is OK for you, it would be great if you could close this ticket.

Thank you

closed 2020-11-04 by Stellarspace

oldman4U, 2020-11-04:
👍🏻

digant73, 2021-07-27:
@oldman4u the parking area is configurable on both marlin and TFT now. Once paused by TFT the user can move to the extruder/(un)load menus etc._. so this FR is already implemented and could be removed from the list. Furthermore this FR was not even requested by the user that opened the FR that simply reported the thougth of another user that cannot even clarify his/her use caseses