bigtreetech/BIGTREETECH-TouchScreenFirmware #1212: Classic Menu & Icon Theme - EOL

oldman4U, 2020-10-28:
Hi.

The Classic Menu & Icon Theme is currently not available for all TFT's bigger than TFT35 and even for the supported TFT's some Icons are already missing.

Is there someone out there, who would like to add the missing icons?

If not, I would remove it from the package and the ReadMe with my next PR.

Thank you

Chronicmemecrafter, 2020-10-28:
No one uses the classic menu so I would just get rid of it or make some sort of legacy section for people who do.

oldman4U, 2020-10-29:
Thank you for your feedback.

I like the theme you maintain. Maybe you should post it a bit more prominent!

No sure there is a legacy area somewhere, but we can put it into a ticket, so it will be always available. Even it is useless;-)

Chronicmemecrafter, 2020-10-29:
Glad you like it but just a reminder it is not mine. I am also currently
working on making it compatible with the latest version.

oldman4U, 2020-10-29:
I understood. But you take care!

Thank you

oldman4U, 2020-10-30:
Finally if someone knows about this theme and really wants it, he can also download an older version. So I will remove it - that's it.

Seems there are not many fans out there using it...;-)

Thank you again for your input

Happy printing

closed 2020-10-30 by oldman4U