bigtreetech/BIGTREETECH-TouchScreenFirmware #1213: Latest Master is not working with the SKR TFT35 V3.0

NicoletoAerials, 2020-10-28:
### Description

When compiling the latest Master and installing the firmware into a SKR TFT35 E3 V3 the screen freezes after restart.

[BIGTREE_TFT35_V3.0_E3.26.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5454258/BIGTREE_TFT35_V3.0_E3.26.x.zip)

### Steps to reproduce

1. In platformio.ini select default_envs = BIGTREE_TFT35_E3_V3_0 as compiling environment.
2. Compile the latest Master via VS Code
3. Install firmware
4. Screen gets stuck in the main window

**Expected behavior**
Screen working normally

**Actual behavior**
Screen gets stuck in the main window

### Hardware Variant

BTT SKR mini E3 V2.0 with latest Marlin Bugfix 2.0.x
BTT SKR TFT35 E3 V3 with the latest Master

### Additional Information
The problem started after merging  #1188 #1144  and #1134

digant73, 2020-10-28:
Did you provide the new config.ini and icons?

NicoletoAerials, 2020-10-28:
You are right. I missed the icons folder. Everything is working fine now.

Thanks.

closed 2020-10-28 by NicoletoAerials