bigtreetech/BIGTREETECH-TouchScreenFirmware #1217: Invalid Value Temperature at 475 degrees for Mosquito Hotend

Matdos, 2020-10-29:
I have installed a Mosquito Hotend and want to set the maximum temperature of 450 degrees, but the error message Invalid Value appears. 

How can I set this?
[image]

oldman4U, 2020-10-29:
Hi.

It is always helpful to attache the config files here.

Matdos, 2020-10-30:
I dont now what is the problem.

[config.docx](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5467735/config.docx)

GregSKR, 2020-11-01:
Currently, temperatures over 300°C are not allowed:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/8a5b9a1ccb0f3c2d3f6b2d14bfdd2e015a6ea0c8/TFT/src/User/API/config.h#L154

If you compiled the firmware youself, change the value to **500** (max temp for mosquito hotends).

You should create a PR for this... :wink:

Matdos, 2020-11-01:
hello, thanks a lot, I didn’t find it, I habe search for more than 4 hours.

closed 2020-11-01 by Matdos