bigtreetech/BIGTREETECH-TouchScreenFirmware #1218: [BUG] Verified Bugs

oldman4U, 2020-10-30:
Dear reader,

This list of verified bugs has three "functions":

1: It allows users to check, if a certain bug has already been requested, before starting a new ticket.

2: To mark bugs which are already under development to avoid, that several developers are working on the same function.

3: There is now a BOT which closes inactive tickets. Some of them are probably useful, so it makes sense to make them available at a more prominent place.

Notes:
Make sure you are using the latest firmware available for your mainboard and TFT, before adding a new bug report.
Read the [Read me before posting!](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1047) to see if you found an already know issue.
To join an existing bug report, please click on the link below. Even the ticket is closed, it does not mean that it can not be used anymore.
Feel free to add your comments related to this BUG ticket below.
New bug reports will be added as fast as possible, once more than one user verified the same issue.
Bug reports shown in Bold are under development and if possible the developer is mentioned next to it.
Notes from the community are added to each BUG in CAPITAL LETTERS.
Bugs which have been merged into the master branch, will be removed from this list.

General Bugs:
#417 TFT35 stops responding with red 'Busy processing please wait....'
#540 and #593 TFT not reading filament detector when connected to SKR
#571 Firmware assumes heated bed - SOUNDS MORE LIKE A FEATURE REQUEST
#893 (MARLIN 2.5.4 - M997 freeze SKR1.4 Turbo with BTT TFT35-E3 V3.0 only in touch mode)
#1199 Terminal outputs "Busy paused for user" after every print

Touchscreen Related Bug:
#629 Filament motion sensor TFT display is not interested in... 
#642 Communication timeout - reset send buffer block
#1135 On-screen Layer info not working printing from onboard SD
#1139 in print fan % change issue
#1193 TFT 35 E3 v3.0 + skr 1.4 Power loss recovery Problem - NO support for UPS module

Marlin Screen Related Bugs:

Bug, Feature or Known Issue?:

PS: A user asked me what happens with this list in case I do not want to maintain it anymore or when I die. Well, I am younger than my nickname makes you believe, but this is a valid question, so here is my answer.

Besides the option to reopen all tickets or to fall back to the status before this list has been made, there is still the option to copy and paste this ticket into a new one at any time. Or I am able to hand it over to another user, so I would do so...

Thank you

@bigtreetech Maybe you can pin this list also. Thank you

oldman4U, 2020-11-16:
@bigtreetech @Msq001 In case you decide to pin this ticket one day, could you please make this ticket the most left, followed by the FR ticket and the Read Me Before Posting the one most to the right. This would place it right above the button to start a "New issue" which would make it more likely that people read it.

Thank you

oldman4U, 2020-11-27:
Close Issue now that it is pinned.

Thank you BTT

closed 2020-11-27 by oldman4U

chris-Neale, 2020-12-09:
I am having the same issue with the x axis grinding when enabling the z offset. I’m using the z min limit switch for homing with a negative home offset. It homes G28, applies the z offset, moves up and then tries to move the x axis past the limit switch.

kisslorand, 2020-12-21:
> #540 and #593 TFT not reading filament detector when connected to SKR

Send M75 to Marlin when print begins. I had the same issue, Marlin doesn't know that the print is started so no filament runout warnings. It's not a bug.

oldman4U, 2020-12-21:
Thank you kisslorand, I informed the two other owners of the related tickets.

chris-Neale, Please try what kisslorand recommended above and let us know.

Thank you