bigtreetech/BIGTREETECH-TouchScreenFirmware #1224: tft24-v1.1 pressing touchscreen doesnt work since firmware upgrade

Roop66, 2020-11-01:
i have updated the firmware with various different files but it looks like its broken - it powers on - lets me do a long press to take me back to marlin/btt menu. touching/pressing anywhere on the screen doesn't work. spent a long time on this quite fed up. any help would be great. cheers.

oldman4U, 2020-11-01:
Read the Read Me to reset the display.

Roop66, 2020-11-01:
Thanks. I'll have a look.

Roop66, 2020-11-02:
Thank you, this worked - partially -  I get a different message (may not be attributed to original post) when switching to marlin mode. i get LCD12864 simulator stuck on screen. i think there may be a setting in marlin 2.0 but am unsure which.
Thanks

oldman4U, 2020-11-02:
There is a read me before posting ticket pinned on top of the main ticket page.

Roop66, 2020-11-03:
OK thank you ill see if i can find the main ticket page.

Roop66, 2020-11-03:
if you mean this one: 
https://docs.github.com/en/free-pro-team@latest/github/working-with-github-support/submitting-a-ticket

the only step i believe i have missed is: 'Steps to reproduce the issue'

so here goes:

power on SKR1.3, TFT24 boots, press jog dial to select Marlin mode, then i get:

LCD12864

problem is reproducible every time.

Hope this is what you were referring to.

Thanks in advance.

Roop66, 2020-11-03:
ok i think i was booking at eh wrong readme.

i have found the one you were referring to:

REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

i shall try this and let you know.

Thanks

oldman4U, 2020-11-03:
And that you have to turn the encoder and not to forget the right cabeling which depends on your mainboard.

oldman4U, 2020-11-04:
Make sure that the CR10 Stockdisplay has // in front of it.

Please let us know

Thank you

oldman4U, 2020-11-05:
Hi.

Any news?

radek8, 2020-11-05:
Roop66 What's your board with Marlin?

oldman4U, 2020-11-06:
Roop66

We try to help you, but to be able to do so, we need you to answer our questions. Thank you.

Please help the community and close this ticket in case you do not need it anymore.

Roop66, 2020-11-06:
> Roop66 What's your board with Marlin?

 SKR1.3 thanks

Roop66, 2020-11-06:
> Make sure that the CR10 Stockdisplay has // in front of it.
> 
> Please let us know
> 
> Thank you

no, i couldnt see where your code should go.  and my board just exploded oops

radek8, 2020-11-06:
What???

Roop66, 2020-11-06:
Driver module caught fire. - nothing to do with the code but just wanted to make people aware of how potentially dangerous thes hobby boards can be  

https://drive.google.com/file/d/15WyXXa7g7k3M7XjcdxVCkMw2CnQV9SRM/view?usp=sharing

oldman4U, 2020-11-06:
I got a TFT24 v1.1 today and the firmware is from April 11th at least 2020;-) Will do the update now.

Your mainboard exploded.... that's bad.

Do you still want support for the TFT? This also works even without mainboard if you want.

Roop66, 2020-11-06:
Hi, not the main board - it turned out it was the stepper driver - A4988. was a bit of a shock.  I might get an alternative driver (I wonder if the a4988 are prone to spontaneous combustion ?) Anyway I'll plod on (although I'm starting to wonder why i didn't just get a replacement OEM board and save all the hassle) I shall see if I can find out where to put the code you sent earlier (CR10 Stockdisplay).. Thanks for your help.

Roop66, 2020-11-06:
I get the below:

very strange was working before.

not sure why this file should have vanished

Executing task: c:\Users\roop\Documents\Marlin-bugfix-2.0.x\platformio.exe run --target upload <

The terminal process failed to launch: Path to shell executable "c:\Users\roop\Documents\Marlin-bugfix-2.0.x\platformio.exe" does not exist.

Roop66, 2020-11-06:
Hi could you mark this as closed please. I have spent many hours trying to get the main board - skr1.3 and the TFT24 working. I am only about 5% of the way through to completing my Geeetech A10 rebuild. With such a steep learning curve I just don't have the spare time to continue (although I'm glad I ordered a fire extinguisher when I first got into 3d printing ;-) ) . It would make more sense to either buy a replacement OEM board or buy a professional printer with warranty. I should be able to get refund for the SKR and the TFT24 boards through amazon, so all is not lost. Once again thanks for your help.

radek8, 2020-11-06:
Why are you giving up so soon? It is fun ...
A4988 driver drivers are the cheapest and the worst.
I recommend using TMC 2209 instead.
Did you set the motor current on the potentiometer? That could be the cause of the fire.

oldman4U, 2020-11-06:
Or 2208.  But sensorless homing is great.

radek8, 2020-11-06:
What oldman4U wrote is to configure Marlin to make Marlin mod work for you.
Unless you compile Marlin yourself, you will never know what configuration you are using.

radek8, 2020-11-06:
I tested the sensorless guidance, but returned to the microswitch. It works fine, but every time I updated Marlin, I had to re-tune the sensitivity. There was always something tuning in there and I didn't enjoy calibrating it anymore. If that doesn't change, maybe I'll come back to it.
I can compare TMC2208 and TMC2209 I have both at home.
TMC 2209 are better in that they heat up much less. At the same current and 2x larger cooler, the TMC2208 was much warmer than the TMC2209.
TMC2209 is the best choice for me.

oldman4U, 2020-11-06:
I agree. And both are incomparable more quiet compared to A4988.

oldman4U, 2020-11-07:
@Roop66 

You are the only person which can close this ticket by clicking on Close at the bottom of the ticket.

Please help the community and close the ticket once you do not need it anymore. Thank you

Roop66, 2020-11-08:
new GEEETECH motherboard in the post ;-)

closed 2020-11-08 by Roop66

oldman4U, 2020-11-08:
👍🏻