bigtreetech/BIGTREETECH-TouchScreenFirmware #1225: [BUG] Installing pre-configured firmware give errors/warnings

ralfiii, 2020-11-01:
### Description

When I update my TFT35 display with the prepared firmware from the folder "Copy to SD Card root directory to update - Unified Menu Material theme" I get errors
* S_load.bmp and one other bitmap is missing.
* during "updating configuration" I get "Invalid keyword: mesh_max_color:2

Also strange: Is it normal that the GUI doesn't show bed and extruder temperature when heating up for a print job? My display stays as in the screenshot until the print starts, only then the current and expected temperatures are shown.

### Steps to reproduce

I downloaded the zip from github, copied
* TFT35 (folder)
* BIGTREE_TFT35_V3.0_E3.26.x.bin
* config.ini
on an SD card and inserted the card into the display to update.

### Hardware Variant

It's this display:
https://www.amazon.de/dp/B089SPPGKJ/?coliid=I3A06Y8U7J9O7E&colid=1Y5TQLAWV23RQ&psc=1&ref_=lv_ov_lig_dp_it

### TFT Firmware Version & Main Board Firmware details
* Marlin 2.0.7.2
* TFT_E3_V3.0.26.x Sept 12 2020

### Additional Information

[image]
[image]
[image]
[image]
[image]

radek8, 2020-11-01:
The problem is in the old version of the precompiled file and the new ini file and and icons.
I will compile my own bin file, if you can't, I'll send you new versions for your display.

radek8, 2020-11-01:
[BIGTREE_TFT35_V3.0.26.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5471930/BIGTREE_TFT35_V3.0.26.x.zip)

radek8, 2020-11-01:
The temperature update problem you describe is standard behavior when printing from an SD card integrated on the Marlin board.
If you print from the SD card on the TFT display, the temperature is updated already during heating.

oldman4U, 2020-11-02:
@radek8 The temp updates with the new firmware also when printing from on board sd card.

radek8, 2020-11-02:
> @radek8 The temp updates with the new firmware also when printing from on board sd card.
> 
> Using Marlin 2.0.7.2.
> 
> Do you know who is responsible for the pre compiled versions?

For me, printing from SD to SKR behaves the same as described.
If you behave differently, it could be due to some choice in the Marlin or TFT configuration.

oldman4U, 2020-11-02:
> 
> For me, printing from SD to SKR behaves the same as described.
> 
> If you behave differently, it could be due to some choice in the Marlin or TFT configuration.

Using pre compiled firmware or in general?

Do you know who is responsible for the pre compiled firmware?

radek8, 2020-11-02:
I compile FW for TFT and SKR myself.

radek8, 2020-11-02:
But we are not solving me but Ralfiii :-)

oldman4U, 2020-11-02:
Hi ralfii.

Not sure if this is already clear to you, but you made everything right. The problem is, that the precompiled firmware is very old and because of this, some elements are missing or wrong in the other parts of the software wich is needed to update your TFT.

Have you ever used VSC and PIO to compile your own firmware? Would you like to do so and would you need any help?

Please let us know

oldman4U, 2020-11-02:
@radek8 And maybe I will get the virus.....amazon.**DE**

;-)

radek8, 2020-11-02:
FW including not yet merged PR 1127 and 1126 :-)

[BIGTREE_TFT35_V3.0_E3.26.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5478258/BIGTREE_TFT35_V3.0_E3.26.x.zip)

oldman4U, 2020-11-02:
BTW. Congrats. Ripper08...

radek8, 2020-11-02:
> BTW. Congrats. Ripper08...

I don't understand some people. He buys a display from China for a few € and waits for a technician to come to install it ....

ralfiii, 2020-11-03:
1000 thanks to everyone!!! What a great, helpful community.
A few hours ago my wife gave birth to a baby,  so I am busy with other
things, but asap I'll try the provided patch...
Thanks again, Ralf

oldman4U, 2020-11-03:
Congratulations and all the best for your family!!

ralfiii, 2020-11-04:
Baby is sleeping, so I found a few minutes time:

After updating the firmware with BIGTREE_TFT35_V3.0_E3.26.x.zip provided by Radek8 yeserday, I get an error
"Failed to enable bed leveling".
And the bed leveling zero height using touch is a big higher than using Marlin.
Shouldn't that be identical?

And another question: 
Is the RESULT different, printing through touch or through Marlin? Or is it identical?

oldman4U, 2020-11-04:
Hi. They sleep a lot at the beginning. Try to get some extra sleep also. You will need it the coming 18y. 😁

So update worked?

You should see identical quality prints independent of the user interface you use.

With leveling I can not help. Start a new ticket for each new topic and close unneeded tickets.

radek8, 2020-11-04:
Ripper is the nickname of an arrogant user from another discussion :-)

radek8, 2020-11-04:
Do you use any kind of leveling?
If so, what is your probe?
Do you have leveling points saved?

ralfiii, 2020-11-05:
Hey Radek8!
I don't have any probe (yet, bltouch will follow).
And I didn't know I could save leveling points.
I clicked around in some menus, a "G-code unknown" error occurred. now the
leveling suddenly is on the right height.
The error message still shows when powering up in touchscreen mode.
Maybe for the moment I'll go with Marlin...

Thanks for your effort!
Ralf

radek8, 2020-11-06:
An unknown code is fine if you run an option that your Marlin does not support or is not configured.
It would be good to compile Marlin yourself so that you can adapt Marlin to your configuration as much as possible.

radek8, 2020-11-10:
> The temperature update problem you describe is standard behavior when printing from an SD card integrated on the Marlin board.
> If you print from the SD card on the TFT display, the temperature is updated already during heating.

Allow to start code and add to start code
M155 S1
this could solve the problem with reporting temperatures when printing from SD to SKR

radek8, 2020-11-11:
> 
> Also strange: Is it normal that the GUI doesn't show bed and extruder temperature when heating up for a print job? My display stays as in the screenshot until the print starts, only then the current and expected temperatures are shown.
> 

I have an idea how to solve the problem with updating temperatures when starting printing from the integrated SD card in Marlin.
I tested it yesterday and it works.
I will create a PR. If Bigtreetech approves it, it would be a permanent solution for everyone who suffers from this problem.

oldman4U, 2020-11-11:
This would be great. Do you know why it works on my machine? And whats
about the layer hight which is shown nowhere!?

radek8, 2020-11-11:
> This would be great. Do you know why it works on my machine? And whats about the layer hight which is shown nowhere!? Radek <notifications@github.com> schrieb am Mi. 11. Nov. 2020 um 12:02:

Why do you see temperatures when you warm up the bed and the nozzle when printing from the integrated SD? That would be good to know. I assume your slicer does it. Can you attach a file to a printable work?
Thank you
 
The problem with updating layer height and task completion percentage is a bit bigger. Printing is controlled by Marlin and TFT has no way of knowing which layer to print and what percentage is complete. This would require Marlin to proactively send the press status over the serial line. But at the moment he probably can't do that

oldman4U, 2020-11-11:
Please check out PR #1244 for a possible solution.

Thank you

oldman4U, 2020-11-11:
[CE3_KeyCap_Opener.gcode.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5526861/CE3_KeyCap_Opener.gcode.zip)

Please find attached the code file which I printed while I made the movie which shows the temp changing while heating: Temp Changing print from SD card.mov.mov.zip #1070

radek8, 2020-11-11:
It's strange.
Same parameters as for me, but different behavior.

oldman4U, 2020-11-12:
What is different?

closed 2020-11-18 by bigtreetech

FoxtrottPL, 2020-11-22:
About missing bmp files - I solved it by merge all bmp files from unified, normal and hybrid tft folders.

PR #1244 (2020-11-18): Temperature update when printing from onboard SD
After printing from the onboard SD, the bed and nozzle temperatures are not updated until they reach the target temperature.
This adjustment solves it.
Adding code "M155 S1" to start G-code does not work.
Adding a code to the start code in Cura also does not work.
First the heating is started and only then the start code is activated.
Therefore, the values on the display are not updated until the target temperature is reached.

resolves #1127
resolves #1225
resolves #1070

Add a comment to Config.ini, with information about the required settings [... cut]