bigtreetech/BIGTREETECH-TouchScreenFirmware #1229: [BUG] TFT35_V2.0 nozzle just stops during Pause and Stop command during Printing

CaptHigh, 2020-11-04:
### Description
When i Pause or Stop during printing from SDcard the nozzle just stops instead of returning to the park postion set at X10,Y10.
Due to this reason i am unable to pause to change filament, and also this causes nozzle to create burn marks on the print.
I have updated the lcd firmware (compiled the latest bin file), still this problem persists

### Steps to reproduce

1. Print
2. Print from Onboard SDcard
3. Pause/Stop
4. Nozzle just stays there

**Expected behavior**
when i pause or stop i expect nozzle to move to the parked location prgrammed in the firware of SKR OR TFT35 (Both have same parking location programmed X10,Y10)

**Actual behavior**
The nozzle just stays there burning my print until i cancel and Home the print

### Hardware Variant
TFT35_V2.0

oldman4U, 2020-11-04:
Have you tried if printing from TFT SD or U disk makes a difference?

Please attach the config.ini which you uploaded to the TFT.

CaptHigh, 2020-11-04:
Yes, I have tried both from onboard SDcard and from TFT sdcard but the problem persists. 
Here is the config.ini
[config.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5490388/config.zip)

oldman4U, 2020-11-05:
I would recommend to activate Pause and End code in config.ini and see if this solves your problem. At the moment they are deactivated.

radek8, 2020-11-07:
After pressing the pause icon, printing stops, the nozzle is raised and parked at the edge.
If this does not happen to you, attach the Marlin configuration.h and configuration_adv.c files

oldman4U, 2020-11-07:
The function is deactivated in config.ini. Not sure how this can work without?!

oldman4U, 2020-11-08:
@CaptHigh 

Any news?

oldman4U, 2020-11-09:
Please help the community and close this ticket or reply.

Thank you

CaptHigh, 2020-11-20:
The problem got soved. Thank you for your help @oldman4U

oldman4U, 2020-11-20:
👍🏻

CaptHigh, 2020-11-20:
Sorry for reopening the ticket but now this problem has started again i have tried enabling start_gcode@line427 and end_gcode@line431 but i dont seem to find pause_gcode line of code. enabling both of them dont seem to work. and the park feature randomly works, sometimes it does park and sometimes, print pause doesnot work (nozzle stops there buring the print). 
Config.ini of the Lcd and configuration and configuration_adv files are added below.
[Configini_configuration_configuration_adv.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5576179/Configini_configuration_configuration_adv.zip)

oldman4U, 2020-11-21:
Please test #1272 to see if this works for you.

CaptHigh, 2020-11-23:
I have updated the firmware. This is an occasional bug, it is working fine till now. ill let you know if it happens again.

closed 2020-12-01 by CaptHigh