bigtreetech/BIGTREETECH-TouchScreenFirmware #1230: [BUG] FileNotFoundError: [Errno 2] No such file or directory: linker.tpl

heavylight-repo, 2020-11-04:
### Description

### Steps to reproduce

1. Download the zip from github "BIGTREETECH-TouchScreenFirmware-master"
2. Open folder -> "BIGTREETECH-TouchScreenFirmware-master" in Visual Studio Code
3. Control that "default_envs = BIGTREE_TFT35_V3_0" is set in platformio.ini
4. In platformio menu choose "Project Task" -> "env:BIGTREE_TFT_V3_0" -> "General" -> "Build"

**Expected behavior**
Compile a .bin file for TFT35 3.0

**Actual behavior**
> Executing task: C:\Users\Admin\.platformio\penv\Scripts\platformio.exe run --environment BIGTREE_TFT35_V3_0 <

Processing BIGTREE_TFT35_V3_0 (platform: ststm32; framework: stm32cube; board: STM32F103VC)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/STM32F103VC.html
PLATFORM: ST STM32 (8.0.0) > STM32F103VC (48k RAM. 256k Flash)
HARDWARE: STM32F103VCT6 72MHz, 48KB RAM, 256KB Flash
DEBUG: Current (blackmagic) External (blackmagic, cmsis-dap, jlink, stlink)
PACKAGES:
 - framework-stm32cube 2.0.200813
 - tool-ldscripts-ststm32 0.1.0
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
Warning! Cannot find a linker script for the required board! Firmware will be linked with a default linker script!
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\Admin\\.platformio\\packages\\framework-stm32cube\\platformio\\ldscripts\\tpl\\linker.tpl':
  File "C:\Users\Admin\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 170:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Admin\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Admin\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Admin\.platformio\platforms\ststm32\builder\main.py", line 99:
    target_elf = env.BuildProgram()
  File "C:\Users\Admin\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Admin\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 61:
    env.ProcessProgramDeps()
  File "C:\Users\Admin\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Admin\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 127:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "C:\Users\Admin\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Admin\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 342:
    SConscript(env.GetFrameworkScript(f), exports="env")
  File "C:\Users\Admin\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 661:
    return method(*args, **kw)
  File "C:\Users\Admin\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Admin\.platformio\packages\tool-scons\scons-local-4.0.1\SCons\Script\SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Admin\.platformio\platforms\ststm32\builder\frameworks\stm32cube.py", line 230:
    LDSCRIPT_PATH=get_linker_script(board.get("build.mcu")))
  File "C:\Users\Admin\.platformio\platforms\ststm32\builder\frameworks\stm32cube.py", line 128:
    with open(template_file) as fp:
================================================================================================ [FAILED] Took 0.87 seconds ================================================================================================

Environment         Status    Duration
------------------  --------  ------------
BIGTREE_TFT35_V3_0  FAILED    00:00:00.874
========================================================================================== 1 failed, 0 succeeded in 00:00:00.874 ==========================================================================================
The terminal process "C:\Users\Admin\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'BIGTREE_TFT35_V3_0'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

oldman4U, 2020-11-04:
Works great on my system:

[image]

Have you tried a simple restart of VSC?

Have you selected "Update All" under PIO?

Delete everything inside .platformio/ and restart VSC. Wait for the index rebuild and try again.

This is most likely a problem with your local environment.

Let us know

oldman4U, 2020-11-07:
Any news?

Please help the community and close the ticket once you do not need it anymore. Thank you

oldman4U, 2020-11-09:
It would be nice if you could reply to the person which tried to help you.

heavylight-repo, 2020-11-10:
Hi, So sorry for late answer. Update all did not work and deleting .platformio did not work. But I reinstalled the VSC and platformio and then it worked.

It was exactly as you said. A problem with my local enviroment! Thank you!

oldman4U, 2020-11-10:
👍🏻

Please help the community and close the ticket once you do not need it anymore. Thank you

oldman4U, 2020-11-12:
Please close the ticket.

closed 2021-01-17 by heavylight-repo