bigtreetech/BIGTREETECH-TouchScreenFirmware #1232: [BUG] Preheat titles color

gordo3di, 2020-11-05:
### Description

### Steps to reproduce

1. Change background color to white.
2. Preheat menu titles are also white.

### Hardware Variant
TFT35 v3
### TFT Firmware Version & Main Board Firmware details
CONFIG_VERSION 20200810

### Additional Information

In preheat.c I changed the title color to the code below. I'm not great with github commits so I was hoping someone could add this. Thank you.

  ```
//set preheat title properties
  lvIcon.lines[0].h_align = CENTER;
  lvIcon.lines[0].v_align = TOP;

  //GTL changed to match font color
  GUI_SetColor(infoSettings.font_color);
  //lvIcon.lines[0].fn_color = LCD_WHITE;

  lvIcon.lines[0].text_mode = GUI_TEXTMODE_TRANS;
  lvIcon.lines[0].pos = preheat_title;
  lvIcon.lines[0].large_font = false;
```

oldman4U, 2020-11-05:
Hi gordo3di.

Great to "see" you again.

You mean preheatMenu.c, right?

I can show you how to make PR's, it is a 10 min action.

oldman4U, 2020-11-05:
So, this would be correct?

[image]

gordo3di, 2020-11-05:
Hey great to esee you as well haha. Yes that's correct. I think I'm working in too many files at once.

Basically I use VS Code to program and I just download the latest repository as a zip file every now and then. For some reason I've tried to grasp the git concept but it seems like I'm not getting it lol. I've tried several tutorials but do you have any suggestions?

I see.you have also gotten these tickets really wrangled in!

radek8, 2020-11-05:
https://desktop.github.com/

oldman4U, 2020-11-05:
;-)

I am working with GitHub on Mac,..... aaaah radek8 my friend was faster;-)

In GitHub you then make a local repository which includes the software you would like to change. Every change is recognised by GitHub and once you are done, you save it to your local repo and after that you simply upload your changes to the server repository. It makes automatically a ticket with a PR - it is MAGIC;-)

Tickets. Slowly the system is usable and many users are really helping a lot, some not...;-) But I had 5 children, I am used to the - I hate you and I wish I would never be born - stuff. For this shit I am really too old.

gordo3di, 2020-11-06:
@radek8 I'll definitely work on it this weekend. Thank you

> Tickets. Slowly the system is usable and many users are really helping a lot, some not...;-) But I had 5 children, I am used to the - I hate you and I wish I would never be born - stuff. For this shit I am really too old.

@oldman4U Haha. I have employees but no kids yet... The system is definitely more useable.

oldman4U, 2020-11-06:
It will improve further. My goal is still <25 tickets and I expect this to happen by the end of the year. Gurmeet installed a Bot which finally will close all those completely useless old tickets. ;-)

Have fun with GitHub

oldman4U, 2020-11-09:
How has your GitHub knowledge improved over the weekend?

gordo3di, 2020-11-10:
No time lately. I'm trying to move so lots of packing. Soon though! I spent
some time today on the BTT UPP power loss/backup module and PLR. Do you
want me to close this topic?

oldman4U, 2020-11-10:
Keep it open until we know who will fix this.

oldman4U, 2020-12-12:
Hi.

Hope you are doing fine.

I will change the issue with my next PR #1380

Please check and let me know if this is correct

Thank you

gordo3di, 2020-12-14:
> Hi.
> 
> Hope you are doing fine.
> 
> I will change the issue with my next PR #1380
> 
> Please check and let me know if this is correct
> 
> Thank you

Looks good. I did notice that sometimes my text will revert back to white after a print has completed and I head back to the preheat menu. I'm not sure why. I'll keep investigating.

oldman4U, 2020-12-14:
Hi. Gurmeet made some additional changes. Because he is the one who knows
the code best I expect it works now. 😁

gordo3di, 2020-12-14:
Perfect! I'll close this ticket. Thank you

closed 2020-12-14 by gordo3di

oldman4U, 2020-12-14:
👍🏻

PR #1380 (2020-12-17): Readme and Preheat titles color
Added information about Reset default settings ... to the ReadMe

Changed Preheat titles color according to gordo3di

resolves #1232