bigtreetech/BIGTREETECH-TouchScreenFirmware #1233: [BUG]Touchscreen not responding after update, TFT43

rennwaffen, 2020-11-06:
### Description

I had just updated my newly acquired TFT43. The touch screen was working before the update. After the update, the touchscreen stopped working. The wiring wasn't changed, and the screen isn't in a case to be pressing on the edges. Also, the LED for the knob is not working, even though it's configured to power on when the printer is on. 

### Steps to reproduce

1. Install firmware
2. Power on printer
3. Touch screen not working

**Expected behavior**
Touch screen to work and knob LED to light up

**Actual behavior**
Touch screen doesn't work and knob LED doesn't light up

### Hardware Variant
TFT43
### TFT Firmware Version & Main Board Firmware details
Marlin 2.0.5.3 and TFT 3.0.26

### Additional Information

blueeagle69, 2020-11-07:
Hi.

Try placing a Reset.txt file on your TFT SD to see if that fixes it.

rennwaffen, 2020-11-07:
> Hi.
> 
> Try placing a Reset.txt file on your TFT SD to see if that fixes it.

I actually don't have the reset.txt file, and I didn't get an sd card with my display. Is it available on GitHub?

oldman4U, 2020-11-07:
There is s readme at the main page. Please read it and come back if it does not help. 

Please help the community and close the ticket once you do not need it anymore. Thank you

rennwaffen, 2020-11-07:
Placed a blank text file in the sd card and named it "reset". This helped recalibrate the display and solve the issue. Thanks!

blueeagle69, 2020-11-07:
> Placed a blank text file in the sd card and named it "reset". This helped recalibrate the display and solve the issue. Thanks!

Good to hear!

oldman4U, 2020-11-08:
Bin mir nicht ganz sicher, weshalb du mir einen Daumen runter gibst, wenn ich dich dazu auffordern muß etwas zu tun, was eigentlich selbstverständlich sein sollte.

closed 2020-11-08 by rennwaffen