bigtreetech/BIGTREETECH-TouchScreenFirmware #1234: [BUG]   BIGTREETECH TFT35 V3.0 udisk error

FranckGib, 2020-11-07:
Hello,
the TFT38 does not detect the insertion of a usb key, while it detects SD cards.
When I click on the USB icon, the tft gives me an error message. "error Udisk". I tested several keys in Fat32 nothing. same problem when there is no usb key in the reading.

the TFT is up to date with version BIGTREE_TFT35_V3.0.26.x.bin.

oldman4U, 2020-11-07:
This is definitely not a bug but a local software or hardware problem. 

Has U disk working before?

More details about hardware and firmware version from info window.

blueeagle69, 2020-11-07:
Yep, Latest firmware on my TFT35 and UDisk is working normally.

CBDesignS, 2020-11-07:
No problems here with the latest git code so I agree with oldman4U & blueeagle69 that this must be a problem that @FranckGib has. check your Marlin build Serial port settings.

#define SERIAL_PORT -1
#define SERIAL_PORT_2 0

oldman4U, 2020-11-07:
As far as I know U disk works independent of serial settings defined.

radek8, 2020-11-07:
TFT38 ???

blueeagle69, 2020-11-07:
I think it was a typo judging by the title.
I looked twice 😀

oldman4U, 2020-11-07:
Must be 35 because 28 does not have a u Disk.

FranckGib, 2020-11-08:
Has U disk working before?

My usb card works perfectly with a TFT28 MKS. The message appears even without a usb card. It will be a material problem, I will see with the supplier ....

it is indeed a tft35 V3.0!

oldman4U, 2020-11-08:
The question was the other way around if the TFTs U Disk has ever been
working before.

But it will finally be a hardware issue I believe.

Please help the community and close the ticket once you do not need it
anymore. Thank you

FranckGib, 2020-11-08:
Sorry, translation problem.
the USB TFT never worked and it is new. TFT just received from the official store. Litigation opened with the supplier. I close the file. Thanks for your help.
Franck

closed 2020-11-08 by FranckGib

oldman4U, 2020-11-08:
Thank you 

Guess the replacement will work instantly.

Happy printing

radek8, 2020-11-08:
> Has U disk working before?
> 
> My usb card works perfectly with a TFT28 MKS. The message appears even without a usb card. It will be a material problem, I will see with the supplier ....
> 
> it is indeed a tft35 V3.0!

I'm crazy about that
this is the third display?
TFT 38
TFT28 MKS
TFT35 V3.0
OMG

FranckGib, 2020-11-08:
yes TFT35 V3.0 
Damn translator who changes my text

oldman4U, 2020-11-08:
I guess that the connector is the problem. Compare it with a working connector.

FranckGib, 2020-11-08:
good idear.

oldman4U, 2020-11-08:
All good, we get used to creative translations;-)