bigtreetech/BIGTREETECH-TouchScreenFirmware #1235: Change Firmware settings for Bltouch

Tobamory, 2020-11-07:
Guys i need to setup the probe offset on the Bltouch but i really cant figure out how to edit the BIN file that we upload, can you actually edit this file, rather than have to compile it all again ?

I also tried the latest firmware, i noticed they have now switched to 25 point BLtouch, but ive also noticed that the TFT 35 doesn't display Temps only working in Marlin old style page

thanks in Advance

radek8, 2020-11-07:
You can change the probe offset in the parameters menu

[image]

[image]

radek8, 2020-11-07:
If you want to change the probe type, you must compile a new BIN file.

charakaSamawry, 2020-11-08:
No guide for compiling a new bin. Recompiling it in vscode or similar is editing it, well you edit the code files in the folder that will be included in the compilation/creation of the .bin that will update the firmware on ur tft. I removed the rgb led menus and recompiled because I didn’t like it assuming I have neopixels and trying to use them by default when it’s an extraneous purchase...

oldman4U, 2020-11-09:
@Tobamory 

Any news related to the asked questions? Anything left open from our side?

Please let us know

Tobamory, 2020-11-09:
Sorry Oldman, yes just one last question, i didnt really understand Charaks response. If i update the printer to the latest Bin file from the V2 firmware site the Graphical TFT display works fine, but the temperatures never activate nor does it print, it will only work if i change to Marlin mode. i stepped back the firmware back to the older 9 point BLtouch version and alls good again, is the firmware broken, or do i need to also update the TFT35 firmware inline with the Printer firmware ? if so can you link the TFT35 one i need please. thanks in advance

oldman4U, 2020-11-09:
I must say I also do not understand what charakasamawry wants to say...;-(

Do you get a "No Printer attached" message on top of the touch screen?

I would strongly recommend to try to use the latest firmware of the TFT and maybe also to compile your own version. Only doing so allows you to get all the functions of it and especially for the TFT it is very easy to do so. Precompiled firmware assumes certain things which can be true but also false, depending on your configuration.

The ReadMe on this page is a very good starting point for this: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware

Please let me know

charakaSamawry, 2020-11-10:
Sorry there is a guide although a bit dated.

I was just trying to say you have to recompile if you want the newest features or your changes to the code. As I’m trying to do just by simply removing some menus from the code, as I don’t have rgb lighting

Tobamory, 2020-11-10:
Thanks guys you can now close this

oldman4U, 2020-11-10:
Hi. Only you can close your own ticket by clicking Close at the bottom of the ticket. 

Thank you

closed 2020-11-10 by Tobamory

oldman4U, 2020-11-10:
Happy printing