bigtreetech/BIGTREETECH-TouchScreenFirmware #1236: [BUG] Unknown Command M503 S0 error

agenttatdaddy, 2020-11-07:
I got this screen a few weeks ago. Since day one I get an error pop-up that says "echo; Unbknown Command; M503 S0" The gcode M503 has something to do with EEPROM memory. I flashed the newest firmware and the problem still exists. I found no info on this issue at all. However, I did come across a site that had a similar issue and recommended creating a file called "eeprom.dat" and placing it on the root of the SD card with no luck as well.. But that was for enabling the EEPROM M500, M501, M502, gcodes, as well as disable M503. In marlin, I disabled that function, but I do not think it applies to the TFT screen. I do not see any issues with prints, and I do not see any issues with the printer that I am aware of, and while it doesn't seem to mess anything up, it is quite annoying to have an error message pop up every time I mess with any settings in the menu. Any help would be greatly appreciated.

Turn on Printer
Boots fine
Then I get Pop_up error "echo: Unknown Command: M503 S0"
I clear it from the error list
Change any setting on the printer and it returns again.

I want to make it stop or at least stop notifying me every 5 mins.

The picture
[image]
 is blurry a bit, hard to catch the damn thing it pops up and disappears so fast.

radek8, 2020-11-07:
In Marlin you have the M503 command disabled
You must enable it by commenting on the line 
//#define DISABLE_M503
It should look like this:

[image]

agenttatdaddy, 2020-11-08:
That's what I have. In fact, when the issue first happened, I went into Marlin and made sure it was disabled, or enabled by commenting out define Diusable_M503. That's what is confusing me. I thought it would correct the issues, but the problem is still persistent.

agenttatdaddy, 2020-11-08:
It seems the issues only come up when it tries to save changed settings to the EEPROM. Such as baby steps, etc. I even tried different firmware. I used both the BIGTREETECH firmware and TH# Studio firmware for the SKR Mini E3,  with the same results.

radek8, 2020-11-08:
If you also have a problem with storing values, it can cause 2 problems.
1. you may have a poorly defined EEPROM in Marlin
2. You have a damaged EEPROM on the Marlin board. The EEPROM has a limited number of writes, if you exceed it, it may be damaged

The solution to the second problem would be to emulate the eeprom in the SD card.

closed 2021-01-14 by stale[bot]