bigtreetech/BIGTREETECH-TouchScreenFirmware #1237: Thumbnail visibility Prusaslicer

neew1, 2020-11-08:
Hi Guys,

first of all, many thanks for your effort.

The Cura plugin for the thumbnails is great, but I actually use the Prusaslicer. Is there such a plugin or something for this one?

According to the Internet, you can (like the Mini) add the item Thumbnail to the corresponding printer profile via Prusaresearch.ini.
But I can't get the thumbnail to be displayed. can someone help me?

friendly greetings

oldman4U, 2020-11-08:
Hi. This is more a feature request. The current version is Cura only and I do not believe that this will be available for other slicers any time soon. If you change this to feature request I will add it to the FR list so that developers are aware of it.

neew1, 2020-11-08:
Ok thanks for the information.

closed 2020-11-08 by neew1