bigtreetech/BIGTREETECH-TouchScreenFirmware #1238: Thumbnail visibility for Prusaslicer

neew1, 2020-11-08:
Hi Guys,

first of all, many thanks for your effort.

The Cura plugin for the thumbnails is great, but I actually use the Prusaslicer. Is there such a plugin or something for this one?

According to the Internet, you can (like the Mini) add the item Thumbnail to the corresponding printer profile via Prusaresearch.ini.
But I can't get the thumbnail to be displayed. can someone help me?
Hardware:
SKR 1.4 Turbo and Btt TFT35 V3.0

friendly greetings

oldman4U, 2020-11-08:
Hi.

I added your feature request to the pinned FR list #1170 . This list has been made so that users which would like to contribute have a single point where they can look for FR and BUGs. This also allows to make sure, that only a single developer works on a certain feature and not several developers - like we had in the past.

It would be great if you could close this ticket. This would allow us to keep this system maintainable.

Thank you and happy printing

traffic-light, 2020-11-09:
Hi, Good idea. But it's not possible to create a plugin for prusaslicer. Then there should be a edited version of prusaslicer which they've to maintain. And that's the hard part...

neew1, 2020-11-09:
It's a shame, I like to work with the slicer because I always had problems with Cura. So many little things that were simply not available with the Prusaslicer.
But today with these screens it should soon be standard. Hopefully

oldman4U, 2020-11-09:
I do not know who finally made the functionality, BTT or a user or someone got paid for it, I do not know. But I could imagine, that they finally followed simply the majority of users and this seems to be 80% plus for Cura.

This is open source, the way is clear, one already walked down the road and now someone else has to go the same way. What the community delivers is what we all can use. There is no manufacturer we can blame or ask to do it.

You can use Prusaslicer and there is nothing wrong with it, but you get the functions which are available for it and those using Cura get the other functions. It is great that there is a choice, isn't it!?

neew1, 2020-11-09:
Sure, of course,
It's a shame anyway;) of course a plus for Cura.
But it's also a shame that I don't know my way around here and can't do anything like programming either.

I wanted to ask if there is something like that and / or someone of you could do something like that, but I also can't estimate the effort.

Since I just manage to configure Marlin to print.

And the most important thing is that with each success, the fun of the thing increases.

And excuse my "English" that is translated from g ....;)

Happy printing

oldman4U, 2020-11-09:
I mean this is a nice to have feature finally and not a must have. Even not all Cura users use this function and use the list view instead.

Maybe one day there will be a similar function for Prusa also, or you give Cura a second try and you decide again afterwards.

traffic-light, 2020-11-09:
[image]

neew1, 2020-11-10:
Oh really?
would that mean that there is a way to possibly set this for the Prusaslicer.

how could i do that

excuse the question.

Kind regards

bigtreetech, 2020-11-10:
Does anyone know the encoding format of prusaslicer thumbnail? Or related document links? another point to consider is whether the resolution of prusaslicer thumbnail can be set. Otherwise, we can't confirm that all TFTs are compatible

traffic-light, 2020-11-10:
I don't know. But I can test or send the file to you @bigtreetech

traffic-light, 2020-11-10:
@bigtreetech I received an awnser:

As answer to Bigtreetech

 

The thumbnail-data from PusaSlicer is a base64 encodet 256 color png.

 

And you can see in my phyton script it writes all four tumbnail sizes to the gcode file

 

def overseek(gfile):

        outdatar = ""

        outdatar = outdatar + overread(QSize(70,70),gfile)

        outdatar = outdatar + overread(QSize(95,80),gfile)

        outdatar = outdatar + overread(QSize(95,95),gfile)

        outdatar = outdatar + overread(QSize(160,140),gfile)

        return outdatar

 

Marc

neew1, 2020-11-10:
So what does that mean exactly?

Can this be inserted anywhere?

excuses these questions,

ETE-Design, 2020-11-11:
@traffic-light So to get this working we need to add it as an Post Processing Script?

traffic-light, 2020-11-16:
[biqu_convert_new.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5549789/biqu_convert_new.zip)

This is the file for thumbnail in prusaslicer.

I don't know how it works. I don't care enough about it at this moment to research it ;-)

effgarces, 2021-01-01:
This works rather nicely, thanks for the script. I already made a small change to it, as it sits now, it saves whatever gcode file it receives with `[Bigtree].gcode` appended to the end of it. I modified it so it simply adds `_btt` (a little bit smaller) and it preserves the original file extension, instead of enforcing `.gcode`:
[biqu_convert_mod.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5758960/biqu_convert_mod.zip)

How to use:
Download and install python 3 (3.9.1 at this moment in time), afterwards, install PyQt5, by going to the command line and issuing `pip install PyQt5`

Next open prusa slicer and add the thumbail generation:
[image]
 

Finaly go to your Print Settings and in Post processing scripts, add the following [python absolute path] [space] [script absolute path] and if any of the paths contain spaces enclose them in "":
[image]

Then when saving the gcode, it will be processed by the script that will replace it with a version with the correct thumbnails for our TFT

sarvenn, 2021-01-10:
@effgarces I tried you instruction but could not go further then installing pyton.
I'm quite far from using pyton, coding etc.
Is it possible to provide a plug and play file? Run and go.
I tried the cura plugin which works great. Now I use prusaslicer and I would be happy to get thumbnail from prusaslicer as well.

effgarces, 2021-01-10:
@sarvenn  I've created a repo with the modded version that I shared above and also "compiled" a windows standalone version, using auto-py-to-exe, if you are on windows give it a try, seems to be working for me.
Grab it here: https://github.com/effgarces/Biqu-Thumbnail-Generator/releases/tag/0.01

sarvenn, 2021-01-10:
Great! Thank you. I'll try it tomorrow and inform you.

sarvenn, 2021-01-12:
> @sarvenn I've created a repo with the modded version that I shared above and also "compiled" a windows standalone version, using auto-py-to-exe, if you are on windows give it a try, seems to be working for me.
> Grab it here: https://github.com/effgarces/Biqu-Thumbnail-Generator/releases/tag/0.01
> With this standalone version you shouldn't have to install python or pyqt, simply copy the exe to the prusalicer.exe or superslicer.exe folder and add the following to the post processing scripts section:
> [image]

@effgarces I tried your solution above recently. It works perfectly. Thank you very much.
I think your solution should drive PrusaSlicer developers to integrate this into slicer software.
How can we take their attention?

3dfixer, 2021-01-17:
works for me. 
when printing from tft it also replaces babystepping icon with the picture!

effgarces, 2021-01-17:
> @effgarces I tried your solution above recently. It works perfectly. Thank you very much.
> I think your solution should drive PrusaSlicer developers to integrate this into slicer software.
> How can we [...]

I really doubt that the prusaslicer developers would integrate this solution as it sits right now. The best approach would probably be rewrite it in c++ and do a pull request on the prusaslicer github but even then I'm not sure if they would be willing to integrate it. Maybe the superslicer developers would be more open to that integration.

effgarces, 2021-01-17:
> works for me.
> when printing from tft it also replaces babystepping icon with the picture!

Yeah, I've also noticed that. The weird is that if start a print from wifi, using esp3d, the picture is missing. Probably a bug in the touchscreen firmware.

sarvenn, 2021-01-19:
@effgarces as you guessed superslicer developers are interested but prusaslicer developer suggested to change BTT's thumbnail view format. Actually it is an another solution may be but this time there Cura will also need to change it's relevant plugin.
Anyway, you solved the problem. Thanks again mate. I wish I knew coding as well to help the community.

gordo3di, 2021-02-06:
Works great! Thank you. I believe it's supposed to replace the babystepping icon intentionally otherwise there's nowhere else to display it.

neew1, 2021-03-01:
Thank you  a lot, this works so nice.
great work.

thanks

bttguy, 2021-03-12:
Nice sollution, thank you!

marcotagliabo, 2021-04-07:
Hi to All...  Someone have done this python "trick" with MacOS?  i've tried but PrusaSlicer get error compiling g-code

Post-processing script /Library/Frameworks/Python.framework/Versions/3.9/bin/python3.9 "/Library/Frameworks/Python.framework/Versions/3.9/bin/biqu_convert_new.py" on file /var/folders/jj/bk768xgs5lsg_r6rrjgj6fgw0000gn/T/.PrusaSlicer.upload.265f-b320-6b5d-a7a7 failed.
Error code: 1
Output:
Python path configuration:
PYTHONHOME = '/Users/marcotagliabo/Library/Application Support/Autodesk/webdeploy/production/aea9db95b176b4e876666d6d1c364de9e4dfc5b8/Autodesk Fusion 360.app/Contents/Frameworks/Python.framework/Versions/Current'
PYTHONPATH = '/Users/marcotagliabo/Library/Application Support/Autodesk/webdeploy/production/aea9db95b176b4e876666d6d1c364de9e4dfc5b8/Autodesk Fusion 360.app/Contents/Frameworks/Python.framework/Versions/Current/lib/python3.7/site-packages'
program name = '/Library/Frameworks/Python.framework/Versions/3.9/bin/python3.9'
isolated = 0
environment = 1
user site = 1
import site = 1
sys._base_executable = '/Library/Frameworks/Python.framework/Versions/3.9/bin/python3.9'
sys.base_prefix = '/Users/marcotagliabo/Library/Application Support/Autodesk/webdeploy/production/aea9db95b176b4e876666d6d1c364de9e4dfc5b8/Autodesk Fusion 360.app/Contents/Frameworks/Python.framework/Versions/Current'
sys.base_exec_prefix = '/Users/marcotagliabo/Library/Application Support/Autodesk/webdeploy/production/aea9db95b176b4e876666d6d1c364de9e4dfc5b8/Autodesk Fusion 360.app/Contents/Frameworks/Python.framework/Versions/Current'
sys.platlibdir = 'lib'
sys.executable = '/Library/Frameworks/Python.framework/Versions/3.9/bin/python3.9'
sys.prefix = '/Users/marcotagliabo/Library/Application Support/Autodesk/webdeploy/production/aea9db95b176b4e876666d6d1c364de9e4dfc5b8/Autodesk Fusion 360.app/Contents/Frameworks/Python.framework/Versions/Current'
sys.exec_prefix = '/Users/marcotagliabo/Library/Application Support/Autodesk/webdeploy/production/aea9db95b176b4e876666d6d1c364de9e4dfc5b8/Autodesk Fusion 360.app/Contents/Frameworks/Python.framework/Versions/Current'
sys.path = [
'/Users/marcotagliabo/Library/Application Support/Autodesk/webdeploy/production/aea9db95b176b4e876666d6d1c364de9e4dfc5b8/Autodesk Fusion 360.app/Contents/Frameworks/Python.framework/Versions/Current/lib/python3.7/site-packages',
'/Users/marcotagliabo/Library/Application Support/Autodesk/webdeploy/production/aea9db95b176b4e876666d6d1c364de9e4dfc5b8/Autodesk Fusion 360.app/Contents/Frameworks/Python.framework/Versions/Current/lib/python39.zip',
'/Users/marcotagliabo/Library/Application Support/Autodesk/webdeploy/production/aea9db95b176b4e876666d6d1c364de9e4dfc5b8/Autodesk Fusion 360.app/Contents/Frameworks/Python.framework/Versions/Current/lib/python3.9',
'/Users/marcotagliabo/Library/Application Support/Autodesk/webdeploy/production/aea9db95b176b4e876666d6d1c364de9e4dfc5b8/Autodesk Fusion 360.app/Contents/Frameworks/Python.framework/Versions/Current/lib/python3.9/lib-dynload',
]
Fatal Python error: init_fs_encoding: failed to get the Python codec of the filesystem encoding
Python runtime state: core initialized
ModuleNotFoundError: No module named 'encodings'
Current thread 0x0000000115a0edc0 (most recent call first):
<no Python frame>

bttguy, 2021-04-10:
@bigtreetech Any possible implementation of native PS thumbnail images without a post processing script?

bigtreetech, 2021-04-12:
> @bigtreetech Any possible implementation of native PS thumbnail images without a post processing script?

Not available yet.

oldman4U, 2021-04-26:
Cura offers what you want.

oldman4U, 2021-06-05:
Good morning. 

I hope you all have seen PR#1973 from user @Cyberdeck which includes the requested feature.

gordo3di, 2021-06-07:
> Good morning.
> 
> I hope you all have seen PR#1973 from user @CyberDeck which includes the requested feature.

Awesome thanks @oldman4U and @cyberDeck !

oldman4U, 2021-06-07:
Cyberdeck did everything I only let you know.

But feedback would be very welcome I believe.

So happy testing

👍🏻

closed 2021-06-11 by bigtreetech

sarvenn, 2021-12-25:
@effgarces hi,
I switched to PrusaSlicer 2.4 beta 1 (final release version does not work on my PC). I used your standalone scripts. It doesn't work. It still works on PS 2.3.3.

This is the error pop up from PS 2.4.0 beta1
[image]

effgarces, 2021-12-25:
I just took a look at this and from my understanding the expected behavior for post-processing scripts changed with prusaslicer 2.4, now the scripts are expected change the g-code in place and not delete/rename the final file. I will see if I can find the documentation on how to properly comply with this and see If I can fix it.

sarvenn, 2021-12-25:
> I just took a look at this and from my understanding the expected behavior for post-processing scripts changed with prusaslicer 2.4, now the scripts are expected change the g-code in place and not [...]

Thanks @effgarces.
Your solution was the best working solution for me.

effgarces, 2021-12-26:
I've updated the script to comply with PrusaSlicer 2.4, the modified version can be found over at https://github.com/effgarces/Biqu-Thumbnail-Generator/ , there's also a "compiled" standalone version, like previously.

carlosjln, 2021-12-28:
@effgarces thanks for updating it 🙏

cronos45, 2022-01-13:
> I've updated the script to comply with PrusaSlicer 2.4, the modified version can be found over at https://github.com/effgarces/Biqu-Thumbnail-Generator/ , there's also a "compiled" standalone version, like previously.

Any idea how I could implement this on Mac OS?

lauchablanca, 2022-02-03:
> > Actualicé el script para cumplir con PrusaSlicer 2.4, la versión modificada se puede encontrar en https://github.com/effgarces/Biqu-Thumbnail-Generator/ , también hay una versión independiente "compilada", como antes.
> 
> ¿Alguna idea de cómo podría implementar esto en Mac OS?

Hola, no puedo encontrar la versión compilada. Sólo el archivo PY. Podrías revisarlo?

effgarces, 2022-02-03:
> > I've updated the script to comply with PrusaSlicer 2.4, the modified version can be found over at https://github.com/effgarces/Biqu-Thumbnail-Generator/ , there's also a "compiled" standalone version, like previously.
> 
> Any idea how I could implement this on Mac OS?

For the Mac you will probably need something like py2app (https://py2app.readthedocs.io/en/latest/) in order to "compile" a standalone version

effgarces, 2022-02-03:
> Hola, no puedo encontrar la versión compilada. Sólo el archivo PY. Podrías revisarlo?

Check the releases area

supermerill, 2022-02-16:
Hi, can someone tell me what's wrong with this 95x95 thumbnail in this file:
[Shape-Box_bqu.gcode.txt](https://github.com/supermerill/SuperSlicer/files/8082187/Shape-Box_bqu.gcode.txt)

I'm trying to implement the thumbnail format natively.

sarvenn, 2022-02-17:
> Hi, can someone tell me what's wrong with this 95x95 thumbnail in this file: [Shape-Box_bqu.gcode.txt](https://github.com/supermerill/SuperSlicer/files/8082187/Shape-Box_bqu.gcode.txt)
> 
> I'm trying to implement the thumbnail format natively.

@effgarces , @CyberDeck , @bigtreetech , @Msq001 , @traffic-light hi guys, could you help @supermerill on his study please?
I did my best but could not help him further. ( https://github.com/supermerill/SuperSlicer/issues/887)

sarvenn, 2022-02-21:
Kind reminder guys.
@effgarces , @CyberDeck , @traffic-light

effgarces, 2022-02-21:
I'm sorry, but right now is a really bad time for me, I really don't have much time at the moment, maybe in two to three days but not guarantees.

alejandrojrey, 2022-05-16:
Hi, is there any way to preserve colors in the preview as the cura plugin does?

Nudekz, 2024-02-15:
[image]
Hi, im getting this error, any sugestions?

emguepo, 2024-04-26:
> ![image](https://private-user-images.githubusercontent.com/160106110/304979393-4c38359b-0a23-485a-937a-aeea2b15ccea.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF [...]

Tengo el mismo problema. tanto en Prusa (que reconoce .exe) como en Orca (reconoce  "C:\Python\python.exe" "D:\biqu_convert_new.py";) . 

El archivo que queda en temporales con extension .pp, es reconocido por la impresora y muestra la miniatura...

[image]
[image]

emguepo, 2024-04-28:
well, i found the solution... only download this other file... 
[image]
[image]

emguepo, 2024-04-28:
> ![image](https://private-user-images.githubusercontent.com/160106110/304979393-4c38359b-0a23-485a-937a-aeea2b15ccea.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF [...]

see my post...

sarvenn, 2024-04-29:
> well, i found the solution... only download this other file... ![Captura de pantalla 2024-04-28 143739](https://private-user-images.githubusercontent.com/88221675/326289651-89cc8e37-c454-491b-8323-7b07daf715bc.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQzNjYyNTgsIm5iZiI6MTcxNDM2NTk1OCwicGF0aCI6Ii84ODIyMTY3NS8zMjYyODk2NTEtODljYzhlMzctYzQ1NC00OTFiLTgzMjMtN2IwN2RhZjcxNWJjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDI5VDA0NDU1OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTZiYTk1NjYwNGM3MDU3OThkYTZkNmEyM2QzMjc2ZDU4YTQwYzc3M2EzYmU4Mjg5Yzg5NDUyNGIyOTJmNjQ0NGEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.2VqHrYzkbSAgMlxWIegBaYhqHtt6RS7Kl-vQoNIZ5DQ) ![Captura de pantalla 2024-04-28 143855](https://private-user-images.githubusercontent.com/88221675/326289658-7070d2ee-705a-46b7-8984-a42f33c8ca83.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQzNjYyNTgsIm5iZiI6MTcxNDM2NTk1OCwicGF0aCI6Ii84ODIyMTY3NS8zMjYyODk2NTgtNzA3MGQyZWUtNzA1YS00NmI3LTg5ODQtYTQyZjMzYzhjYTgzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDI5VDA0NDU1OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTE4NDExNjE3NTRmZmU1NWUzZWZkY2I3YzY1NGM2YTcwZDZhYTJiN2Y1MDc3MWE4NWIwZTY2ODRiMDU0NzU5ZDMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.bf0J1PUSRpQZp4oplu4OO6K6W9lx_0k0iGRCyRx8nxY)

You can move that .py file to your PrusaSlicer folder (where prusaslicer.exe exists) and just  put biqu_convert_p24.py wording into post prosessing script section in your slicer.

emguepo, 2024-04-29:
In PS show this error... 
[image]