bigtreetech/BIGTREETECH-TouchScreenFirmware #1239: [Q] Bootloader for TFT70

soul1988, 2020-11-09:
Hi, I have a TFT70 with a faulty CPU. I replaced the CPU with a new one. Is it possible to get a bootloader for TFT70?

oldman4U, 2020-11-09:
I am just curious to know- why do you replace the CPU of a hardware which is clearly under warranty?

soul1988, 2020-11-09:
Because I was the one who damaged it, and now I'm trying to repair it.

oldman4U, 2020-11-09:
Ok. 👍🏻

darkspr1te, 2020-11-10:
i'am not 100% sure of the cpu/pinouts of the 7inch screen but i have two bootloaders that work with mkstft and btt tft screens, if your device is different it should not take much to port to the device.

soul1988, 2020-11-10:
I have burned the bootloader for the TFT50 that made the screen work, but the touchscreen does not work. STM32F407VGT6
The processor on this screen is the STM32F407VGT6, I don't know if those bootloaders are compatible, but I can try and see if it works.

darkspr1te, 2020-11-10:
this bootloader is for the stm32f407 https://github.com/darkspr1te/mkstft35_V1_open_bootloader , in mks circles it works on the 3.5/7 , not sure about the 5" ,

radek8, 2020-11-10:
> I have burned the bootloader for the TFT50 that made the screen work, but the touchscreen does not work.
> The processor on this screen is the STM32F407VGT6, I don't know if those bootloaders are compatible, but I can try and see if it works.

Did you try to reset the touch layer?
Upload an empty reset.txt file to the SD

soul1988, 2020-11-15:
Ok, is working now.
I just had to rename the files From TFT70 to TFT50 with the bootloader of TFT50.

darkspr1te, 2020-11-15:
@soul1988. If the issue is resolved, could you close this issue.

closed 2020-11-15 by soul1988

radek8, 2022-06-13:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/discussions/1596#discussioncomment-2935828