bigtreetech/BIGTREETECH-TouchScreenFirmware #1240: Onboard SD card non readable

sadbass, 2020-11-10:
### Description
Board skr 1.4
TFT35 V.2 
i install the latest Firmware marlin 2.0.x >(https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3) 
and latest TFT35 v2 Firmware (https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware)
abilitate on Marlin 
#define SERIAL_PORT 0
#define SERIAL_PORT_2 -1
#define SDSUPPORT
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
in the TFT35 config.ini

# Options: [enable: 1, disable: 0, auto-detect: 2]
onboard_sd_support:1

in any case i can see the SD inside the motherboard (skr1.4) via the TFT35 touch mode (RS232)

### Steps to reproduce

1. turn on the printer
2. wait the connection
3. press print
4. select onboard SD

**Expected behavior**
the TFT35 show the file insiede the SD insered in the SKR 1.4 board

**Actual behavior**
3 short beeps and 

### Hardware Variant
TFT35 V2.0
SKR 1.4 
connection RS232 (5 wire)

### TFT Firmware Version & Main Board Firmware details
TFT firmware Vx.x.26
MARLIN Bugfix 2.0.x (https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/tree/master/BTT%20SKR%20V1.4/Firmware/Marlin-bugfix-2.0.x-SKR-V1.4)

sorry for my bad english

oldman4U, 2020-11-10:
Sd card support is DISABLED by default in Marlin. Check and enable.

sadbass, 2020-11-10:
I have uncommented #define SDSUPPORT
You refer to this?

oldman4U, 2020-11-10:
Yes.

oldman4U, 2020-11-10:
@radek8

Do you know if mainboard sd card is supported without EXP connection?

sadbass, 2020-11-10:
Don't know, how can I find this information?

radek8, 2020-11-10:
Yes, but it must be allowed in Marlin
  #define SDCARD_CONNECTION ONBOARD

sadbass, 2020-11-10:
https://github.com/MarlinFirmware/Marlin/issues/19457 in this photo I see touch mode and onboard SD working

radek8, 2020-11-10:
Must be in FAT32 format
Large capacities (> 32G) may not work properly

radek8, 2020-11-10:
allowed in Marlin 
#define AUTO_REPORT_SD_STATUS

radek8, 2020-11-10:
allowed in  config.ini
M27_always_active:1
long_filename_support:2
onboard_sd_support:2

sadbass, 2020-11-10:
Ok after I try and I let you know

radek8, 2020-11-10:
ok

oldman4U, 2020-11-10:
Radekpedia.

oldman4U, 2020-11-10:
For me it looks like the TFT can not access the card.

sadbass, 2020-11-10:
Yeeeeesssssssssssssss is workinggggggggg
#define SDCARD_CONNECTION ONBOARD
#define AUTO_REPORT_SD_STATUS
#define LONG_FILENAME_HOST_SUPPORT

sadbass, 2020-11-10:
Thanks to everyone a lot

closed 2020-11-10 by sadbass

oldman4U, 2020-11-10:
👍🏻😁

sadbass, 2020-11-10:
For all those who already had this problem you absolutely have to know that to allow the runout sensor function if is connected to the skr 1.4, if you print from the gcode on the SD of the TFT35 the sensor does not work well. To unlock the  advanced pause feature you have to launch the print from the gcode in the sd of skr1.4. xoxo pizzamandolino