bigtreetech/BIGTREETECH-TouchScreenFirmware #1245: [Q] (8.3 file names on TFT35 print screen when print from board SD)

lompyka, 2020-11-14:
On the print screen top left corner I can see only the current gcode file in 8.3 format when I print from the motherboard's SD card. In the file list when I browsing the sd for the *.gcode the long filenames are appears but when I choosing one in the popup window also the 8.3 format appears and then on the printing screen too.

I'm using a TFT35 v3.0 with latest master branch FW (TFT35, firmware, lang, config.ini are updated to latest)
and SKR v1.3 with latest bugfix-2.0.x FW

in config.ini:
long_filename_support:2
in board Configuration_adv.h:
#define LONG_FILENAME_HOST_SUPPORT
#define SCROLL_LONG_FILENAMES

What else am I set for display long file names everywhere?
Thanks
[image]

blueeagle69, 2020-11-15:
Hi.
Unfortunately it's a known issue.
At the moment, when printing from the onboard SD you will see long filenames in the file list, but short 8.3 when printing etc.
I hopeful that some clever soul will find the answer.

oldman4U, 2020-11-15:
I add this to the known issues list.

Thank you for reporting this, you can close the ticket if you want.

closed 2020-11-15 by lompyka

kisslorand, 2021-01-09:
Check PR #1488.

oldman4U, 2021-01-09:
Or #1487

kisslorand, 2021-01-09:
> Or #1487

That's not complete, it is not taking in account if printing from USB host and onboard SD hence wrong filename is shown and it is not showing long filename in filament statistics.

guruathwal, 2021-01-10:
@kisslorand  #1488 is still using +4bytes of RAM

kisslorand, 2021-01-10:
My VSCode tells me otherwise. Compiling "master" branch and compiling PR #1488 yields the exact same RAM usage.

guruathwal, 2021-01-10:
@kisslorand  check again

kisslorand, 2021-01-10:
OK. I just checked again. Same results, 0 (ZERO) extra RAM usage.

[image]

guruathwal, 2021-01-10:
There is clearly something wrong in the calculation for MKS28

[image]
[image]

kisslorand, 2021-01-10:
> There is clearly something wrong in the calculation for MKS28

... or in the calculation for the other TFT or you are checking against the wrong master.

kisslorand, 2021-01-10:
> Or #1487

That PR added some more bugs, go figure! Nice tip @oldman4U ;)

carryreynolds, 2021-01-14:
The misreporting of reduced RAM usage is being caused by `"\a"` in `FILAMENTDATA filData = {"\a", 0, 0, 0, 0};` in PrintingMenu.c line number 39 in #1488. The `\a` is escape sequence for `BEL` character with ASCII index of 7 which is used to play beep sounds on computers. In this case, for some devices, the character is not being interpreted correctly by the compiler.

kisslorand, 2021-01-14:
Almost... Replace "\a" with "???", or "abcdefgh" or whatever else you like and check the compiled size again.

carryreynolds, 2021-01-14:
replace `"\a"` with `""` and see the result

kisslorand, 2021-01-14:
FILAMENTDATA filData = {"", 1, 0, 0, 0}; and check ;)

oldman4U, 2021-01-15:
Hello @lompyka

As of today, the reported issue is solved and therefore removed from the FR list. Please install the master firmware to solve the reported issue.

Happy printing

PR #1488 (2021-01-15): Onboard SD print - Long filename; TFT aware of other printing host
### Requirements

BTT TFT or MKS TFT

### Description

If printing from onboard SD (from TFT), printing screen title and filament data displays the long filename (if exists).
I spent more than 10 hours but couldn't make it to display long filename if using Octoprint (or any other USB host) and printing from onboard SD. Sorry for that.

With this PR the TFT is aware when onboard SD print is started from another host (Octoprint, Pronterface, etc). File name, print time, filament usage statistics are shown even if printing from on [... cut]