bigtreetech/BIGTREETECH-TouchScreenFirmware #1248: [Q] On TFT35 V3.0 the layer remains on 0.00 during print when I print from board SD

lompyka, 2020-11-14:
On the printing screen when I print from motherboard SD the Layer is remain on 0.00 mm, if I print from TFT-SD the layer is growing during the print.

I'm using a TFT35 v3.0 with latest master branch FW (TFT35, firmware, lang, config.ini are updated to latest)
and SKR v1.3 with latest bugfix-2.0.x FW

Am I need to set something in the board's marlin fw or in the FTF firmware or it is a bug?

Thanks

[image]

abacabie, 2020-11-14:
Yip, i think this is well known, its never worked for me.

oldman4U, 2020-11-14:
Please see the pinned Feature Request ticket. 

Please help the community and close the ticket once you do not need it anymore. Thank you

closed 2020-11-15 by lompyka