bigtreetech/BIGTREETECH-TouchScreenFirmware #1249: [Q] (Your Question?)

deeniko, 2020-11-15:
Hi all,

I have a TFT35 over an SKR mini E3 v1.2 and I updated the latest version of the firmware (on the TFT35) version 3.0.26.1, for my Ender 3 pro.
This version has a new function that can display the notification messages on the top of the screen for a few seconds. It is called TOAST in the config.ini file (line 112).

https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/blob/master/Copy%20to%20SD%20Card%20root%20directory%20to%20update%20-%20Unified%20Menu%20Material%20theme/config.ini

The issue is that it shows only for a few seconds and it disappears again and I may not be looking at the screen at the time.

Currently I am using Cura to slice my prints and using an extension to display the layers with M117 gcode commands. But I may be able to display other useful information too, if needs to.

[image]

I wonder if someone knows how to change the time (maybe in seconds) for how long the notification display stays active, so I know at what layer the print is at any time I look at the screen.
Otherwise I have to wait for 2 or 3 minutes starring on the screen for the next layer to kick in to show the next message briefly again.
Beneath the notification is the filename, which is not much of use after the print has started.

My preference is to make the notifications to appear all the time until another M117 gcode command refreshes the content (or set it to a long number of seconds e.g. 999, so it stays up for longer).

By the way, this is not a bug so it is not something urgent, but I would like to mention this to the community.

I appreciate your input in advanced,

Dee.

digant73, 2020-11-15:
the duration of the toast notification is currently hardcoded. If you are able to recompile the FW, the parameter you have to set is  TOAST_DURATION in Configuration.h. It is currently set to 3 seconds

deeniko, 2020-11-15:
Thank you! It worked perfectly.

oldman4U, 2020-11-16:
Hello deeniko.

Please help the community and close this ticket.

Thank you

oldman4U, 2020-11-16:
BTW. The Layer Information will be shown while printing from u disk or sd card from TFT.

deeniko, 2020-11-16:
Hi oldman4U,
It shows in millimeters and I have to bring the calculator out to find the layer number. 
The extension in Cura adds the layer number in the message box. There's also an extension to display the remaining time, which could be handy too.
Thank you both for your contribution :)
Dee

closed 2020-11-16 by deeniko