bigtreetech/BIGTREETECH-TouchScreenFirmware #1250: TFT 24 V1.1 on a GT2560

Halefopa, 2020-11-15:
Is it possible to use the TFT24 V1.1 display on a GT2560 (3D printer board)?
Is there any firmware for this combnation?
Normal the GT 2560 is used by the LCD 12864 / LCD 2004.

[image]
[image]
[image]
[image]

radek8, 2020-11-15:
This board does not have pins on the serial interface.
So only the simulated Marlin mode will work for you.
For touch mode you will have to solder and it will be to your risk

https://arduino.stackexchange.com/questions/39414/where-are-rx-and-tx-pins-on-the-gt2560

radek8, 2020-11-15:
Depending on the connection, additional RX and TX outputs are output on the LCD connector. They might go to use but will need to edit the PINout file in Marlin
[image]

Halefopa, 2020-11-16:
Thank you radek8 for your prompt reply.
The Iformatons were very helpful.
At the moment I'm still looking for a suitable Marlin firmware that I can modify for this Projekt (GT2560 + TFT 24 V1.1).

oldman4U, 2020-11-17:
Hello.

What is the status of this ticket?

Please close it once you do not need it anymore.

Thank you

Halefopa, 2020-11-17:
Hey,

No there is a remaining problem...

Status: 
+ I think, the GT2560 and TFT24V1.1 are wired correctly.
+If I plug in the USB-cable of the GT2560, first the System starts updating.
+ after the update it loading all icons.

- but... after loading the Icons, I getting the message:

[image]

I m naot sure you can read that, that message calls:

"Found outdated data:
 Config: Update required
 Icons: Update required
Insert the SD card with the required files and press the reset button to update."

So now I have to fix that problem, but at the Moment i have no idea how to do that.
Thats the Firmware on SD-Card:

[image]

(https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware)

Can anyone help to fix that?

radek8, 2020-11-17:
Read the manual on the title page !!!

A config.ini file with your configuration is required.
download config.ini and customize to your needs.
copy to SD card and paste to TFT.

Halefopa, 2020-11-17:
Ok thanks,

just the icon-error-message is left.
I found no setting in config.ini to fix that.
Can you tell me the concerned file / setting, to fix that last error?

oldman4U, 2020-11-17:
Do the same with tft24 folder. 

Have you read the readme?

radek8, 2020-11-17:
Use this BIN file :-)
[BIGTREE_TFT24_V1.1.26.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5556072/BIGTREE_TFT24_V1.1.26.x.zip)

radek8, 2020-11-17:
Exposed bin files are 2 months out of date and not compatible with BMP files.

Halefopa, 2020-11-17:
It works!

Thank you so much radek8! for all your helpfull comments and the final .bin!

I'm so happy if I meet helfull people like you in a Forum.

oldman4U, 2020-11-21:
Please help the community and close the ticket once you do not need it anymore. Thank you 

radek rules😁

oldman4U, 2020-11-23:
@Halefopa 

Please close this ticket by pressing the Close button at the bottom of this ticket. This will help us to maintain this ticket system and therefore to help other users like we helped you.

closed 2020-11-24 by Halefopa