bigtreetech/BIGTREETECH-TouchScreenFirmware #1252: [BUG] Extended unicode characters are only displayed in the default-language on 3.5" V3.0 E3

arminth, 2020-11-15:
### Description

Today, I changed the language_de.h as follows

  #define STRING_CANCEL                 "\u088F Abbruch" //Cancel Sign
...

  #define STRING_CONFIRM                "\u088E OK" //Confirm Sign

which matches the actual language_en.h now.

I had english as the default language and copied LANGUAGE_DE.ini to the SD-card as secondary language.

I compiled and flashed the 3.5" E3 version and changed from deafult english to secondary german. The Check sign and cross-sign on OK/Cancel messages don't appear anymore..

What I found is that the extended unicode-characters are only displayed, when the display is used in default language. I set German to be the default language and with my above changes, the unicode-characters showed up. I also copied the language_en.ini file to the SD to make english the secondary language. When switching to english as secondary, the unicode-characters immediatly disappeared.

### Steps to reproduce

Change your language_xx.h file as described above with enhanced unicodes. Compile and copy the language_xx.ini to the SD card. Flash display. Switch to secondary language. Try any EPROM save dialog (eg. TMC current change).

**Expected behavior**
Correct interpretation of unicode-symbols both in default and in secondary language.

**Actual behavior**
Unicode will not appear. 
Make your language the default language in configuration.h. Recompile and flash.
Unicode is there.

### Hardware Variant
3.5" V3.0 E3
### TFT Firmware Version & Main Board Firmware details
Compiled from sources dated November 15, 2020

oldman4U, 2020-11-15:
I believe Gurmeet is the language king here.

radek8, 2020-11-15:
The check box and cross are available only in the default language at compile time. Unable to upload from INI file. Thus, the secondary language will always be without these symbols
A check mark and a cross are practical, if you do not speak the default language, you will be oriented according to the symbols.
If these symbols are important to you, you can add them to your language version and compile your own BIN files.
I can create PR and add code to the language files, but you will have to compile it yourself.

arminth, 2020-11-16:
@radek8 
I had already changed the language_de.h file and included the symbols and compiled it. Worked for me and thats when I realized what you just confirmed, that the unicode-characters are only available in the precompiled default language.
This is not obvious to the average user. Is there no possibility to include this feature in the secondary language.ini as well? Or is this due to memory limitation of the display?

What we should definetly do is add the symbols to all language_xx.h files. Currently, it is only in the english version.

Cheers
Armin

radek8, 2020-11-16:
Yes, due to lack of memory, other languages had to be removed from the compilation.
I assure you that the average user will not even notice the missing symbols :-)
Conversely, in many languages, the word OK is translated with a longer expression, and a syllabus at the beginning would mean that the word should be wrapped to the next line. Then it looks better without the missing symbol.

[image]

[image]

PS: Try to compile Russian. There are symboli that are not even in English ;-)

oldman4U, 2020-11-21:
Hello Armin. 

Are you going to add the symbols and make a PR?

Or what is the outcome of this bug report?

arminth, 2020-11-21:
I can make a PR for German, where it doesn't result in a wrap around. But as it is an issue in other languages, I am not sure which is the correct way. It looks nice but it is inconsistent as
a) it might not be usable in every language

and 

b) it isn't evaluated in the secondary language. 

If it would be me to decide, I would actually remove it from the English language file for consistency reasons. 

Not my decision, though as I am not calling the shots here. 

Cheers 
Armin

oldman4U, 2020-11-22:
If you do not like it like it works with a language you would like to use, I would recommend to change it privately or to change it and make a PR. In case another user would also like to change it, you can describe what you did.

At the moment it seems consistent that only the most important language has icons and all others not. My experience is, that localisation in most cases does not work and if you do not understand the few words shown on TFT, the language is most likely not the problem but the lack of understanding the feature/function at all. If I do not know what Leveling does, the translation to another language will also not help.

My 2 cents

closed 2021-01-28 by stale[bot]