bigtreetech/BIGTREETECH-TouchScreenFirmware #1253: Styleguide

traffic-light, 2020-11-15:
Hello everyone,

Because BTT doesn't have a styleguide I'll make some sort of styleguide here.
It's by far not ready. But it's more then nothing.
More styleguide parts are alway's welcome.
Place them in the comments and I'll bring them into this post.

I don't know if there is a good location for this.

@bigtreetech I don't know what you guys think about this, but I think this is something that this project need. At this moment everyone is minding his own way of writing code. That makes the project messy and one time created a bug.
Currently I made some changes in my current PR that will fix some wrong styles ;-)
(if you guys think it should be a pinned issue, feel free to do that. Otherwise there should be another way to inform developers)

### Comments in code
Comments in the code should always be in English. After the comment marks '//' their should be a space.
Example:
```
//Wrong comment
// Correct comment
```

### If else statements
A if else statement should always look like this:
```
if (statementA == checkup)
{
  //Do something for B
}
else if (statementB == checkup)
{
  //Do something for B
}
else
{
  //Default
}
```
_Notice the space in between the if and '('_

When using a if else statement you should always use the curly braces. Even if it's one line!
Some examples:
```
if (whattodo == true) return; // This is correct
if (whattodo == true)
  return;                                // This should be with curly braces
if (whattodo == true)
{
  return;                                //This is correct. 
}
```

oldman4U, 2020-11-15:
👍🏻

If possible I would like to see the known bugs pinned but what about adding this information to the FR and Bug ticket!?

traffic-light, 2020-11-15:
Yeah. I understand that. In my opinion not the best location for a styleguide. But I don't know where it should be.

Maybe a styleguide.md in the files instead..

traffic-light, 2020-11-15:
@oldman4U as we speak, it's already in my PR ;-)

oldman4U, 2020-11-15:
👍🏻

um 21:07:

digant73, 2020-12-03:
My suggestion is to use the general styleguide common to both C/C++, Java developers. It is more or less the one you currently described (and I used in this repo) with just some more clarifications:
1) The general styleguide does not force to use different naming convention for function and variables. In that sense I will allow to use the preferred style
2) related to 1), function and variable declaration/definition MUST always start with a lower case and use and initila capital letter just for any new word or even acronym (e.g. saveSetting() instead of SaveSetting() or "int saveSetting")
3) any type definiton MUST always start with capital letter Just to differenciate type from variable/function definition
4) in C/C++ any constant or macro definition (using #define) MUST always be reported in capital letters (e.g. "#define USE_MBL_MENU", "#define MY_MACRO {int i=0.....}".
This naming convention clearly distinguish function/variable from macro, avoiding users to wrongly consider MY_MACRO as needing ending ";" such as already hanppened in this repo
5) pre-processing operators such as #if,  #ifdef, #if defined(...) etc... must start on column 1. It's the content of the #if operators that must be aligned the the code (as it is provided in the 3rd parties libraries such all the ST libs).

So:

____int i;
____for (i = 0; i < 10; i++)
____{
______// do something
______
______#if USE_MENU == 1
________int j;
________for (j = 0; j < 20; j++)
________{
__________// do something
________}
______#endif

should be:

____int i;
____for (i = 0; i < 10; i++)
____{
______// do something
______int k;
#if USE_MENU == 1
______int j;
______for (j = 0; j < 20; j++)
______{
________// do something
______}
#endif

See that "int j;"and the following instructions are correctly aligned with the previous instruction "int k;"

I apreciate both of you efforts on trying to reduce the anarchy followed in this repo trying to establish some basic rules, bugs tracking, feature list etc.., However, without some rules imposed and overseen by the owner of the repo (so BTT), one of most typical situation is that people tend to consider the repo as its own, changing here and there without any ageement or any real improvement but based on their only opinion and tastes. That's not good at all in a team and simply let other people to leave the project.

traffic-light, 2020-12-03:
Thank you @digant73. I'll merge it into the current one

traffic-light, 2020-12-03:
By the way. I'm not trying to force anything.
When people think it should be different when looking at my PR: Be my guest. I'll revert the thinks I did (at least when it's a well defined criticism/ arguments)

traffic-light, 2020-12-03:
And feel free, when my pr is finished and merged, to adjust the styleguide ;-)

digant73, 2020-12-03:
I agree with your opinions. That was also my attitude working in this repo.
I already had a look to your PR. it is ok (you properly limited changes on some basic styles). As I said some of my recommendations above are best practice and I didn't applied them because it will have big impact on source code written by others (in that case I tried to respect their style even if it was a pretty home made style not compliant to any guide line)

oldman4U, 2020-12-03:
My colorful friend. Your PR is the next on my test list😁

traffic-light, 2020-12-03:
> My colorful friend. Your PR is the next on my test listgrin

Thank you. only problem is that it currently isn't  working (pre-heat alert but didn't finish it)

oldman4U, 2020-12-03:
Take your time. I know you are learning.

um 21:49:

kisslorand, 2020-12-03:
It hurts my eyes seeing all over the place
`if (a>b){
   random();
}`
instead of:
`if (a>b)
{
   random();
}`
Maybe I'm just too old-school...

traffic-light, 2020-12-04:
And it's not consistent.
For example:
```
If (a > b){
  Do();
}
Else{
  Do();
}
```
When you do this make the else this way: `}else{`

I'm trying to help people using one guideline.
I hope that the code will be more consistent....

closed 2021-02-09 by stale[bot]