bigtreetech/BIGTREETECH-TouchScreenFirmware #1254: [Q] BIQU_TFT35_APP1_V2.0.26.x.bin - No touch screen after update. Old bin to go back?

brosskgm, 2020-11-16:
TFT35 V2.0

After running the BIQU_TFT35_APP1_V2.0.26.x.bin and the files said to copy to SD Card I lost all touch.

Is there a way to go back to the old version to get touch back?

Thanks
Bob

MacedoRAC, 2020-11-16:
Same, but on TFT24

radek8, 2020-11-16:
create an empty reset.txt file on the SD, insert it into the TFT and reset the printer.

MacedoRAC, 2020-11-16:
I've done that 2 times. The calibration succeeds but the screen stil gets stuck on display mode selection screen and I'm unable to select any of the options (Touch or classic marlin mode)

MacedoRAC, 2020-11-16:
Also I saw on an issue, someone made  a pull request to fix this and it was accepted. But I tried to use the firmware version on Master branch but the same still happens

radek8, 2020-11-16:
Calibration was successful?

MacedoRAC, 2020-11-16:
@radek8 yes.

Right now I'm not with the printer but tonight I can record a video showing the problem

radek8, 2020-11-16:
The touch may not work if the frame is pushing on the edge of the screen.

Ok, send the video

radek8, 2020-11-16:
BIN files are obsolete, use custom compilation

MacedoRAC, 2020-11-16:
> The touch may not work if the frame is pushing on the edge of the screen.
> 
> Ok, send the video

That can be the problem once I use a 3D printed case for the display.

> BIN files are obsolete, use custom compilation

I use custom compilation all the time. I never use the bin files from the repo.

radek8, 2020-11-16:
> TFT35 V2.0
> 
> After running the BIQU_TFT35_APP1_V2.0.26.x.bin and the files said to copy to SD Card I lost all touch.
> 
> Is there a way to go back to the old version to get touch back?
> 
> Thanks
> Bob

@brosskgm Do not use the old version. The new one is better and must work

brosskgm, 2020-11-16:
> Calibration was successful?

Yes.

brosskgm, 2020-11-16:
> The touch may not work if the frame is pushing on the edge of the screen.
> 
> Ok, send the video

> > TFT35 V2.0
> > After running the BIQU_TFT35_APP1_V2.0.26.x.bin and the files said to copy to SD Card I lost all touch.
> > Is there a way to go back to the old version to get touch back?
> > Thanks
> > Bob
> 
> @brosskgm Do not use the old version. The new one is better and must work

I tried both. It worked prior to update. Some have said compile myself. Looking for that also. Will try the reset.txt shortly.

brosskgm, 2020-11-16:
> > TFT35 V2.0
> > After running the BIQU_TFT35_APP1_V2.0.26.x.bin and the files said to copy to SD Card I lost all touch.
> > Is there a way to go back to the old version to get touch back?
> > Thanks
> > Bob
> 
> @brosskgm Do not use the old version. The new one is better and must work

The BIQU_TFT35_APP1_V2.0.26.x.bin was in both folders. I don't see a newer one in the download.

brosskgm, 2020-11-16:
reset.txt worked.

closed 2020-11-16 by brosskgm

MacedoRAC, 2020-11-16:
> reset.txt worked.

@brosskgm  do you have your TFT in any kind of enclosure?

brosskgm, 2020-11-16:
> > reset.txt worked.
> 
> @brosskgm do you have your TFT in any kind of enclosure?

https://www.thingiverse.com/thing:4091563

brosskgm, 2020-11-16:
> > > reset.txt worked.
> > 
> > 
> > @brosskgm do you have your TFT in any kind of enclosure?
> 
> https://www.thingiverse.com/thing:4091563

Now all I have to do is get Marlin 2.0.7 to compile configured with this TFT.

radek8, 2020-11-16:
@brosskgm   try this
[BIQU_TFT35_APP1_V2.0.26.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5548622/BIQU_TFT35_APP1_V2.0.26.x.zip)