bigtreetech/BIGTREETECH-TouchScreenFirmware #1256: [BUG] (speed increasing when using encoder to increase temperature)

Stillbeer, 2020-11-16:
### Description

### Steps to reproduce

1. start print job
2. while heating the bed goto: -more - heating 
3. use the encoder to increase the temperature.
4. go back and speed is also increased!

**Expected behavior**

Only temperature change
**Actual behavior**

Temperature AND speed change
### Hardware Variant

btt tft35-e3 v3.0
### TFT Firmware Version & Main Board Firmware details

BIGTREE_TFT35_V3.0_E3.26.x - Marlin 2.0 bugfix

### Additional Information

oldman4U, 2020-11-16:
Hi.

Maybe you want to read the very first entry of the Read Me First, which is pinned at the top of the main page:
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1047

Please close this ticket after that.

Thank you

radek8, 2020-11-16:
694/5000
It's a familiar problem.
Read the FAQ pinned to the front page
if you have SKR1.4 and load the latest Marlin with error correction, compiling a new FW for TFT will solve the problem.

If you have another board with Marlin connected via EXP1 and EXP2
It is possible to solve the problem by editing the pin file in Marlin.
         #if ENABLED (REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER)
           #define BTN_ENC_EN LCD_PINS_D7 // Detect the presence of the encoder
         #endif

If you have Marlin connected via EXP3 (it works in the previous case as well) the solution is to turn off in Marlin the automatic return to the status screen.
#define LCD_TIMEOUT_TO_STATUS -1

Then just click on the encoder after turning on the printer 1.

Stillbeer, 2020-11-17:
Oke, now i feel SO STUPID ! ( READ THE FAQ )
Thanks for the kind answers!

closed 2020-11-17 by Stillbeer

oldman4U, 2020-11-17:
No problem at all!!!

Thank you and happy printing