bigtreetech/BIGTREETECH-TouchScreenFirmware #1264: [FR] Add GUI-support for G35 - Tramming assistant

OlavStornes, 2020-11-17:
**Is your feature request related to a problem? Please describe.**
It is a bit cumbersome to use [G35](https://marlinfw.org/docs/gcode/G035.html) as a custom gcode-command, since the return value is often hidden behind the `recv: busy processing`-messages that the board receives when waiting on the result.

**Describe the solution you'd like**
A solution could be as simple as having a separate button under the leveling-menu which calls G35 and return the intended text (which adjustments the user must make) when it is complete.

**Describe alternatives you've considered**
[Next marlin release will maybe implement something similar](https://github.com/MarlinFirmware/Marlin/pull/20000)

oldman4U, 2020-11-18:
Hi.

Seems not many users are interested in Tramming but I added your feature request to our centralised FR list #1170.

It would be great, if you could close this ticket.

Thank you

ETE-Design, 2020-11-19:
@oldman4U Tramming is a relative new feature added to Marlin so that could be why not many users are interested... But would like to see that feature added to the TFT... Mabye @traffic-light can help?

traffic-light, 2020-11-19:
Maybe later. First I've got to finish my current pr. I've got a lot to do at this moment

oldman4U, 2020-11-19:
I have seen tramming in Marlin and like many other features it would be great to have it. Looking at the FR list I can see many things and only a few developers.

oldman4U, 2020-11-19:
Most users are not able to close the tickets they have started, or even do not respond once they start their first ticket. This all takes time and resources.

traffic-light, 2020-11-19:
Make a PR with all the tickets tagged with resolves in it and hope that btt will merge it 🤣

oldman4U, 2020-11-19:
Ohoh. Bad boy😎

um 09:33:

OlavStornes, 2020-11-20:
> Seems not many users are interested in Tramming but I added your feature request to our centralised FR list #1170.
> 
> It would be great, if you could close this ticket.

The feature came in late summer. so i would assume not many have explored it yet. I dont think it is due to people not being interested in it.

I feel that closing the ticket just because it is referenced in a FR list is not a good practice. Do you still want to close it?

oldman4U, 2020-11-20:
Read #1170 and decide by yourself.

closed 2021-01-26 by stale[bot]