bigtreetech/BIGTREETECH-TouchScreenFirmware #1266: [BUG] TFT 35 Touch slow respondig - SKR Mini E3 V2.0

KVC87, 2020-11-17:
Hello,

I have a SKR Mini E3 V2 with the TFT 35 Touchscreen.

Some features have slow reaction, if i press the fan to go on or off, it takes about a minut to go on or of. Also with the move > home en stof or pause button, i takes time to stop.

Temperatue from nozzle and bed goes directly.

Is this normal?

oldman4U, 2020-11-18:
Which firmware versions?

Any hint why you believe it is TFT and not SKR related?

KVC87, 2020-11-18:
> Which firmware versions?
> 
> Any hint why you believe it is TFT and not SKR related?

The latest one, but somer research made me do think that when you send the command,the code have to wait in line...

oldman4U, 2020-11-18:
Send a picture of Menu - Settings - Info.

Stop and Pause has to clear the buffers first, so this depends on the buffer settings. Fan should change immediately, or at least it should not take a minute.

KVC87, 2020-11-18:
[image]

oldman4U, 2020-11-18:
Please let the Upload end before you click on Comment.

KVC87, 2020-11-18:
[image]

oldman4U, 2020-11-18:
Please follow the update description from here:
https://github.com/oldman4U/BIGTREETECH-TouchScreenFirmware - scroll down on the page -
using the firmware from here:
https://github.com/radek8/BIGTREETECH-TouchScreenFirmware/tree/updateBIN

KVC87, 2020-11-18:
Thanks, I will update it right now and make a print to see if this works out.
I

oldman4U, 2020-11-18:
Wonderful.

KVC87, 2020-11-18:
Owkay, its faster. Bu it is also a smaller print. I see when i push to start the fan sometimes its start after 5 seconds sometimesafter 10. So i think its more thathe is waiting in line.

oldman4U, 2020-11-18:
You will have to change the RT and TX buffers in Marlin firmware for faster response.

Are you printing from SD card of the mainboard or from something connected to TFT?

KVC87, 2020-11-21:
I’m printing from the sd slot on the screen. The slot on the board is not accesible because i build it onther case. 

Their is still a little delay, but not that long i can live with it 😀
[image]

oldman4U, 2020-11-21:
A little delay is normal.... as long as it is less than a minute;-)

Please close this ticket.

Happy printing

closed 2020-11-21 by KVC87