bigtreetech/BIGTREETECH-TouchScreenFirmware #1267: [BUG] No response for M503 in Terminal

kisslorand, 2020-11-17:
### Description

When an M503 gcode is issued in Terminal there's no response shown. Actually only the last few characters are shown:
cho: M410 S1
OK P:15 B:15

### Steps to reproduce

1. [Open Terminal]
2. [Type M503]
3. [Press Send]

**Expected behavior**

The full response for M503

**Actual behavior**

Only the last few characters are shown:
Rcv: cho: M410 S1
Rcv: OK P:15 B:15

### Hardware Variant

MKS TFT clone on Artillery Sidewinder X1 with SKR 1.4Turbo mainboard
### TFT Firmware Version & Main Board Firmware details
TFT FW: 0.0.26.1, Marlin 2.7.2

### Additional Information

When M503 S0 is issued the response is full.

radek8, 2020-11-17:
I can confirm this behavior.

@digant73 
Was the output on the M503 command terminal suppressed in your last PR?

radek8, 2020-11-17:
Comment on these 3 lines.
It might help ....

[image]

kisslorand, 2020-11-17:
Would be easier to allow "knownEcho" to be shown in terminal, I just cannot find a way to do it.

radek8, 2020-11-17:
I wrote that to you

kisslorand, 2020-11-17:
In "menu.c"  "parseACK()" is called where "syncL2CacheFromL1(SERIAL_PORT)" fills "dmaL2Cache". The "knownEcho" elements must be removed somewhere from "dmaL2Cache" since they are not shown in terminal.

radek8, 2020-11-17:
Unfortunately, I won't help you anymore.

kisslorand, 2020-11-17:
It is not about me.

radek8, 2020-11-17:
I can't help, this is where my knowledge ends :-(

digant73, 2020-11-18:
no changes were made on Terminal related to gcode handling. Also M503 correctly works on my side (5 pages of output)

kisslorand, 2020-11-18:
M503 S0 works for me too, the plain M503 gives me trouble.
M503 gives a larger response than the "ACK_MAX_SIZE", M503 S0 gives a smaller response than "ACK_MAX_SIZE".

oldman4U, 2020-11-18:
Looks like this on my TFT. [image]

[image]

[image]

kisslorand, 2020-11-18:
From the pictures I can see it has much less response characters than in my situation. Also this TFT uses different MCU than mine. This one is with STM32F207 and mine has STM32F107. My TFT has no issue to display the response from M503 S0 which has around 1500 opposed to the ~2500 characters it gets from the simple M503 gcode (no S0 parameter).

oldman4U, 2020-11-18:
So there is a problem with your TFT and others like mine work well. What is your screen size?

kisslorand, 2020-11-18:
It is the original 2.8" TFT on the Artillery Sidewinder X1.
Radek8 also confirmed the behaviour I described.

radek8, 2020-11-18:
I can confirm the same behavior as kisslorand.
But my HW is a TFT 35 E3 V3.0 like Oldman4U.

radek8, 2020-11-18:
If I send the M503 command from the PC via Wifi, the response is correct and complete.

oldman4U, 2020-11-18:
Can u send a picture?

radek8, 2020-11-18:
I'll send in the evening

oldman4U, 2020-11-18:
Not sure how many characters it is, but on my TFT28 M503 gives me 6 pages
[IMG_7445.HEIC.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5561107/IMG_7445.HEIC.zip)

radek8, 2020-11-18:
M503
[image]

radek8, 2020-11-18:
M503 S0
[image]
[image]
[image]
[image]
[image]
[image]

kisslorand, 2020-11-18:
[image]
Disregard the SmileyFace, it's from my phone.

oldman4U, 2020-11-18:
Strange. So it works on my machine, but not on yours. I will test another machine and let you know...

oldman4U, 2020-11-19:
Thats my SKR E3 DIP TFT35 E3 v3 based Ender 3.

[image]
[image]
[image]
[image]

kisslorand, 2020-11-19:
What Marlin version are you on? 
I am on 2.7.2 bugfix.
I went as far with debugging as checking the "infoHost.rx_ok[SERIAL_PORT] " after sending "M503". It gets "true" only two times. I checked it in "parseACK.c" after "syncL2CacheFromL1(SERIAL_PORT);" (line 263).
Since it gets "true" only two times it is no surprise the terminal shows only two lines of response.

oldman4U, 2020-11-19:
Marlin 2.0.7.2 from Oct. 22nd. I guess this is the date where this firmware version has been released officially.

I must say that I always use 115200 baud rate and set the TFT connection as the first serial connection and -1 or nothing as the second serial connection.

Let me know in case I can help somehow but must add, that I am no developer. So may it is more work for you to explain to me what I have to do, than the amount of work I can do;-)

kisslorand, 2020-11-20:
Problem solved, the size of the response for M503 didn't fit the DMA and Terminal buffer. Solved it by enlarging those buffers.

radek8, 2020-11-20:
May I ask a solution to the problem?
Thank you

oldman4U, 2020-11-21:
@kisslorand 

Are you going to make a PR for this?

kisslorand, 2020-11-21:
I fixed it for my printer, others might check if it is OK with theirs.
Here's the PR: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/1276

radek8, 2020-11-21:
Good job.
It already works as it should :-)

[image]
[image]

closed 2020-11-27 by bigtreetech