bigtreetech/BIGTREETECH-TouchScreenFirmware #1268: is the smart runnout filament sensor (by BTT) working fine with TFT 35 V3.0 in toutchscreen mode?

Chukozor, 2020-11-18:
is the smart runnout filament sensor (by BTT) working fine with TFT35 3.0 in toutchscreen mode?

digant73, 2020-11-18:
if you have some filament runout wrong error messages sometimes during a print try to increase the value for "fil_noise_threshold" in config.ini (e.g. set it to 150 or more)

Chukozor, 2020-11-18:
I mean I didn't bought it yet, I saw some videos and read some topics saying that the smart runnout filament sensor only work in marlin emulated mode, not in touchscreen mode, so that's why I asked if it has been fixed.

oldman4U, 2020-11-18:
Hi.

There are some open and closed tickets like #312 related to this topic. Maybe it would be best to check them out to get an overview and how it works.

Connecting to the mainboard seems to make it work using Marlin mode and connecting to TFT seems to make it work in TFT mode. But please see by yourself.

Please help the community and close this ticket once you do not need it anymore.

Thank you

blueeagle69, 2020-11-19:
The only success I've had with the BTT sensor is my connecting it to the mainboard instead, and disabling Runout on the TFT. Then with the relevant Marlin settings it appears to work fine with the TFT in both modes.
See here;
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1121

oldman4U, 2020-11-21:
@Chukozor 

Do you have any questions left regarding the smart filament sensor?

Chukozor, 2020-11-22:
Hi, thanks all for your replies. I don't, I'm gonna order one and hope I won't need to reopen a question about it :)

closed 2020-11-22 by Chukozor