bigtreetech/BIGTREETECH-TouchScreenFirmware #1270: Card Inserted on boot when no SD in the TFT35 E3

Doyle4, 2020-11-19:
Hi guys,

Installed the TFT35 E3 V3 on my MKS SGEN L 32bit V2.0

I have tried many wire configs, and the way I have it now is the only way it'll work, one change of a cable and back to not working.

Problem I have is, on boot it says Card Inserted when there isnt, if I put one in it reads fine.
If I boot the printer with an SD card in, I get Illegal Flash App warning.
My rotary LED does not work either.

Latest Firmware for TFT35 E3 Installed.

oldman4U, 2020-11-19:
Hi.

What is latest exactly?

Menu - Settings - Info has to show today or yesterday. otherwise it is no the latest.

Doyle4, 2020-11-19:
> Hi.
> 
> What is latest exactly?
> 
> Menu - Settings - Info has to show today or yesterday. otherwise it is no the latest.

Very true! I’ll add that info when home thanks.

Doyle4, 2020-11-19:
@oldman4U im using the very latest from https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware as I downloaded it and flashed.

info shows: TFT35_E3_V3.0.26.X Sep 12 2020

Still have Card Inserted message on boot, and Illegal File App if SD left in and printer rebooted.

oldman4U, 2020-11-19:
@bigtreetech 

Could you please merge radek8 and my PR. Those poor users depend on you to release it. THANK YOU

@Doyle4 
Please use the firmware from here: https://github.com/radek8/BIGTREETECH-TouchScreenFirmware/tree/updateBIN
following the instructions from here:
https://github.com/oldman4U/BIGTREETECH-TouchScreenFirmware - scroll down.

Very likely this will not solve your problem, because it is more likely hardware related, but this is the next step.

Doyle4, 2020-11-19:
@oldman4U Thank you very much for your quick responses.
I'll try this soon and will report back thanks 👍

Doyle4, 2020-11-19:
@oldman4U 
Sorry for another question, wanting to make sure on this.
On the 2nd link it says "make sure that **ONLY** the "REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER" is activated in Configuration.h and that all other controllers (especially the "CR10_STOCKDISPLAY") are Deactivated" 

But in config.ini it says you must have these enabled:
M115_GEOMETRY_REPORT (in Configuration_adv.h)
#  REPORT_FAN_CHANGE (in Configuration_adv.h)
#  EMERGENCY_PARSER (in Configuration_adv.h)
#  SERIAL_FLOAT_PRECISION 4 (in Configuration_adv.h)

This is still correct to must have enabled also? Thanks again

Doyle4, 2020-11-19:
Sorry, unsure why its copying bold

oldman4U, 2020-11-19:
> @oldman4U Thank you very much for your quick responses.
> I'll try this soon and will report back thanks 👍

You are very welcome.

oldman4U, 2020-11-19:
On the 2nd link it says "make sure that ONLY the "REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER" is activated in Configuration.h and that all other controllers (especially the "CR10_STOCKDISPLAY") are Deactivated"

The important part is CONTROLLERS, which means displays. We had issues where users had the CR10 AND the RepRap controller activated. All other functions can be activated if needed.

Doyle4, 2020-11-19:
@oldman4U Flashed, reset eeprom on printer, restored screen to default settings.

Sadly "Card Inserted!" on boot and Rotary LED non functioning still.

radek8, 2020-11-19:
Can I join?

Doyle4, 2020-11-19:
Feel free :)

Love the new terminal look btw! great job!

oldman4U, 2020-11-19:
Almost thought so. What is the firmware version and have you tried to change the color of the encoder in the TFT menu?

Do you have a broken usb cable which you do not need any more?

radek8, 2020-11-19:
[BIGTREE_TFT35_V3.0_E3.26.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5568471/BIGTREE_TFT35_V3.0_E3.26.x.zip)

Use this bin file

radek8, 2020-11-19:
> Ale v config.ini říká, že musíte mít tyto povoleny:
> M115_GEOMETRY_REPORT (v Configuration_adv.h)
> 
> REPORT_FAN_CHANGE (v Configuration_adv.h)
> EMERGENCY_PARSER (v Configuration_adv.h)
> SERIAL_FLOAT_PRECISION 4 (v Configuration_adv.h)
> Je to stále správné, musíte také povolit? Ještě jednou děkuji

radek8, 2020-11-19:
What's your board with Marlin?

radek8, 2020-11-19:
also enable:

 M114_DETAIL  (in Configuration_adv.h)
 HOST_ACTION_COMMANDS  (in Configuration_adv.h)
For printing from onboard SD
 SDSUPPORT (in Configuration.h)
 AUTO_REPORT_TEMPERATURESS (in Configuration_adv.h)
 AUTO_REPORT_SD_STATUS (in Configuration_adv.h)
 LONG_FILENAME_HOST_SUPPORT (in Configuration_adv.h)
 SDCARD_CONNECTION ONBOARD (in Configuration_adv.h)

Doyle4, 2020-11-19:
@radek8 I have the MKS SGEN L V2.0 (2020 version) with Marlin 2.0.7.2

I'll make sure they are enabled also thanks!

radek8, 2020-11-19:
when flashing, you must also use the folder with icons and the TFT35 font
Config.ini with its own configuration and language ini file according to your own preference.

radek8, 2020-11-19:
I recommend adding this code.
This should solve the speed change problem when using the encoder in touch mode.

[image]

Doyle4, 2020-11-19:
Verbose mode can be enabled via `-v, --verbose` option
In file included from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../pins/pins.h:423,
                 from buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/MarlinConfig.h:34,
                 from buildroot/share/PlatformIO/scripts/common-dependencies.h:31:
buildroot/share/PlatformIO/scripts/../../../../Marlin/src/inc/../pins/lpc1769/pins_MKS_SGEN_L_V2.h:244: error: unterminated #elif
  244 | #if IS_TFTGLCD_PANEL
      | 
CalledProcessError: Command 'arm-none-eabi-g++ -D__MARLIN_FIRMWARE__ -DU8G_HAL_LINKS -D__MARLIN_DEPS__ -w -dM -E -x c++ buildroot/share/PlatformIO/scripts/common-dependencies.h' returned non-zero exit status 1.:
  File "/Users/Doyle/.platformio/penv/lib/python3.8/site-packages/platformio/builder/main.py", line 168:
    env.SConscript(item, exports="env")
  File "/Users/Doyle/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Script/SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "/Users/Doyle/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Script/SConscript.py", line 287:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "/Users/Doyle/Desktop/Custom Firmware/MKS SGEN L V2/Marlin-2.0.7.2/buildroot/share/PlatformIO/scripts/common-dependencies.py", line 286:
    apply_features_config()
  File "/Users/Doyle/Desktop/Custom Firmware/MKS SGEN L V2/Marlin-2.0.7.2/buildroot/share/PlatformIO/scripts/common-dependencies.py", line 118:
    if not env.MarlinFeatureIsEnabled(feature):
  File "/Users/Doyle/.platformio/packages/tool-scons/scons-local-4.0.1/SCons/Environment.py", line 219:
    return self.method(*nargs, **kwargs)
  File "/Users/Doyle/Desktop/Custom Firmware/MKS SGEN L V2/Marlin-2.0.7.2/buildroot/share/PlatformIO/scripts/common-dependencies.py", line 262:
    load_marlin_features()
  File "/Users/Doyle/Desktop/Custom Firmware/MKS SGEN L V2/Marlin-2.0.7.2/buildroot/share/PlatformIO/scripts/common-dependencies.py", line 250:
    define_list = subprocess.check_output(cmd, shell=True).splitlines()
  File "/Users/Doyle/.platformio/python3/lib/python3.8/subprocess.py", line 411:
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/Users/Doyle/.platformio/python3/lib/python3.8/subprocess.py", line 512:
    raise CalledProcessError(retcode, process.args,

Doyle4, 2020-11-19:
Im getting this error when compiling

Doyle4, 2020-11-19:
#if ENABLED(ULTIPANEL)
          #define LCD_PINS_D5              P0_17
          #define LCD_PINS_D6              P1_00
          #define LCD_PINS_D7              P1_22
        
        **#if ENABLED(REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER)
          #define BTN_ENC_EN         LCD_PINS_D7**
        
        #endif

      #endif // !FYSETC_MINI_12864

    #endif // !MKS_12864OLED_SSD1306

  #endif // !CR10_STOCKDISPLAY

#endif // HAS_WIRED_LCD

Doyle4, 2020-11-19:
possible for the MKS SGEN L pins.h file? if I change mine compiling fails.

radek8, 2020-11-19:
Sorry my mistake

radek8, 2020-11-19:
You wrote that you have MKS SGEN L V2.0

Doyle4, 2020-11-19:
Sorry, my mistake also, Possible for the MKS SGEN L V2.0 pins.h file please with the added code?

The ** I thought made it bold on here, not added in pins.h file.

Thanks for your help :)

radek8, 2020-11-19:
Use Marlin 2.0 with bug fixes

radek8, 2020-11-19:
You must add 3 lines

I forgot the last line :-(

[image]

Doyle4, 2020-11-19:
Thank you, will download bugfix when home, I had to pop out quick during your help. Will update when done.

oldman4U, 2020-11-19:
Question. With NO SDcard inserted, do you see a green led lid up at the backside of the TFT close to the SDcard slot and the WIFI connector?

Doyle4, 2020-11-19:
Not that Im aware, i’ll add that to my check list. Theres a red led always on, but im assuming thats the power indicator.

Doyle4, 2020-11-19:
@oldman4U Can confirm green LED is on without an SD card inserted.

oldman4U, 2020-11-19:
So you have a hardware issue.

Doyle4, 2020-11-19:
Crap, I need to get in touch with supplier then
uggghhhhhh

oldman4U, 2020-11-19:
Or you see if you can find the issue and fix it.

Doyle4, 2020-11-19:
Not too sure what id be looking for

oldman4U, 2020-11-19:
Best would be to compare it with a working sd card reader.

Doyle4, 2020-11-19:
Ahh, Dont think im at that level just yet.

Iv opened a ticket with who I bought it from.

Doyle4, 2020-11-19:
Something else I noticed with the screen, it would change the speed% without me touching anything, changing it was slow as it would go back to wrong% if too fast. This was the same with Hotend fan speeds.

Doyle4, 2020-11-19:
faulty cable that came with it, swapped it and the SDcard problem has gone, but the led on rotary doesn't work, and Marlin Simulator mode stays black, the rotary controls the hotend fan and also plays with nozzle and bed temp whilst attempting to use M.S mode

Doyle4, 2020-11-20:
@radek8 @oldman4U Thank you, I think we have solved it.

Did everything you guys have suggested, apart from SDCARD_CONNECTION ONBOARD (in Configuration_adv.h) I had to change to LCD as ONBOARD the sd port is covered and I have to open my printer up to access it to update, where as LCD mode lets me connect to my computer via usb and drag the file across.

This is where I wanted to kick my self.. The SDCARD INSERTED message was from the mainboard... yeahh... I forgot to remove it a while ago now, really sorry about that.

So far everything seems fine, only thing not working still is the rotary LED, but I can live with that.

Thanks again.

radek8, 2020-11-20:
So the simulated Marlin mode also works

Doyle4, 2020-11-20:
Only thing not working in simulated mode is it will always so NO MEDIA,

oldman4U, 2020-11-20:
Regarding the SD Card always detected. When you look at the card holder, there is a small pin (green) which is pushed up against the red element once a card is inserted. You can also take a screwdriver or similar and short circuit those two elements. doing so, the TFT recognises an SD Card has been inserted. I expect your pin is pushing against the red element even no sd card is inserted, so try to push the pin away from the red element to see if this makes the TFT "release" the SD Card.

[image]

Doyle4, 2020-11-20:
@oldman4U Checked and thats ok.

I think what was/is, I have a micro SD card in my MKS SGEN L V2 mainboard for firmware updates, the LCD is reading it at boot. Issue I have is If i change it to onboard, I can no longer update my printer unless I remove the bottom panel each time (I own a Artillery Genius).

problems remain are:

Rotary LED non functioning
Cant access media in simulation mode unless simulation is default at startup.

oldman4U, 2020-11-20:
OK.

Then I would close this ticket join 1271 and open a new ticket for the second issue. Not many have your mainboard here, so hopefully someone can help.

Doyle4, 2020-11-20:
@oldman4U BTT emailed, going through that procedure now also as not really happy with the product, much prefer TFT35, so going to see if can swap.
Thanks very much for all your help and also too @radek8

closed 2020-11-20 by Doyle4

oldman4U, 2020-11-20:
👍🏻