bigtreetech/BIGTREETECH-TouchScreenFirmware #1271: knob led tft35-E3 v3.0 it does not work

ivindor, 2020-11-19:
Hi I wanted to ask to see if someone can solve my problem.
I have an ender 3 pro with a skr 1.4 turbo and the tft35-E3 v3.0 and I can't get the button led to work.
I am attaching my configuration in case you can help me solve my problem, thank you very much.

[conf marlin 2.0.7.2.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5568496/conf.marlin.2.0.7.2.zip)
[conf tft35-E3 v.3.0.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5568515/conf.tft35-E3.v.3.0.zip)

traffic-light, 2020-11-19:
That's strange. It should work.

You did turn in on in feature settings?
You can change the color over there

radek8, 2020-11-19:
Compile your own FW.
The precompiled BIN file is 2 months out of date

oldman4U, 2020-11-19:
@bigtreetech @Msq001 

Could you please take over support for all those users using several months old firmware because you are not able to merge PRs which are already available.

Thank you

traffic-light, 2020-11-19:
The bin files should be removed and replaced with a good instructive readme. (That's my opinion)

radek8, 2020-11-19:
Not every user is able to compile on their own

oldman4U, 2020-11-20:
@ivindor 

The LED of the knob should work independent of your Marlin or TFT settings. Please take a picture from Menu - Settings - Info and also from Menu - Settings - Feature (go down two pages) and attach it here.

Doyle4, 2020-11-20:
@ivindor You are not on your own, im trying to sort this also.

oldman4U, 2020-11-20:
Then please add the same photos here.

Doyle4, 2020-11-20:
[image]

Doyle4, 2020-11-20:
[image]

Doyle4, 2020-11-20:
[image]

oldman4U, 2020-11-20:
OK.

No idea what it could be.

If you want you can make the final test by removing all cables from the TFT, cut a USB cable and connect the red wire to 5V and the black wire to GND of the TFT to see if it works then.

Doyle4, 2020-11-20:
If the little trick works, this could mean the led can power on if the LCD was connected to a PSU as it has the option.

I find it odd, on my SKR 1.4T the LED worked fine, I wonder if it has to do with my board

Doyle4, 2020-11-20:
[image]

Doyle4, 2020-11-20:
[image]

traffic-light, 2020-11-20:
I think I know what happened. Can you dissable rotary knob idle and restart the screen?

Doyle4, 2020-11-20:
How would I disable it sorry?

Doyle4, 2020-11-20:
Changed Rotary knob idle LED to on, restarted screen, after reboot is was off, no LED function still

traffic-light, 2020-11-20:
[image]
the one on the bottom

Doyle4, 2020-11-20:
Tried with a LED colour set, enabled, restarted.

This time it stayed set to on and LED colour showed Blue, still no function.

oldman4U, 2020-11-20:
No.

This only shows if the TFT works properly or not.

Doyle4, 2020-11-20:
Not sure if helps, but the problems started when I installed it onto my MKS SGEN L V2.0 (2020 edition)

traffic-light, 2020-11-20:
> Tried with a LED colour set, enabled, restarted.
> 
> This time it stayed set to on and LED colour displayed set to Blue, still no function.

Did you try disable the Rotary Knob LED idle? (red dot next to it)

Doyle4, 2020-11-20:
yeah, tried both methods :(

radek8, 2020-11-20:
nemůže ten rámeček krytu bránit šíření světla LED diody kolem hřídele knoflíku?

Doyle4, 2020-11-20:
It could do for sure, but the enclosure I have for mine was designed to have the LED light shine through

ivindor, 2020-11-20:
I have compiled again with the most recent fimrware and I put the config knob on but it still does not turn on I attach photos and the config for you to review
[image]
[image]
[image]

ivindor, 2020-11-20:
plug it into an external source and it still doesn't work.
I also put a multimeter on the knob LEDs and they pass 4.98v

[image]
[image]

oldman4U, 2020-11-20:
Works on my TFT.

Have you tried to remove the TFT from the Ender 3 metal frame to see if it works then. Evergreen alarm;-)
[image]

ivindor, 2020-11-20:
> Works on my TFT.
> 
> Have you tried to remove the TFT from the Ender 3 metal frame to see if it works then. Evergreen alarm;-)
> [image]

I just tried it too and nothing does not light the led either

oldman4U, 2020-11-20:
Ok. Guess this is a hardware related issue because i am not aware that the
TFT firmware can cause this.

Sorry

Doyle4, 2020-11-20:
Could it be anything to do with Pins? on the SKR1.4T the Rotary LED worked.
Also whats different with the MKS SGEN L V2 is the pins for the black cable with 4 connections and 1 reset, on MKS there is no where for the reset connector.

Just a thought.

[image]

radek8, 2020-11-20:
Yes, on the MKS it is necessary to turn the connectors EXP1 and EXP2 by 180 °
There is no need to remap anything.
The EXP2 connector has a reset pin.
Even if you do not connect the reset pin to the RS232 connector, the reset will work.

Doyle4, 2020-11-20:
One of mine is 180, if i do both it fails as a whole.

radek8, 2020-11-20:
The ice on the rotary knob is switched by the processor.
so it has nothing to do with the type of board with Marlin.
Ice should be neopixel. The 4.98V power supply is OK.
The question is whether it is okay to wake them up.
Therefore, the diode may be defective or excited. you can't measure it with a voltmeter

radek8, 2020-11-20:
> Jeden z mých je 180, pokud udělám obojí, selže jako celek.

You have to turn them both over, otherwise the Marlin mod won't work

radek8, 2020-11-20:
What kind of display driver did you use in Marlin?
must be
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER

radek8, 2020-11-20:
[image]

radek8, 2020-11-20:
The LED at the knob is controlled by pin PC7
PC 7 is connected to connector EXP1
Therefore, if you connect it incorrectly (rotate 180 ° or connect to EXP2), the LED will not light up, or the input may be destroyed.

[image]

oldman4U, 2020-11-20:
@Doyle4 Tty the 5v and ground only and see if it works.

ivindor, 2020-11-20:
> The LED at the knob is controlled by pin PC7
> PC 7 is connected to connector EXP1
> Therefore, if you connect it incorrectly (rotate 180 ° or connect to EXP2), the LED will not light up, or the input may be destroyed.
> 
> [image]
> 
> Connect the cables properly, if that doesn't help, take a picture of how you're connected and connect the marlin configuration.h

[image]
my tft only works in ext3 I don't know why it doesn't connect in ext1-ext2

Doyle4, 2020-11-20:
Not manage to try the 5V trick yet.

Finding this all strange, think soon im going pretend there isn’t an LED.

Exp1 is connected exp1 and same with 2.
One of them had to be rotated, cant remember which. 
How I have it now is the only way it works, if i change to any other combination, whole lcd stops working.

radek8, 2020-11-20:
send photo of engagement

radek8, 2020-11-20:
@ivindor  what kind of record do you have marlin?

ivindor, 2020-11-20:
> @ivindor what kind of board do you have marlin?

#define SERIAL_PORT  0

/**
 * Select a secondary serial port on the board to use for communication with the host.
 * :[-1, 0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT_2  -1
  this??

radek8, 2020-11-20:
No.
What hardware do you have on Marlin? so send a photo of where the connector is going

radek8, 2020-11-20:
SKR 1,4 turbo?

ivindor, 2020-11-20:
> SKR 1,4 turbo?

yes i have skr 1.4 turbo

radek8, 2020-11-20:
Then you have it badly connected

ivindor, 2020-11-20:
> Then you have it badly connected

How do I connect it?

radek8, 2020-11-20:
You have to connect two cables. EXP1 and EXP2
In Marlin you have to set
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
Send configuration.h

I'm writing this for the second time. Please read what I write.

ivindor, 2020-11-20:
> You have to connect two cables. EXP1 and EXP2
> In Marlin you have to set
> #define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
> Send configuration.h
> 
> I'm writing this for the second time. Please read what I write.

You have to connect two cables. EXP1 and EXP2
In Marlin you have to set
#define REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
Send configuration.h

I'm writing this for the second time. Please read what I write.
I have it defined the same in my marlin that I put configuration.h in my first post

radek8, 2020-11-20:
You have an error here - configuration.h
You must comment on a line
// #define CR10_STOCKDISPLAY

radek8, 2020-11-20:
Then connect the 2 connectors EXP1 and EXP2
It has to work

ivindor, 2020-11-20:
[image]

> Then connect the 2 connectors EXP1 and EXP2
> It has to work

It is also commented

ivindor, 2020-11-20:
with that configuration it only connects in ext3 for a second that I take a picture

ivindor, 2020-11-20:
[image]
[image]

radek8, 2020-11-20:
You have an error in the attached file.
[image]

radek8, 2020-11-20:
[image]

ivindor, 2020-11-20:
ok sorry I compile it and try again to see if it works

[47 more replies]

closed 2021-03-28 by stale[bot]