bigtreetech/BIGTREETECH-TouchScreenFirmware #1273: BTT TFT35 sending M190 or M109 command yields M140 and M104 instead [Q] (Your Question?)

drlohwk, 2020-11-20:
I have this BTT TFT35 screen connected to SKR1.4Turbo. when I send M190 or M109 code using the terminal screen, the TFT sends M140 and M104 to the mainboard instead. any idea on how to fix this? I tried print from SD card using M190 code too but it always send M140. I suspect this inherently affect my NEOPIXEL event lighting as well. any pointers?

kisslorand, 2020-11-20:
The way I see it written, it is intentional, so the gcodes sent after an M109 or M190 are executed immediately, without waiting for the temp to reach to the desired value. However it is only if M109/M190 is sent from the TFT.

drlohwk, 2020-11-21:
this issue seems to prevent me to run Printer Event LEDs subroutine which I was told only can be implemented by using M109 and M190 to trigger the heating of both bed and hotends. not sure what I should do now.

// Support for Adafruit NeoPixel LED driver
#define NEOPIXEL_LED
#if ENABLED(NEOPIXEL_LED)
  #define NEOPIXEL_TYPE   NEO_GRB // NEO_GRBW / NEO_GRB - four/three channel driver type (defined in Adafruit_NeoPixel.h)
  #define NEOPIXEL_PIN     P1_24       // LED driving pin
  //#define NEOPIXEL2_TYPE NEOPIXEL_TYPE
  //#define NEOPIXEL2_PIN    P1_24
  #define NEOPIXEL_PIXELS 16     // Number of LEDs in the strip. (Longest strip when NEOPIXEL2_SEPARATE is disabled.)
  #define NEOPIXEL_IS_SEQUENTIAL   // Sequential display for temperature change - LED by LED. Disable to change all LEDs at once.
  #define NEOPIXEL_BRIGHTNESS 100  // Initial brightness (0-255)
  #define NEOPIXEL_STARTUP_TEST  // Cycle through colors at startup

  // Support for second Adafruit NeoPixel LED driver controlled with M150 S1 ...
  //#define NEOPIXEL2_SEPARATE
  #if ENABLED(NEOPIXEL2_SEPARATE)
    #define NEOPIXEL2_PIXELS      15  // Number of LEDs in the second strip
    #define NEOPIXEL2_BRIGHTNESS 127  // Initial brightness (0-255)
    #define NEOPIXEL2_STARTUP_TEST    // Cycle through colors at startup
  #else
    //#define NEOPIXEL2_INSERIES      // Default behavior is NeoPixel 2 in parallel
  #endif

  // Use a single NeoPixel LED for static (background) lighting
  //#define NEOPIXEL_BKGD_LED_INDEX  0               // Index of the LED to use
  //#define NEOPIXEL_BKGD_COLOR { 255, 255, 255, 0 } // R, G, B, W
#endif

/**
 * Printer Event LEDs
 *
 * During printing, the LEDs will reflect the printer status:
 *
 *  - Gradually change from blue to violet as the heated bed gets to target temp
 *  - Gradually change from violet to red as the hotend gets to temperature
 *  - Change to white to illuminate work surface
 *  - Change to green once print has finished
 *  - Turn off after the print has finished and the user has pushed a button
 */
#if ANY(BLINKM, RGB_LED, RGBW_LED, PCA9632, PCA9533, NEOPIXEL_LED)
  #define PRINTER_EVENT_LEDS
#endif

oldman4U, 2020-11-21:
Please check #1209 and see if this solves the problem.

kisslorand, 2020-11-21:
Did you test M109 and/or M190 from within a gcode?

discip, 2020-11-23:
This behavior was pointed out as early as **May 19 https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/663#commitcomment-39291724** and is still present.
Even if the *'Custom G-Code Commands'* contain this, it is not handled right.
### Only if this command is used in a *printjob* it gets executed.

Would be very handy, if this would be made functional and if only for the *'Custom G-Code Commands'* - section.

thank you in advance

RepRapster, 2020-11-28:
neopixel event leds on btt's tft screens don't work as m109 and m190 are intentionally changed to m104 and m140 the same for chamber m191 is changed to m141 but if you have dual screen ie marlin mode they will work, i have found the offending file responsible for the m code changes and have now got event leds working through tft though the added "and wait" to heating does put the tft in a loop until temperature is reached and will not respond to any other commands until such set temperature is reached unless reset is triggered, the offending file is interfaceCmd.c found in TFT, User, API the header in file is case109 and case190 in the first line of each there is the cmd index of position to change character +3 on m109 and +2 on m190 with the change character after the = which is set to 4 to change the 9 to a 4 and simply changing the 4 to a 9 in both case109 and case190 will stop the commands changing to m104 and m140 but like i said this is intentional  due to the processor loop induced!!!

Felix-ArduPilot, 2020-12-07:
Wouldn’t it be better to modify marlin to also show the LEDs when an M104 or M140 command is sent? That way the problem of the processor loop would be solved…

RepRapster, 2020-12-07:
yes it would be better for m104 and m140 to activate event leds for btt's tft screens but think that possibly linking event leds to m105 temp reports might be an easier option though i am not a programmer just someone with basic knowledge of it...

drlohwk, 2020-12-08:
This question is no longer a problem as @traffic-light has sorted the firmware to trigger the Print Event LED running on neopixel. So thanks to all and Merry Christmas.

guruathwal, 2020-12-14:
The reason that **M109** & **M190** are changed to **M104** & **M140** by the TFT is that the **M109** & **M190** are blocking g-codes, i.e. the printer will not respond to any other command until it reaches the target temperature. This will leave the TFT unusable as many background commands-responses are going on between the TFT and the printer continuously.
The user will not be able to use any other feature like moving any axis, homing, or even changing the target temperature.

RepRapster, 2020-12-14:
i have this working on my tft35 e3 v3 and only use m109 and m190 whilst printing from tft sd card, yes it does stop me inputting other commands until temps are reached but at the same i do not need to run other commands as i have set it to print, if errors occur the stop/cancel function will work once temps are reached but i still have the option to reset or power off if serious errors occur, it would be more prevalent for btt to workaround this by using subroutines or another way of handling this as marlin does...

discip, 2020-12-15:
I am not quite sure about this, but is there a way to let Marlin handle this, by pushing all the commands to it, instead of them being processed by the tft?

traffic-light, 2020-12-15:
For the meopixels, maybe 
For the encode LED, no

Felix-ArduPilot, 2020-12-15:
I still think, it should be possible, to tell marlin to use the LEDs for M104 and M140. In that case it wouldn’t hurt, that BTT has to change the commands to prevent being locked out for some time. Or is there a reason why it’s not done like that?

RepRapster, 2020-12-15:
marlin is a fantastic firmware and a credit to all the developers to which i gratefully thank all who has worked on it and with that i think this issue should falls on btt's development team and i don't see why they can't use m155 report to monitor the temps and use that info in the "and wait" part of m109 and m190 without stopping further inputs, the tft itself displays both temps and periodically updates them though not as often  as marlin but still does this without blocking any commands and think they could use this reporting to good effect !!!

traffic-light, 2020-12-15:
I don't see any solution to retreive the current event led mode.
This is build in into marlin

That's why i made my PR

RepRapster, 2020-12-15:
traffic-light, i downloaded your working branch of btt's tft for working event leds to try but when extracting it won't let me extract a couple of files so unfortunately i couldn't try it out though i am quite happy with the way the event leds work on my tft35 e3 v3 with the little modifications i made to interfaceCmd.c file to not change the gcodes

closed 2021-02-20 by stale[bot]