bigtreetech/BIGTREETECH-TouchScreenFirmware #1274: How can I not use the .ini file to update the firmware ?

Chukozor, 2020-11-21:
I'm really frustrated by the "maximum of 50 characters" for the code running at "pause/end" of a print, I'm used to compile the firmware and use the .bin where there were not this limitation of 50 characters.
I tried updating the firmware with the .bin file but it does not override the .ini.
How can I update the firmware ONLY by using the .bin file so I can write more characters for the code at the pause/end of print than the 50 limit of the .ini ?
Thank you

Doyle4, 2020-11-21:
if you still have the config.ini.CUR that is working, remove the .CUR and change it back to config.ini and drag it across with the firmware and fonts. Just a guess, how I have done it, but not exceeding the 50 max.

radek8, 2020-11-21:
There was an error in the comment. The maximum is 75 characters.
It is already fixed in the new ini file

radek8, 2020-11-21:
The limit does not apply to the end code you have in the slicer :-)

radek8, 2020-11-21:
By using current files, you will avoid many mistakes.
Compile the bin file yourself, it is obsolete or you download it from #1265

oldman4U, 2020-11-21:
There is no way to "not" use the ini file besides using a very old version of the firmware. The ini overrules all other configurations and can be used "stand alone".

closed 2020-11-21 by Chukozor