bigtreetech/BIGTREETECH-TouchScreenFirmware #1277: Trying to compile BIGTREE_TFT_V"_0 Fails

mquelcutti, 2020-11-21:
### Description

### Steps to reproduce

1. download lastest firmware
2.  open project in vs
3. select build

**Expected behavior**
Sucess

**Actual behavior**
Fails with errors
*** [.pio\build\BIGTREE_TFT35_V3_0\src\src\Libraries\Startup\stm32f2xx\startup_stm32f2xx.o] Error 1
Compiling .pio\build\BIGTREE_TFT35_V3_0\src\src\Libraries\fwlib\stm32f2xx\src\stm32f2xx_crc.o
*** [.pio\build\BIGTREE_TFT35_V3_0\src\src\Libraries\cmsis\stm32f2xx\system_stm32f2xx.o] Error 1
*** [.pio\build\BIGTREE_TFT35_V3_0\src\src\Libraries\fwlib\stm32f2xx\src\misc.o] Error 1
*** [.pio\build\BIGTREE_TFT35_V3_0\src\src\Libraries\fwlib\stm32f2xx\src\stm32f2xx_adc.o] Error 1
'arm-none-eabi-gcc' is not recognized as an internal or external command,
operable program or batch file.
*** [.pio\build\BIGTREE_TFT35_V3_0\src\src\Libraries\fwlib\stm32f2xx\src\stm32f2xx_can.o] Error 1
'arm-none-eabi-gcc' is not recognized as an internal or external command,
operable program or batch file.
*** [.pio\build\BIGTREE_TFT35_V3_0\src\src\Libraries\fwlib\stm32f2xx\src\stm32f2xx_crc.o] Error 1

### Hardware Variant

### TFT Firmware Version & Main Board Firmware details

### Additional Information
* Include a ZIP file containing your `Configuration.h` or use [Pastebin](https://pastebin.com/) and paste a link in this issue.
* Provide pictures or links to videos that clearly demonstrate the issue.
[image]

mquelcutti, 2020-11-21:
ps no changes were made to the files , just downloaded and tried to compile

oldman4U, 2020-11-21:
What is the creation date of the files inside the BIGTREETECH-TouchScreenFirmware-master folder?

mquelcutti, 2020-11-24:
> What is the creation date of the files inside the BIGTREETECH-TouchScreenFirmware-master folder?

hi level 21/11/2020
[image]

oldman4U, 2020-11-24:
Seems to be OK.

I expect a restart did not help, so I would recommend to delete current pio configuration. On macOS I delete the content of the folder /Users/myusername/.platformio/ and restart VSC after that. Check Google for the corresponding folder on Win.

Your problem is VSC/PIO related and it will not be the last time that you have a problem with it.

mquelcutti, 2020-11-25:
Hi deleting everything from (WINDOIWS) C:\Users\<USER>\.platformio and running build again worked thanks

closed 2020-11-25 by mquelcutti