bigtreetech/BIGTREETECH-TouchScreenFirmware #1278: [BUG] stuttering Print with BIGTREETECH TFT35-E3 V3.0

Bastlwastl84, 2020-11-21:
If printing from TFT a cylinder for testing spiralize mode and the ttf is at start-screen its stuttering. if i'm in a submenu, only about all 3-10 seconds a short stuttering. If the same file at marlin-mode no stuttering. tried
 higher Baud and the fast mode for gcode, but not made it better.
the High-value at z-axis is flickering, too.

Used the BTT TF Cloud (copied on it before directly at the MicroSD with cardreader, cause Wifi not working to transfer with Win10)

A lot of Blobs and Zits at Surface

Same file from same SD Card at Marlinmode plugged in at Mainboard are working without any Problems.

[Btt.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5578575/Btt.zip)

radek8, 2020-11-22:
If you have problems start with a lower communication speed.
500000 is a lot and interference may occur.
Try 250000 or better yet 115200

Bastlwastl84, 2020-11-22:
> If you have problems start with a lower communication speed.
> 500000 is a lot and interference may occur.
> Try 250000 or better yet 115200

with 115200 i started and gone step by step upwards, but was not better

radek8, 2020-11-22:
How old are your FW in TFT

Bastlwastl84, 2020-11-22:
> How old are your FW in TFT?

TFT FW downloaded today from the BTT Github and Marlin Bugfix today from the Marlin Github

radek8, 2020-11-22:
Did you download the precompiled BIN, or did you compile your own bin file?

oldman4U, 2020-11-22:
Hi radek.

I believe it is enough, so I made this ticket: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/issues/1282

Bastlwastl84, 2020-11-22:
> Did you download the precompiled BIN, or did you compile your own bin file?

Compiled with vs code. The config files incl STL  and Video and pictures are at zip-file

Bastlwastl84, 2020-11-22:
Same Config, but with now with 115200 Baudrate:
https://drive.google.com/file/d/19hSKkBuUWvKoLZ9B8F7ySwUMvJO7HY-n/view?usp=drivesdk

https://drive.google.com/file/d/19RmHP9KzZJHkLU3jw9eeTq7snzhh5mT0/view?usp=drivesdk

Bastlwastl84, 2020-11-22:
[25MM mittig _DURCHMESSER_210_HOCH DasFilament PETG lw 0.4mm lh 0.2mm if 10% ext1 225C bed 70C.gcode.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5579819/25MM.mittig._DURCHMESSER_210_HOCH.DasFilament.PETG.lw.0.4mm.lh.0.2mm.if.10.ext1.225C.bed.70C.gcode.zip)

Bastlwastl84, 2020-11-22:
With Baud 115200 if i want to connect from Touchscreen Mode to the onboard SD the TFT35 E3 v3 says: Error and reboots. This was not happening with Baud 250000 and 500000.

radek8, 2020-11-22:
Strange behavior.
Can you try to change some parameters?

[image]

Bastlwastl84, 2020-11-22:
> Strange behavior.
> Can you try to change some parameters?
> 
> [image]

All into 8,  16 or should i use 32?

radek8, 2020-11-22:
uncomment
#define AUTO_REPORT_SD_STATUS
#define SERIAL_FLOAT_PRECISION 4
#define M114_DETAIL
#define REPORT_FAN_CHANGE
#define HOST_ACTION_COMMANDS
#define HOST_PROMPT_SUPPORT

do you need a permit? I would comment on that.
#define CNC_COORDINATE_SYSTEMS

Bastlwastl84, 2020-11-22:
> uncomment
> #define AUTO_REPORT_SD_STATUS
> #define SERIAL_FLOAT_PRECISION 4
> #define M114_DETAIL
> #define REPORT_FAN_CHANGE
> #define HOST_ACTION_COMMANDS
> #define HOST_PROMPT_SUPPORT
> 
> do you need a permit? I would comment on that.
> #define CNC_COORDINATE_SYSTEMS

Ok. Let's have a try. Firmware flashed, 115200 Baud, resetted to factory defaults.

Bastlwastl84, 2020-11-22:
Disabled CNC... Same Problem with higher Buffer (also tested 32) at 115200 Baud

Bastlwastl84, 2020-11-22:
Next Try with rx Buffer 1024 and Xon Xoff

Bastlwastl84, 2020-11-22:
[image]
[image]

radek8, 2020-11-22:
I can confirm the same behavior when printing your file that you attached.
You press a cylinder that is cut like a continuous spiral. Therefore, the height of the Z axis is constantly changing and the display constantly updates the height of the layer.
Unfortunately, it uses the TFT processor 100%. as a result, the display does not manage to send data to Marlin.
If you click the More icon, printing will become continuous.
You will need to extend the time interval for updating the layer height value on the display.

kisslorand, 2020-11-22:
> Ihe height of the Z axis is constantly changing and the display constantly updates the height of the layer.
> Unfortunately, it uses the TFT processor 100%. as a result, the display does not manage to send data to Marlin.
The layer is queried only every 2 seconds and it's updated only if actual differs from the old one. Hence the update also happens only every 2 seconds.

radek8, 2020-11-22:
That's what I thought, too.
Watch the video made by Bastlwastl84
See at what intervals the layer height is updated.
I have the same behavior of printing the same file.

[Uploading 20201121_155031.zip…]()

Bastlwastl84, 2020-11-22:
the small edges came from freecad (setted there a high resolution). sometimes i print with 0,2mm Nozzle. than i need a high speed and high resolution. can try new solutions again next weekend. i hope some one had a good solution, cause at marlinmode its printing without any problems.

oldman4U, 2020-11-22:
The solution would then to print in Marlin mode using such a setting...

Bastlwastl84, 2020-11-22:
> The solution would then to print in Marlin mode using such a setting...

same Problem is with a high detailed Fur from Dogfigure with 0,2 nozzle with tft35 E3 V3

radek8, 2020-11-22:
No, the solution is to adjust the refresh interval of the layer height icon.
Currently, this does not happen once every 2 seconds as intended, but after each change in height

Bastlwastl84, 2020-11-22:
> No, the solution is to adjust the refresh interval of the layer height icon.
> Currently, this does not happen once every 2 seconds as intended, but after each change in height

where to find?

kisslorand, 2020-11-22:
Yup, in fact those 6 icons are redrawn every loop.What interval do you guys think it would be appropriate? I can make  a PR in a few minutes just cannot decide to delay all of those 6 icons refresh or only Layer.

radek8, 2020-11-22:
1 second would be optimal?

radek8, 2020-11-22:
if it is for all icons, the CPU load will be reduced.

kisslorand, 2020-11-22:
If I include the time, it will skip once in a while.
Just let me know guys what should it be. I was thinking about 500ms...

radek8, 2020-11-22:
500ms will be enough. Can you create a PR?
Thank you

kisslorand, 2020-11-22:
Let's try with only the Layer, I don't see how anything else would change so rapidly. PR done in a few minutes.

Bastlwastl84, 2020-11-22:
> 500ms will be enough. Can you create a PR?
> Thank you
What is a PR?
Where i could change the setting with the Time?
i could do next WE after Hospital.

radek8, 2020-11-22:
PR is Pull Requests
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pulls

kisslorand, 2020-11-22:
Here it goes: https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/1284

Bastlwastl84, 2020-11-22:
[image]
is it there at Move.c ?

radek8, 2020-11-22:
> Try this PR: #1284

Aren't you wrong?

kisslorand, 2020-11-22:
I messed up a little, commited to the wrong repository, check again.

kisslorand, 2020-11-22:
Of course I messed up again, a new commit is on the way.

radek8, 2020-11-22:
That's not it yet :-(
reDrawLayer (Z_ICON_POS);
has not changed

[image]

kisslorand, 2020-11-22:
Wait for it... :)

radek8, 2020-11-22:
Bastlwastl84  Can you test it?
[BIGTREE_TFT35_V3.0_E3.26.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5580494/BIGTREE_TFT35_V3.0_E3.26.x.zip)

Bastlwastl84, 2020-11-22:
> Bastlwastl84 Can you test it?
> [BIGTREE_TFT35_V3.0_E3.26.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5580494/BIGTREE_TFT35_V3.0_E3.26.x.zip)

i could test it after Hospital (friday/saturday)

radek8, 2020-11-22:
Tested.
Repair failed.
The layer height icon is constantly redrawn. (more often than 500ms)

kisslorand, 2020-11-22:
My bad, new commit uploaded.

radek8, 2020-11-22:
I'll try it tomorrow, I have to sleep

kisslorand, 2020-11-23:
I just tested it. Without the delay the cylinder stutters badly, with the 500ms delay prints nicely, without any hiccup. The layer is displayed with enough precision, I do not think anyone would need a higher refresh rate.

Bastlwastl84, 2020-11-23:
What did you changed? Can you Upload the file here?

radek8, 2020-11-23:
[BIGTREE_TFT35_V3.0_E3.26.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5585872/BIGTREE_TFT35_V3.0_E3.26.x.zip)

kisslorand, 2020-11-23:
> What did you changed? Can you Upload the file here?

This: #1284

Bastlwastl84, 2020-11-23:
> [BIGTREE_TFT35_V3.0_E3.26.x.zip](https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/files/5585872/BIGTREE_TFT35_V3.0_E3.26.x.zip)

Thank you. Will try it at Friday.

radek8, 2020-11-23:
It works great. Updating the Z axis is ideal.

The file to print has many segments, but at a communication speed of 250,000 and a set speed in the slicer, it prints well.
During acceleration to 115% - 120% are already being felt jerking. but it will already be caused by the great burden of something else.
https://youtu.be/UIMzwTqPtLE

PR #1284 (2020-11-27): Layer refresh on the tft delayed during print
### Requirements

BTT TFT or MKS TFT.

### Description

Delays the refresh of the layer on TFT during print.
### Benefits

Avoids stuttering (especially on vase mode when it is updated at every program loop).
### Related Issues
Hopefully fixes #1278
closed 2020-11-27 by bigtreetech